open62541/open62541 #2666: Decrypt the message encodeable

Sun-Dordje, 2019-04-25:
Hello the community,

I work on a NIDS for OPC UA. I use the tool Zeek to catch and parse opc ua's packets.
I have install the framework open62541 to have some client/server and dump the trafic to make some test.
When the security mode is None, there is no problem to have each fields.

When the security mode is basic256sha256 i have a problem, i want to decrypt the "Message : Encodeable Object"
but i dont know how it is really encrypted, there is my opc ua trafic on a pcap file:

[image]

I see the security policy define in the part 7: So the OpenSecureChannel use the AsymmetricEncryptionAlgorithm_RSA-OAEP-SHA1
when the security mode is basic256sha256 (http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256).
I have implement it with python.

There is my outputs from my NIDS when the OpenSecureChannel "basic256sha256" is catch.

[image]

When i catch the sendercertif, i build it like a hexa-file and i have the right certificat.

[image]

When i catch the encrypted message (wich is exactly the same as the pcap file), i want to decrypt it with RSAES-OAEP-DECRYPT and with the good private key.
But I dont know how can i do because to decrypt a message, the message must have 256 bytes. But here i have 504 bytes.
Maybe there is a padding or anything else but i tried to decrypt each 256 bytes block from the message i had catch, but no success.

If someone can help me it is very cool, sorry for my english.

thank you
Seunam

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlgiraud, 2019-04-25:
The encrypted part starts at the sequence header, which wireshark incorrectly tries to decrypt.
If you want to decrypt the message, you have to take the bytes starting from the sequence header up to the end of the packet. This should then correspond to 512 bytes in your case.

See this figure as well (OPC UA specification v1.04 part 6 page 45):
[image]

Sun-Dordje, 2019-04-26:
Thank you @mlgiraud 

> This should then correspond to 512 bytes in your case.

you right, so with this information i tried to decrypt the first 256 bytes block and the second.. but no success again.

my private key is kk.pem
so with this command: openssl rsa -in kk.pem -text -noout
i can extract the privateExponent and i put it on my python RSAES-OAEP-DECRYPT script (from the rfc3447). My script works when i encrypt and decrypt a random message with kk.pem, so i dont think its about my script but i think my problem is about the block to decrypt that i catch with my NIDS... Im not sur so if someone want my script i can put it here.

Any idea to how can i cut the block to decrypt?

Sorry 4 my english
Sun-Dordje

jpfr, 2019-06-06:
If you know C-programming, you can start from the ua_securechannel.c file to do the decryption.
Otherwise, we can not help a lot because we don't see the scripts you are using.

Reopen if a question specifically for open62541 remains.

closed 2019-06-06 by jpfr