open62541/open62541 #2668: Load XML NodeSet without compiling?

paddor, 2019-04-26:
Just getting started with this project, hopefully to replace another UA stack. So far we've been generating our own XML nodesets. We then provide the generated XML file to the server during startup.

Is there a way to do this with Open62541? Am I missing something or is it absolutely necessary to include a compilation step to be able to load a new nodeset?

jpfr, 2019-04-26:
We currently do not parse XML at runtime.
Renaming as a feature-request for future reference.

jpfr, 2019-08-01:
@matkonnerth has proposed a PR for this in #2937.

paddor, 2021-05-13:
How would one have to proceed to avoid having to compile a nodeset in advance?

Is dynamically inserting folders, objects, and variables during runtime the only way? Is this possible before starting the server?

matkonnerth, 2021-05-13:
maybe this helps: https://github.com/matkonnerth/nodesetLoader

closed 2021-09-23 by jpfr