open62541/open62541 #2672: Error in building the PubSub code in Microsoft Visual Studio 2017

jadhavshital, 2019-04-26:
Enviorment :
  - Visual Studio 2017 
  - CMAKE 3.13.2
  - Windows 7 64 bit

CMake Build Steps :  
  I have set below flags
   - UA_BUILD_EXAMPLES
   - UA_ENABLE_DISCOVERY
   - UA_ENABLE_METHODCALLS
   - UA_ENABLE_NODEMANAGEMENT
   - UA_ENABLE_SUBSCRIPTIONS
[image]

Open Solution open62541.sln
  - This solution was built and installed successfully.
  - But the Example solution 'open62541-examples.sln' is not building successfully.
  - The Pubsub code is throwing some errors.

I got following errors:-
35>tutorial_pubsub_connection.c
36>tutorial_pubsub_publish.c
37>tutorial_pubsub_subscribe.c
35>path\open62541\include\open62541/server_pubsub.h(120): error C2061: syntax error: identifier 'UA_KeyValuePair'
35>path\open62541\include\open62541/server_pubsub.h(122): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(126): error C2143: syntax error: missing ')' before '*'
35>path\open62541\include\open62541/server_pubsub.h(126): error C2143: syntax error: missing '{' before '*'
35>path\open62541\include\open62541/server_pubsub.h(127): error C2371: 'UA_NodeId': redefinition; different basic types
35>path\open62541\include\open62541/types.h(318): note: see declaration of 'UA_NodeId'
35>path\open62541\include\open62541/server_pubsub.h(127): error C2143: syntax error: missing ';' before '*'
35>path\open62541\include\open62541/server_pubsub.h(127): error C2059: syntax error: ')'
35>path\open62541\include\open62541/server_pubsub.h(133): error C2143: syntax error: missing ')' before '*'
35>path\open62541\include\open62541/server_pubsub.h(133): error C2081: 'UA_PubSubConnectionConfig': name in formal parameter list illegal
35>path\open62541\include\open62541/server_pubsub.h(133): error C2143: syntax error: missing '{' before '*'
35>path\open62541\include\open62541/server_pubsub.h(133): error C2059: syntax error: ')'
35>path\open62541\include\open62541/server_pubsub.h(161): error C2061: syntax error: identifier 'UA_DataSetMetaDataType'
35>path\open62541\include\open62541/server_pubsub.h(163): error C2143: syntax error: missing '{' before '*'
35>path\open62541\include\open62541/server_pubsub.h(164): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(172): error C2061: syntax error: identifier 'UA_DataSetMetaDataType'
35>path\open62541\include\open62541/server_pubsub.h(177): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(186): error C2061: syntax error: identifier 'UA_PublishedDataItemsTemplateConfig'
35>path\open62541\include\open62541/server_pubsub.h(188): error C2061: syntax error: identifier 'eventTemplate'
35>path\open62541\include\open62541/server_pubsub.h(188): error C2059: syntax error: ';'
35>path\open62541\include\open62541/server_pubsub.h(189): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(190): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(193): error C2143: syntax error: missing ')' before '*'
35>path\open62541\include\open62541/server_pubsub.h(193): error C2143: syntax error: missing '{' before '*'
35>path\open62541\include\open62541/server_pubsub.h(193): error C2059: syntax error: ')'
35>path\open62541\include\open62541/server_pubsub.h(199): error C2061: syntax error: identifier 'UA_ConfigurationVersionDataType'
35>path\open62541\include\open62541/server_pubsub.h(200): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(203): error C2061: syntax error: identifier 'UA_Server_addPublishedDataSet'
35>path\open62541\include\open62541/server_pubsub.h(203): error C2059: syntax error: ';'
35>path\open62541\include\open62541/server_pubsub.h(203): error C2059: syntax error: '<parameter-list>'
35>path\open62541\include\open62541/server_pubsub.h(210): error C2143: syntax error: missing ')' before '*'
35>path\open62541\include\open62541/server_pubsub.h(210): error C2081: 'UA_PublishedDataSetConfig': name in formal parameter list illegal
35>path\open62541\include\open62541/server_pubsub.h(210): error C2143: syntax error: missing '{' before '*'
35>path\open62541\include\open62541/server_pubsub.h(210): error C2059: syntax error: ')'
35>path\open62541\include\open62541/server_pubsub.h(227): error C2061: syntax error: identifier 'UA_ConfigurationVersionDataType'
35>path\open62541\include\open62541/server_pubsub.h(230): error C2061: syntax error: identifier 'publishParameters'
35>path\open62541\include\open62541/server_pubsub.h(230): error C2059: syntax error: ';'
35>path\open62541\include\open62541/server_pubsub.h(231): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(241): error C2061: syntax error: identifier 'UA_DataSetVariableConfig'
35>path\open62541\include\open62541/server_pubsub.h(243): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(244): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(247): error C2143: syntax error: missing ')' before '*'
35>path\open62541\include\open62541/server_pubsub.h(247): error C2143: syntax error: missing '{' before '*'
35>path\open62541\include\open62541/server_pubsub.h(247): error C2059: syntax error: ')'
35>path\open62541\include\open62541/server_pubsub.h(251): error C2061: syntax error: identifier 'UA_ConfigurationVersionDataType'
35>path\open62541\include\open62541/server_pubsub.h(252): error C2059: syntax error: '}'
35>path\open62541\include\open62541/server_pubsub.h(255): error C2061: syntax error: identifier 'UA_Server_addDataSetField'
[... 262 lines cut ...]
37>path\open62541\include\open62541/plugin/pubsub.h(71): error C2365: 'close': redefinition; previous definition was 'function'
37>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\corecrt_io.h(459): note: see declaration of 'close'
37>path\open62541\include\open62541/plugin/pubsub.h(75): error C2059: syntax error: '}'
37>path\open62541\include\open62541/plugin/pubsub.h(89): error C2143: syntax error: missing ')' before '*'
37>path\open62541\include\open62541/plugin/pubsub.h(89): error C2143: syntax error: missing ';' before '*'
37>path\open62541\include\open62541/plugin/pubsub.h(89): error C2059: syntax error: '*'
37>path\open62541\include\open62541/plugin/pubsub.h(89): error C2059: syntax error: ')'
37>path\open62541\include\open62541/plugin/pubsub.h(90): error C2059: syntax error: '}'
37>path\open62541\include\open62541/plugin/pubsub.h(103): error C2143: syntax error: missing ')' before '*'
37>path\open62541\include\open62541/plugin/pubsub.h(103): error C2081: 'UA_PubSubTransportLayer': name in formal parameter list illegal
37>path\open62541\include\open62541/plugin/pubsub.h(103): error C2143: syntax error: missing '{' before '*'
37>path\open62541\include\open62541/plugin/pubsub.h(103): error C2059: syntax error: ')'
37>path\open62541\plugins\include\open62541/plugin/pubsub_udp.h(15): error C2061: syntax error: identifier 'UA_PubSubTransportLayerUDPMP'
37>path\open62541\plugins\include\open62541/plugin/pubsub_udp.h(15): fatal error C1003: error count exceeds 100; stopping compilation
37>Done building project "tutorial_pubsub_subscribe.vcxproj" -- FAILED.
 
[image]

[image]

[image]

Kindly help.

Thanks you

Pro, 2019-04-30:
How did you enable PubSub?

You do not mention it in your text, but VS tries to build it. Did you manually set `UA_ENABLE_PUBSUB`?

Can you post the CMake settings you are using:

If you are using cmake-gui, go to "Tools > Show my Changes" and paste the content of "Command Line Options"

On the command line use `cmake -L` (or `cmake -LA` if you changed advanced variables)

I just tried to build the PubSub examples on my VS2017 and it works fine. The only change i made is: Use default CMake settings, and click on Advanced, then Enable `UA_ENABLE_PUBSUB`.

jadhavshital, 2019-05-01:
Yes, I have enabled it manually by adding variable in cmake, but it did not work. Then I tried to add UA_ENABLE_PUBSUB in preprocessor definition in C/C++ compiler in Project properties. Still it is giving me errors. 
    If your build worked, can you explain the procedure you have followed in detail?

Pro, 2019-05-01:
What do you mean by it did not work?

Any error message?

Manually setting the define will definitely not work.

Just clone the repo, open CMake Gui, enable Advanced settings and then enable PubSub

jadhavshital, 2019-05-06:
Thank you for your help.
Setting UA_ENABLE_PUBSUB flag in Advanced setting, pubsub code worked.

Thanks again.

closed 2019-05-06 by jadhavshital