open62541/open62541 #2673: Publish/Subscribe temperature value does not work?

theresa95, 2019-04-26:
## Description
I am trying to publish a temperature sensor value (datatype = double) but I am a little bit confused on how to address the temperature sensor node. 

## Background Information / Reproduction Steps
In order to publish a temperature sensor value, I have defined a TemperatureSensorType (ObjectType) because I want to add some other temperature sensors later on.
The TemperatureSensorType node does have three attributes held by variables: SensorName, Location, Temperature.

In order to instantiate one TemperatureSensor, I have written an "addTemperatureSensorInstance" method which is an adapted version from the open62541 tutorial.                                                                      

Now I want to publish the TemperatureValue of the instantiated TemperatureSensor node. In order to do so, I have added a method called "addTemperatureDataSetField" which is also an adapted version of the "addDataSetField" of the Publish tutorial. But I have add two input parameters in order to call this method with different nodeids.

I hope, I have done every correctly till now. Because this is the part where I need some help. The thing is: I have problems with understanding the concept with the nodeIds here: In order to publish the temperature value, I need to specify the nodeId of the Temperature Variable node (or from the whole TemperatureSensor node?). But how can I specify a nodeId when creating a TemperatureSensor instance with the "addTemperatureSensorInstance" method? I am only specifying the "QualifiedName" in this method (as it is done in the tutorial as well). But the QualifiedName is not a NodeId, or am I wrong? 

Ad Subscribing to this temperature value:
I have altered the original tutorial_pubsub_subscribe.c a bit so that I can subscribe to the timestamp AND to the temperature value. In the section where the algorithm loops over the fileds and prints well-known content types, I have added another else if to filter DOUBLE values. But how do I know which double values belongs to which node if I have more than one publishing temperature sensor nodes? Where can I specify that?

I have attached source files where you can look up those methods, please have a look!  If you need any further information on my question, please feel free to comment down below and I will go more into detail. I hope I have explained my issues understandable enough! Thank you very much in advance for any help you can possibly give me.

[ClientSubscriber.txt](https://github.com/open62541/open62541/files/3121144/ClientSubscriber.txt)
[TemperatureSensor _source.txt](https://github.com/open62541/open62541/files/3121145/TemperatureSensor._source.txt)
[ServerPublisher.txt](https://github.com/open62541/open62541/files/3121147/ServerPublisher.txt)
[TemperatureSensor_header.txt](https://github.com/open62541/open62541/files/3121148/TemperatureSensor_header.txt)

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): the latest one
 - [ ] Other OPC UA SDKs used (client or server): No OPC UA SDKs used.
 - [x] Operating system: OPCUA Server/Publisher Raspian (Raspberry Pi) ; OPCUA Client/Subscriber: Windows 10
 - [x] Self-contained code example attached
 - [x] Critical issue: for me it is critical! :)

jh-isw, 2019-04-28:
Hi @theresa95 unfortunately just by reading the text-files, I can't tell what your code is (not) doing. Could you provide us with a self-contained repo of your PublisherServer, so we can see what is going wrong?

Let me know if you need help with that.

johndzilva, 2019-04-30:
@theresa95 from your code, you have created an instance with initializing qualified name. In order to reference the node you need either the nodeid or the callback nodeid from the function

the second parameter in the `UA_Server_addObjectNode` function is NULL. Add the initialized nodeid in the second parameter like:

```
 UA_Server_addObjectNode(server, nodeId,
                             UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                             UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                             UA_QUALIFIEDNAME(nsIdent, name),
                             temperatureSensorTypeId, /* this refers to the object type
                                            identifier */
                             oAttr, NULL, NULL);
```
then you can use the same nodeid in the `addTemperatureDataSetField` function to add the dataset field. 

This code is vague to understand. If you need more help share your repo link

theresa95, 2019-05-02:
@johndzilva  and @jh-isw , I have now set up a repository with my files so you can try it out! In the repository, you will find the files to build the publish application. I have also corrected the publish application as suggested by @johndzilva . First I have added a temperature sensor instance by defining the nodeid (nsIndex = 1, numeric identifier = 2001) and passing it to the UA_Server_addObjectNode() method as 2nd parameter. Then I call the addTemperatureDataSetField() with the same nodeId (1,2001) to publish this node. 

I can run the publish example without errors, but in the subscribe example it only prints out the datetime value although the fieldcount of the dsm variable is 2! So I have managed to add a second publishing field which should be the temperature field. But it is not entering the "temperature parsing elseif branch" within the parsing published fields loop.  Is it because I add the temperature sensor instance as publishing field? Do I have to add the temp.value variable node WITHIN the temperature sensor instance??

What am I doing wrong here? In the subscribe application, how can I distinguish the temperature values if I have more than one temperature sensor instance ?

Please have a look at my repository, you can clone it by using:
git clone https://tlichten@bitbucket.org/tlichten/pubsub-with-open62541.git

Also have a look at the readme.txt file. I have written down some notes there as well. If you have any further questions, please let me know! I will try to help as much as I can in order to solve this issue!
Thank you so much in advance!

jpfr, 2019-05-02:
Your code is missing the addTemperatureSensorInstance method.

> UA_NodeId temp1NodeId = UA_NODEID_STRING(1, "temp.value"); 

Can you instead write the temperature to the node "ns=1;i=2001"?
I suspect you create the variable. But there is no value inside.

Can you read the temperature variable in the OPC UA server with UAExpert or a similar GUI client?

theresa95, 2019-05-02:
Hello @jpfr , what exactly do you mean by "missing the addTemperatureSensorInstance method"? I am calling the method in the run function of the ServerPublisher.c file. Through calling this method, I am creating an instance node of the TemperatureSensorType (ObjectType). This node has the numeric nodeId (1,2001).  As defined in the method "defineTemperatureSensorType", each TemperatureSensorType does have child variable nodes (for SensorName, Location and Value). The value variable node does have the string identifier "temp.value". And I think if I want to set the temperature value, I will have to address the value variable node, right?

If I call the "UA_Server_writeValue" method on the TemperatureSensor node (1,2001), do I update the temperature value then? 

I have UAExpert installed on my windows pc, but the publishing server is running on a raspberry pi. I will have to figure out how to access the running server on the raspi via UAExpert running on Windows.

jh-isw, 2019-05-03:
Hi @theresa95 
using wireshark, I can see that the temperature doesn't get published at all. The Data Field has the same length (34 bytes) like in the pub-example. Your temperature is missing.

By the way:
[image]

theresa95, 2019-05-03:
hey @jh-isw , 
thank you for your answer, I just saw this while building and fixed this error!
How is it that the fieldCount = 2 when running the subscribe sample? How can I add the temperature node?

jh-isw, 2019-05-03:
Hi @theresa95 
I have it:
[image]

Your TemperatureSensor has ns1;i=2001
But the temperature value iteself has ns1;i=20550 (which I found out with UAExpert)

Lines 204 through 210 do nothing useful, because you have no ns1;i="temp.value"
Line 216 is the money shot.
[image]

And to your Subscriber:
`"Received temperature: %f", //not %d`
%f is for floats/double, %d is for signed integers

theresa95, 2019-05-03:
@jh-isw  cool, now this works! Thank you so much for your help. 
But I have two more questions though:
1. If I create two TemperatureSensor instances now, how will I be able to publish the second one? In the "defineTemperatureSensorType(..)" method I define the temperature variable node with the nodeId = (1, "temp.value"). So this nodeId will be the same for each instantiated temperatureSensor node. But as you already said, it seems that the variable node gets also a numeric node id (1, 50220) as well.  So I guess I will be able to distinguish between the temperature value nodes only through the NUMERIC nodeIds, not the string identifier (temp.value). Is there a way to retrieve those numeric nodeIds _without_  looking them up on UAExpert first? Or is it possible to give them a custom numeric ID while instantiating the temperature value node?

2. When multiple temperature sensor node are instantiated, how is it possible to distinguish between them in the subscribe application. I mean, everytime I parse a double, it is automatically a temperature value. But how do I know which temperature value it is? I guess I will have to publish the location node as well, but then there is the question on how to assign each location to each double value. Does the subscriber have to know the exact sequence of the published values?

Thank you for your help so far, I hope you can help me one more time!

jh-isw, 2019-05-03:
1. A nodeId is unique within a server's address space. It cannot be assigned twice. And it is either numeric or string (or... the standard knows more nodeId types). To answer your question: Look at the [Objects Tutorial](https://open62541.org/doc/current/tutorial_server_object.html) again, especially the part with _UA_Server_translateBrowsePathToNodeIds(...)_.
2. Either it has to know the sequence or you publish an own datagram for each sensor.

jh-isw, 2019-05-09:
@theresa95 can you close your issues, when they are resolved for you?

theresa95, 2019-05-12:
@jh-isw  Thank you for your help! I really appreciate it :)!

closed 2019-05-12 by theresa95