open62541/open62541 #2682: fail to call UA_Server_new with error UA_STATUSCODE_BADNODEIDUNKNOWN

davesliu, 2019-04-29:
I use the latest version of open62541.c and .h files.  The return value of UA_Server_new() is NULL.
I use gdb to debug the and found it's cause by fail of line 23420 of UA_Server_initNS0().
The return value retVal  is UA_STATUSCODE_BADNODEIDUNKNOWN (0x8035000).
Anyone could help to solve issue 

#ifdef UA_ENABLE_HISTORIZING
    /* ServerCapabilities - HistoryServerCapabilities - AccessHistoryDataCapability */
    retVal |= writeNs0Variable(server, UA_NS0ID_HISTORYSERVERCAPABILITIES_ACCESSHISTORYDATACAPABILITY,
                               &server->config.accessHistoryDataCapability, &UA_TYPES[UA_TYPES_BOOLEAN]);

## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

davesliu, 2019-04-29:
When I disable the line  "#define UA_ENABLE_HISTORIZING" in open62451.h and comment UA_Server_AccessControl_allowHistoryUpdateUpdateData and UA_Server_AccessControl_allowHistoryUpdateDeleteRawModified in open62451.c. The return value of UA_Server_new() will be OK.

So how shall we configure the UA_ENABLE_HISTORIZING?  

Thanks if anyone could help.

davesliu, 2019-04-29:
I also tried to use open62541/examples/server_ctt.c source code and compile this example.
It also failed to call UA_Server_new. My OS is ubuntu16.04 64 bit and with gcc 5.4

========================
Breakpoint 1, main (argc=1, argv=0x7fffffffde58) at /hom/test/open62541/examples/server_ctt.c:446
446	    if(server == NULL)
(gdb) p server
$1 = (UA_Server *) 0x0
(gdb) l
441	int main(int argc, char **argv) {
442	    signal(SIGINT, stopHandler); /* catches ctrl-c */
443	    signal(SIGTERM, stopHandler);
444	
445	    UA_Server *server = UA_Server_new();
446	    if(server == NULL)
447	        return EXIT_FAILURE;
448	    UA_ServerConfig *config = UA_Server_getConfig(server);
449	
450	#ifdef UA_ENABLE_ENCRYPTION
(gdb)

davesliu, 2019-04-29:
For the server_ctt, it failed at line 801 of ua_server_ns0.c
=====================================
Hardware watchpoint 5: retVal

Old value = 0
New value = 2150891520
UA_Server_initNS0 (server=0x692010) at /home/test/src/open62541/src/server/ua_server_ns0.c:801
801	                               &server->config.insertDataCapability, &UA_TYPES[UA_TYPES_BOOLEAN]);
(gdb) bt
#0  UA_Server_initNS0 (server=0x692010) at /home/test/src/open62541/src/server/ua_server_ns0.c:801
#1  0x00000000004098a7 in UA_Server_new () at /home/test/src/open62541/src/server/ua_server.c:236
#2  0x00000000004053d1 in main (argc=1, argv=0x7fffffffde58) at /home/ldw/src/open62541/examples/server_ctt.c:445
(gdb) p server->config.insertDataCapability
$4 = false
(gdb) p /x retVal
$5 = 0x80340000
(gdb) l 801
796	    retVal |= writeNs0Variable(server, UA_NS0ID_HISTORYSERVERCAPABILITIES_MAXRETURNEVENTVALUES,
797	                               &server->config.maxReturnEventValues, &UA_TYPES[UA_TYPES_UINT32]);
798	
799	    /* ServerCapabilities - HistoryServerCapabilities - InsertDataCapability */
800	    retVal |= writeNs0Variable(server, UA_NS0ID_HISTORYSERVERCAPABILITIES_INSERTDATACAPABILITY,
801	                               &server->config.insertDataCapability, &UA_TYPES[UA_TYPES_BOOLEAN]);
802	
803	    /* ServerCapabilities - HistoryServerCapabilities - InsertEventCapability */
804	    retVal |= writeNs0Variable(server, UA_NS0ID_HISTORYSERVERCAPABILITIES_INSERTEVENTCAPABILITY,
805	                               &server->config.insertEventCapability, &UA_TYPES[UA_TYPES_BOOLEAN]);
(gdb)

davesliu, 2019-04-29:
It seems that  the file "tools/schema/Opc.Ua.NodeSet2.Minimal.xml" in below link  is not merged to the master branch. So we can't find UA_NS0ID_HISTORYSERVERCAPABILITIES_INSERTEVENTCAPABILITY in 
ua_server_ns0.c:801.

===========================================
https://github.com/open62541/open62541/pull/1862/commits/9511b6f66c52ddf0696c59a8e80fee6796d42300#diff-de41118085658fe4cd58a60c97ae7e8cR1209

Pro, 2019-04-30:
Thanks for your issue report!
This will be fixed with #2689

Can you please test that version and give feedback?

closed 2019-04-30 by Pro

davesliu, 2019-04-30:
Hi, Pro

To fix this issue. I only replaced the "tools/schema/Opc.Ua.NodeSet2.Minimal.xml" with the one in link https://github.com/open62541/open62541/commit/9511b6f66c52ddf0696c59a8e80fee6796d42300#diff-de41118085658fe4cd58a60c97ae7e8cR1209 and compile the example code and it works fine.

PR #2689 (2019-04-30): Set minimal nodesets for Historizing and Events
Fixes #2682
commit b9225f9f (2019-04-30): Historizing: Add minimal nodeset required for historizing

Fixes #2682