open62541/open62541 #2692: Server configuration and initialisation flaws

jojastahl, 2019-04-30:
## Description

The server configuration possibilities have some flaws in th current master, IMHO.

## The concerns

1. Calling `UA_Server_new` already does some initialisation, but the configuration is not yet setup. This leads to the server namespace index 1 to refer uri `UA_STRING_NULL`. This is since fd08d012d95b5837c90b18ba3fd7230b2c4d8ef6 as far as I can tell.

2. The custom logger cannot be setup before calling any of the `UA_ServerConfig_setXXX` methods, as they cleanup the complete config again. You could setup the logger after this call, but several other things, like `UA_SecurityPolicy` and `UA_ServerNetworkLayerTCP` copy this before you can change it.

3. `UA_ServerConfig_setXXX` do usefull things where many of them you don't want to repeat in your code. But they use a relative fixed default configuration. For example, you cannot setup a (dynamic) custom `config->applicationDescription`. You can change it afterwards, but 

   a. `UA_Server_new` uses the applicationUri before you can ever get access to the config. (See point 1).

   b. `createEndpoint` copies the applicationDescription into the endpoints that are created by UA_ServerConfig_setXXX kind of function before you can change it.

## Proposal

Split the monolithic UA_ServerConfig_setXXX functions so that someone who is interested to do so can setup a config default, and then probably change some aspects, add his network layers and enpoint configuration with some helpers that hide most of the complexity but leave some more customization possibilities.

1. Expose `setDefaultConfig` to the library consumer as `UA_ServerConfig_setDefaults`. This function shouldn't setup anything which copies something of its init values, especially `logger` and `applicationDescription`.

2. The library consumer could modify the values which he doesn't like from the defaults.

3. Expose `addDefaultNetworkLayers` as `UA_ServerConfig_addNetworkLayerTCP`

4. Introduce `UA_ServerConfig_addSecurityPolicy_None`, `UA_ServerConfig_addSecurityPolicy_Basic128Rsa15`, `UA_ServerConfig_addSecurityPolicy_Basic256` and `UA_ServerConfig_addSecurityPolicy_UA_SecurityPolicy_Basic256Sha256`

5. Add `UA_ServerConfig_addSecurityPolicy_all`

6. Introduce `UA_ServerConfig_addEndpoint` which takes the securityPolicyUri and securityMode and uses createEndpoint to setup the endpoint.

7. Add `UA_ServerConfig_addDefaultEndpoints`

8. `UA_Server_new` must not do anything that is based on config. This must be moved to `UA_Server_run_startup`.

Any comments welcome. I would probably prepare a PR if we agree on how / what to do.

jpfr, 2019-04-30:
Good analysis! Yes, let’s adress these points!

With respect to point 8: some things need to be done already before run_startup. For example populating ns0. So that the users can add nodes.

jojastahl, 2019-05-01:
Point 8 really seems to be difficult. Ns0 is not a problem as it is constant. But ns1 depends on config and I'm not sure if that could savely be moved to later because adding nodes will probably depend on it.It would be easier if config would be setup before UA_Server_new. What is the reason that you changed this in fd08d012d95b5837c90b18ba3fd7230b2c4d8ef6? Von meinem Samsung Galaxy Smartphone gesendet.

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/open62541/open62541/issues/2692#issuecomment-488088339",

"url": "https://github.com/open62541/open62541/issues/2692#issuecomment-488088339",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

jpfr, 2019-05-01:
The lifetime of the server config was very messy.
We initialized the config outside of the server. Then the config was copied into the server.
So changes at runtime were not reflected in the original config.
Then the config had to be deleted manually after the end of the server lifetime.

This is much much cleaner now.
And the API has become much simpler.

>Calling UA_Server_new already does some initialisation, but the configuration is not yet setup. This leads to the server namespace index 1 to refer uri UA_STRING_NULL. This is since fd08d01 as far as I can tell.

ns1 always needs to have the URI of the server's applicationdescription.
We can just use a datasource that copies over the URI from the config every time the array is read.

> The custom logger cannot be setup before calling any of the UA_ServerConfig_setXXX methods, as they cleanup the complete config again. You could setup the logger after this call, but several other things, like UA_SecurityPolicy and UA_ServerNetworkLayerTCP copy this before you can change it.

Let's just set the default logger as the first thing we do in the default config.
We are in full control over this. So we can set the logger apropriately.

> UA_ServerConfig_setXXX do usefull things where many of them you don't want to repeat in your code. But they use a relative fixed default configuration. For example, you cannot setup a (dynamic) custom config->applicationDescription. You can change it afterwards, but

Eventually we should have a way to parse the config from an config file.
I like TOML quite a bit. But other formats are nice as well.
https://github.com/toml-lang/toml
https://github.com/cktan/tomlc99

> createEndpoint copies the applicationDescription into the endpoints that are created by UA_ServerConfig_setXXX kind of function before you can change it.

That's problematic. There should be a *single source of truth* for the applicationdescription.
I think at some point we manually replaced the applicationdescription with the server config version **in the getendpoints service**.

jojastahl, 2019-05-08:
I just created a PR for splitting the configuration functions. This addresses points 1, 3, 4, 5 and 7.

jojastahl, 2019-05-08:
I just created a PR to fix the namespace 1 init problem. Since fd08d012d95b5837c90b18ba3fd7230b2c4d8ef6 the namespace 1 uri remains `UA_STRING_NULL` as the setup was done when the application URI could not have been set.

I opted now to set this to `UA_STRING_NULL` in `UA_Server_new`. Then I tried to find all functions using the namespace array and setting it to the URI from the application description the configuration if it still is `UA_STRING_NULL`.

This solution is not really what I was looking for - I would preffer setting this either before `UA_Server_new`, which is impossible with the new config system from fd08d012d95b5837c90b18ba3fd7230b2c4d8ef6, or in `UA_Server_run_startup`, which isn't possible either, as between these to functions Nodes will be added and probably (at least in my use case) `UA_Server_addNamespace` / `UA_Server_getNamespaceByName` is called with different namespaces, including the servers own namespace that equals the application Uri. So we need to set this up as soon as it may be used. That implies, that the application Uri must be setup as soon as possible, at least before adding nodes.

@jpft: Can you have a look on PR #2718? Should I add a check in `UA_Server_run_startup` to verify that namespace 1 still equals the application description?

jojastahl, 2019-05-08:
> ns1 always needs to have the URI of the server's applicationdescription.
> We can just use a datasource that copies over the URI from the config every time the array is read.

See my [previous comment](https://github.com/open62541/open62541/issues/2692#issuecomment-490430863) and PR #2718.

> Let's just set the default logger as the first thing we do in the default config.
> We are in full control over this. So we can set the logger apropriately.

I rechecked for the logger and found that I was wrong. The logger is not copied, but only referenced. So this is OK.

> Eventually we should have a way to parse the config from an config file.
> I like TOML quite a bit. But other formats are nice as well.
> https://github.com/toml-lang/toml
> https://github.com/cktan/tomlc99

Implementing this is not currently possible for me. Perhaps this should be moved to a seperate issue? Anyway, a programmatic way should exist for fine grained configuration. So that proprietary config sources can be implemented.

> That's problematic. There should be a _single source of truth_ for the applicationdescription.
> I think at some point we manually replaced the applicationdescription with the server config version [...]

I didn't find a prove that this is changed in code. In my opinion this is still copied from the endpoint. However, with the split config functions of my PR #2717 it is possible to first setup some defaults, than modify these, including the application description, than adding the rest of the configuration. 

IMHO once the UA_Server_run_startup has been called, config changes do not need to be considered. That would add to much complexity or how would you detect and treat addition or modification of a network layer?

jh-isw, 2019-05-08:
@jpfr you said "ns1 always needs to have the URI of the server's applicationdescription."
Where in the Specification can I find that? I don't want to question it. I'm just unable to locate this rule.
Thanks in advance.

jojastahl, 2019-05-10:
From my point of view, the PR are ready and this ticket can be closed. Or do you have any open points on this?

PR #2718 (2019-05-28): Fix setup of namespace 1 URI
Fix namespace 1 initialisation problem that was introduced in commit fd08d012d95b5837c90b18ba3fd7230b2c4d8ef6.

Fixes #2692
closed 2019-05-28 by Pro