open62541/open62541 #2693: problem read from UA_DataValue with servertimestamp and sourcetimestamp

bnadem, 2019-04-30:
## Description
when  variabe ist scalar,  the timesstamp "03/10/4752 07:15:23.165.016.200", i think is wrong value for servertimestamp!
when variabel is array (liek example below), i becomme no timestamp for servertimestamp or SOURCETIMESTAMP
here is examplar from json data
"VALUE": {
                "ARRAYDIMENSIONSSIZE": 0,
                "ARRAYLENGTH": {
                    "BINARYENCODINGID": "funturcheckein",
                    "MEMBERSIZE": 10,
                    "MEMSIZE": 8,
                    "NODEID": "ns=0;i=11",
                    "OVERLAYBLE": 1,
                    "TYPEINDEX": 1,
                    "TYPENAME": "Double"
                },
                "HASSERVERTIMESTAMP": 0,
                "HASSOURCEPICOSECONDS": 0,
                "HASSOURCETIMESTAMP": 0,
                "HASSTATUS": 0,
                "HASVALUE": 0,
                "STATUS": 0,
                "STORAGETYPE": "UA_VARIANT_DATA",
                "VALUE": {
                    "[0]": 1.0,
                    "[1]": 2.0,
                    "[2]": 3.0,
                    "[3]": 4.0
                },
                "arrayDimensions": 0
            },

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-05-01:
Which node are you reading?
Can you supply a minimal working code example where the error shows?

bnadem, 2019-05-01:
Node from class variable

bnadem, 2019-05-13:
UA_VariableAttributes *varAttributes = (UA_VariableAttributes *)nodeAttributes;
UA_Variant *help =  &varAttributes->value;
 UA_DataValue *datavalueType = (UA_DataValue *)help->data;

i read timestamp over datavalueType

teccheck, 2025-03-18:
I've tested it with the current version. It should work

closed 2025-03-18 by teccheck