open62541/open62541 #2702: Prosys OPC UA Client not working anymore with open62541

alex-lotz, 2019-05-03:
## Description

I know this is maybe not entirely related to open62541, but I am using the Prosys OPC UA Client that has worked fine with an older checkout of the open62541 library. After updating my local git clone, the Prosys client fails to connect to any server examples from the open62541 library. While attempting to connect, the Prosys Client displays the error message "Cannot discover applications from ...", which I interpret such that the client fails reading the server space for some reason.

Maybe this is not an issue of the open62541 library by itself, but something must have been changed that prevents the Prosys client to work. Do you have any experience with the Prosys Client, or can you recommend any other generic clients that particularly work on Linux?

## Background Information / Reproduction Steps

For reproducing the error, I cloned the most recent update of the open62541 library and built the examples in a standard way using cmake. I have been using the "server_mainloop" example for testing. After starting the server_mainloop example, I tried to connect the Prosys client with the above mentioned error.

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=ON ..
```

## Checklist

Please provide the following information:

 - [X] open62541 (most recent update of the master branch, from 3rd may 2019):
 - [X] Other OPC UA SDKs used: [Prosys OPC UA Client](https://www.prosysopc.com/products/opc-ua-client/)
 - [X] Operating system: Ubuntu 18.04 64 Bit

Pro, 2019-05-03:
Thanks for the report!

Do you see any output on the open62541 Server side?

Can you try to build with `-DUA_LOGLEVEL=100` (=Trace) and post the output of the server here?

alex-lotz, 2019-05-03:
```
[2019-05-03 12:06:18.456 (UTC+0200)] info/network	TCP network layer listening on opc.tcp://nb-alotz:4840/
[2019-05-03 12:06:25.730 (UTC+0200)] trace/network	Connection 5 | New TCP connection on server socket 3
[2019-05-03 12:06:25.730 (UTC+0200)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2019-05-03 12:06:25.780 (UTC+0200)] trace/network	Connection 5 | Activity on the socket
[2019-05-03 12:06:25.780 (UTC+0200)] trace/network	Connection 5 | Received a packet.
[2019-05-03 12:06:25.780 (UTC+0200)] trace/network	Connection 5 | No channel attached to the connection. Process the chunk directly
[2019-05-03 12:06:25.829 (UTC+0200)] trace/network	Connection 5 | Activity on the socket
[2019-05-03 12:06:25.829 (UTC+0200)] trace/network	Connection 5 | Received a packet.
[2019-05-03 12:06:25.829 (UTC+0200)] trace/network	Connection 5 | No channel attached to the connection. Process the chunk directly
[2019-05-03 12:06:25.829 (UTC+0200)] trace/network	Connection 5 | Process OPN message
[2019-05-03 12:06:25.829 (UTC+0200)] info/channel	Creating a new SecureChannel
[2019-05-03 12:06:25.829 (UTC+0200)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 0 | Decrypting chunk
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 0 | Chunk size before and after decryption: 132, 132
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 0 | Verifying chunk signature
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 0 | Decrypted and verified chunk with request id 1 and sequence number 1
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 0 | Sequence Number processed: 1
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 0 | Process an OPN on an open channel
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Generating new local keys
[2019-05-03 12:06:25.829 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Generating new remote keys
[2019-05-03 12:06:25.829 (UTC+0200)] info/channel	Connection 5 | SecureChannel 1 | Opened SecureChannel
[2019-05-03 12:06:25.878 (UTC+0200)] trace/network	Connection 5 | Activity on the socket
[2019-05-03 12:06:25.878 (UTC+0200)] trace/network	Connection 5 | Received a packet.
[2019-05-03 12:06:25.878 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Decrypting chunk
[2019-05-03 12:06:25.879 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Chunk size before and after decryption: 93, 93
[2019-05-03 12:06:25.879 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Decrypted and verified chunk with request id 2 and sequence number 2
[2019-05-03 12:06:25.879 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Sequence Number processed: 2
[2019-05-03 12:06:25.879 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Process a MSG
[2019-05-03 12:06:25.879 (UTC+0200)] debug/server	Sent ServiceFault with error code BadInvalidTimestamp
[2019-05-03 12:06:25.977 (UTC+0200)] trace/network	Connection 5 | Activity on the socket
[2019-05-03 12:06:25.977 (UTC+0200)] trace/network	Connection 5 | Received a packet.
[2019-05-03 12:06:25.977 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Decrypting chunk
[2019-05-03 12:06:25.977 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Chunk size before and after decryption: 57, 57
[2019-05-03 12:06:25.977 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Decrypted and verified chunk with request id 3 and sequence number 3
[2019-05-03 12:06:25.977 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Sequence Number processed: 3
[2019-05-03 12:06:25.977 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Process a CLO
[2019-05-03 12:06:25.977 (UTC+0200)] info/channel	Connection 5 | SecureChannel 1 | CloseSecureChannel
[2019-05-03 12:06:26.026 (UTC+0200)] trace/network	Connection 5 | Activity on the socket
[2019-05-03 12:06:26.026 (UTC+0200)] info/network	Connection 5 | Closed
```

Pro, 2019-05-03:
Thanks, looks like there is an issue with the timestamp from the client:

```
[2019-05-03 12:06:25.879 (UTC+0200)] debug/server	Sent ServiceFault with error code BadInvalidTimestamp
```

I guess Prosys Client sends a request with a timestamp set to 0, which is checked here:
https://github.com/open62541/open62541/blob/ec9a1573389e32ebb2872c970150924740187daf/src/server/ua_server_binary.c#L564

This was introduced here:
#2569

@jpfr this change was introduced due to CTT tests as mentioned in #2569

Is this check really necessary to pass the tests? Couldn't find anything about mandatory timestamp value in the spec.

It only says:
[image]

So the timestamp is only used for logging purposes. In the stack we do not even use this value at all.

jpfr, 2019-05-03:
This is now tested in the CTT. We must keep it in to be conformant.
PR #2705 proposes a configuration option where we can relax certain tests if required.

jpfr, 2019-05-04:
Fixed via #2705.
The behaviour can be configured now.

closed 2019-05-04 by jpfr