open62541/open62541 #2707: STM FreeRtos LWIP mbedtls

embbo, 2019-05-06:
## Description

I am trying to compile for stm32 (LWIP ,Freertos) certificate using mbetls libraries using Eclispe. 
-> Its works fine if I Disabaled  //#define UA_ENABLE_ENCRYPTION
-> after giving all the paths of mbedtls library i still have problem ""mbedtls_platform_entropy_poll' undeclared (first use in this function)""
Can anyone help about it .

Used CMake options:

../cmake .. -G "Eclipse CDT4 - Unix Makefiles" -DUA_ARCHITECTURE=freertosLWIP  -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_AMALGAMATION=ON -DMBEDTLS_INCLUDE_DIRS="/../mbedtls/mbedtls-development/build/include" -DMBEDTLS_LIBRARY="/../mbedtls/mbedtls-development/build/library" -DMBEDX509_LIBRARY="/../mbedtls/mbedtls-development/build/library" -DMBEDCRYPTO_LIBRARY="/../mbedtls/mbedtls-development/build/library"

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

embbo, 2019-06-24:
@Pro

Pro, 2019-06-24:
@cabralfortiss maybe you can help out with stm32 stuff?

cabralfortiss, 2019-06-25:
I never worked with encryption but I gave it a quick try. I pulled the https://github.com/ARMmbed/mbedtls repo, and tried to compiled to my platform. The error I receive is:

#error "Platform entropy sources only work on Unix and Windows, see MBEDTLS_NO_PLATFORM_ENTROPY in config.h"

So I think that's the problem. I took the configs/config-no-entropy.h (the name says literally that it has no entropy) paste it in the include/ folder, renamed as config.h and compiled without any problem. (this isn't news, the procedure is described in the repo). 

With my encryption library compiled, I set CMake for open62541 with the proper values. In your case I think you might have a problem since your MBEDXXX variables are pointing to the folder and not the library itself. Anyway, I pointed to the libraries, the include folder of the repo and compiled. I have my cool open62541 library with (I assume) encruption there. Of course, it doesn't have entropy, since "Platform entropy sources only work on Unix and Windows, see MBEDTLS_NO_PLATFORM_ENTROPY in config.h".

Now, I have a library, but I didn't test it and have no idea if it works. But it compiles

embbo, 2019-06-26:
@cabralfortiss , I have comment out these line , 
//    mbedErr = mbedtls_entropy_add_source(&pc->entropyContext,
//                                         mbedtls_platform_entropy_poll, NULL, 0,
//                                         MBEDTLS_ENTROPY_SOURCE_STRONG);

and Now i can also compile without  erorr but I am not it is great idea to comment these lines .

cabralfortiss, 2019-06-26:
I think it's a bad idea. Did you try as I did using the configs/config-no-entropy.h file?

embbo, 2019-06-26:
yes i did but it still have erorrs .. let me try it again will come to you back .
2. i am using this example from encryption.
*****************************
/* Load certificate and private key */
	UA_ByteString certificate;
	certificate.length = CERT_DER_LENGTH;
	certificate.data = CERT_DER_DATA;

	UA_ByteString privateKey;
	privateKey.length = KEY_DER_LENGTH;
	privateKey.data   = KEY_DER_DATA;

	/* Load the trustlist */
	size_t trustListSize = 0;
	UA_ByteString *trustList = NULL;

	/* TODO test trustList
        if(argc > 3)
            trustListSize = (size_t)argc-3;
        UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
        for(size_t i = 0; i < trustListSize; i++)
            trustList[i] = loadFile(argv[i+3]);
	 */

	/* Loading of a revocation list currently unsupported */
	UA_ByteString *revocationList = NULL;
	size_t revocationListSize = 0;

	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);

	UA_StatusCode retval =  UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840,
			&certificate, &privateKey,
			trustList, trustListSize,
			revocationList, revocationListSize);

	UA_ByteString_clear(&certificate);
	UA_ByteString_clear(&privateKey);

embbo, 2019-06-28:
@cabralfortiss  I did the same as you mentioned , it complies fin BUT when I call the encryptions functions,  i have same issue ::::::::: error: 'mbedtls_platform_entropy_poll' undeclared (first use in this function)

cmbahadir, 2019-06-28:
Hello @embbo and @cabralfortiss ,

I have also faced with the same issue and IMHO it is not logical to support entropy for "freeRTOSLWIP" architecture build of open62541, since it is only supported for Windows and Linux in the mbdetls side;

https://github.com/ARMmbed/mbedtls/blob/66b7edb10865809a972362ce362c62820075ade3/include/mbedtls/entropy_poll.h#L57

I am not familiar with entropy and encryption topics but it seems security policy plugins should be architecture dependent.

embbo, 2019-07-01:
@cmbahadir, exctaly what I am thiking , but i don't have any clue how to do it, i hope , someone from the group will help .

embbo, 2019-07-01:
// mbedErr = mbedtls_entropy_add_source(&pc->entropyContext,
// mbedtls_platform_entropy_poll, NULL, 0,
// MBEDTLS_ENTROPY_SOURCE_STRONG);

I think we can comment out this function , beacuse we are are using config-no-entropy and this is netropy based or we can change that fucntion with mbedtls_null_entropy

cmbahadir, 2019-07-01:
Hi @embbo ,

I am not sure about the impact of the commenting out mbedtls_entropy_add_source function. For me the safest way is to use open62541 without UA_ENABLE_ENCRYPTION, but i guess it is not useful for you. 

So as a result i am not sure how encryption would be supported on embedded platforms. :)

Regards

embbo, 2019-07-10:
@cabralfortiss . I tried to connect and it gives error Baddecooding Error...

embbo, 2019-07-18:
@Pro @cabralfortiss ...I think, you can replace "mbedtls_platform_entropy_poll" with mbedtls_null_entropy_poll it complies fine and i  think also works fine . I am still testing and will coem back to you . (I am talking about stm32 and freertos)

embbo, 2019-07-30:
@cabralfortiss .. it works fine

embbo, 2019-07-31:
can I use .PEM and .key string alternative instead key  use certificate.der and ker.key.der ? if anyone can help .. @Pro @Pro @cmbahadir

schroeder-, 2019-07-31:
You can use .PEM, you just have to add a '\n' at the end of the ByteString. 

`UA_ByteString loadCertificate(const std::string& cert, bool pem) {
	UA_ByteString certificate = UA_STRING_NULL;

	FILE *fp = NULL;
	auto err = fopen_s(&fp, cert.c_str(), "rb");
	if (!fp && err != 0) {
		UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER, "Could not open certificate file err_no: %s", err);
		return certificate;
	}

	fseek(fp, 0, SEEK_END);
	certificate.length = (size_t)ftell(fp);
	size_t size = certificate.length;
	if (pem)
		certificate.length++;
	certificate.data = (UA_Byte *)UA_malloc(certificate.length * sizeof(UA_Byte));
	if (!certificate.data) {
		fclose(fp);
		return UA_STRING_NULL;
	}

	fseek(fp, 0, SEEK_SET);
	if (fread(certificate.data, sizeof(UA_Byte), size, fp) < size)
		UA_ByteString_deleteMembers(&certificate); // error reading the cert
	fclose(fp);
	if (pem)
		certificate.data[certificate.length - 1] = '\0';
	return certificate;
}`

embbo, 2019-08-02:
@schroeder-  I don't want to read from card my intensions are just to write a string like this ..
UA_ByteString certificate = "-----BEGIN CERTIFICATE-----\n"
			"MIID0TCCArmDQEBCwUAMDwxFzAVBgNV\n"
	"BAMMDmp1YW5nYnVyZ29zIENBMSEwHwYDVQQKDBhqdWFuZ2J1cmdvcyBPcmdhbml6\n"
			
			"-----END CERTIFICATE-----";

schroeder-, 2019-08-02:
You need to a "\0" to your string like:

```
	UA_ByteString certificate;
	char cert[] = "-----BEGIN CERTIFICATE-----\n"
		"MIID0TCCArmDQEBCwUAMDwxFzAVBgNV\n"
		"BAMMDmp1YW5nYnVyZ29zIENBMSEwHwYDVQQKDBhqdWFuZ2J1cmdvcyBPcmdhbml6\n"
		"-----END CERTIFICATE-----\0";
	certificate.data = (UA_Byte*)cert;
	certificate.length = sizeof(cert) / sizeof(cert[0]);
```

embbo, 2019-08-05:
Thank you .. @schroeder-  with above, I got error BadEncodingLimitsExceeded..

embbo, 2019-08-07:
@schroeder-  to find size of strinf is problem here ..

embbo, 2019-10-14:
Can I only use .DER formate . can I used .pem, .cert  or .key for certification. anyone can aswer ?

saperlot, 2020-02-25:
i just also run in this problem. 
All calls of `mbedtls_entropy_add_source(&pc->entropyContext, mbedtls_platform_entropy_poll ....` should be removed from the stack, since `mbedtls_platform_entropy_poll` is not available on non win or linux plattforms and sources will be added in the Init function `mbedtls_entropy_init` depending on your mbedtls configuration. In my case, i define `MBEDTLS_NO_PLATFORM_ENTROPY` and `MBEDTLS_ENTROPY_HARDWARE_ALT` and implement my `mbedtls_hardware_poll` which gets a random number from a hardware random generator. 
Also check out the entropy size which get requested. It can be defined by `#define MBEDTLS_ENTROPY_MAX_GATHER                32`

And just to conclude all problems: Certifications and Keys need to be zeroterminated. 

I also run the code on a STM32, I am not able to test if it also works without the manual calls of mbedtls_entropy_add_source on a linux or win plattform, But i expect that it does. If so, somebody should remove those calls in the next version.
If i could i would add a new PR for this, but i can't do this from out network.

zxiang1996, 2020-03-26:
I want to use stm32 to set up a server, but I don't know how much ROM and ram I need, can you tell me

mohamedalielghoula, 2020-09-29:
Hello all
I use stm32 as an OPC server, and the timestamp starts at 01:00
I want to change this time manually or is there another way to get this time from the host automatically ?

NoelGraf, 2021-04-18:
Does your problem still exist? Have you tried the 1.2 version in the meantime?

embbo, 2022-09-27:
@NoelGraf id you try ?

NoelGraf, 2022-09-27:
@embbo No, I didn't try it.

embbo, 2022-09-27:
@NoelGraf   what your version are you working ?

NoelGraf, 2022-09-28:
@embbo Not sure what you mean? I just wanted to know if you could still detect the problem on the current master or on the 1.3 branch.

teccheck, 2025-03-18:
Support for FreeRtos was in 1.3. Because of the new event loop, it had to be reworked quite a bit. The merge for 1.4 is expected soon. As there's no further response, I'll close this issue

closed 2025-03-18 by teccheck