open62541/open62541 #2708: How to publish and subscribe random data using pubsub code?

jadhavshital, 2019-05-06:
The default pubsub code is publishing server's current datetime in "Datetime" datatype. When I am sending random data by using "rand()%40" it is not subscribed by subscriber.
I have attached the screenshots of same. (Highlighted lines are used for generating and publishing random values between 0 to 39.)
[image]
[image]

In Subscriber, they are comparing for datatypes received from publisher. So I am not able to get how for what datatype I should compare this.
[image]

Kindly help.

jh-isw, 2019-05-06:
The second parameter of UA_NODEID_NUMERIC() is not the value to publish, but the identifier of the node in the server's address space.

Your server needs to have a node that is UA_Int32 and you assign it your random value. This node is your publishedVariable, identified by a NodeID.

jadhavshital, 2019-05-07:
Can you please elaborate more in detail?

jpfr, 2019-05-07:
You should not set the writerId to a random value.
You should set the value in the variable in the information model that is referenced in the DataSet.

Please read the spec and example code.
We are glad to help. But you need to put in the work as well.

closed 2019-05-07 by jpfr