open62541/open62541 #2715: Should it be safe to shutdown an uninitialized Server?

ccvca, 2019-05-07:
## Description
Should it be safe to shudwon a server that has not been initialized by UA_Server_run_startup?

Currently this code crashes with a segfault, because UA_Server_run_shutdown calls mdnsd_step with a NULL pointer.

```
UA_Server *pServer = UA_Server_new();
configure_lds_server(pServer);
UA_Server_run_shutdown(pServer); /// Segfault in mdns library
UA_Server_delete(pServer);
```

## Background Information / Reproduction Steps

Used CMake options:

```
MDNSD_LOGLEVEL:STRING=300
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=ON
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=ON // <------
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=REDUCED
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: windows, linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-05-12:
> Should it be safe to shutdown an uninitialized Server?

Of course it should be safe.
Rewording the title...

Pro, 2019-05-13:
Thanks! Will be fixed with #2724

commit b6ad325a (2019-05-13): Stack: Only shut down mDNS if it was initialized before.

Fixes #2715
PR #2724 (2019-05-14): Stack: Only shut down mDNS if it was initialized before.
Fixes #2715
commit 8cb236af (2019-05-14): Stack: Only shut down mDNS if it was initialized before.

Fixes #2715
closed 2019-05-13 by Pro