open62541/open62541 #2716: UA_Server_run_startup is executed multiple times as server->state unused

ccvca, 2019-05-07:
## Description
UA_Server_run_startup is executed multiple times because this if-condition never triggers:

https://github.com/open62541/open62541/blob/004d7d23af82f2be366a3f0a6a3fef86dc249adb/src/server/ua_server.c#L395-L396

as server->state is never set to anything else than UA_SERVERLIFECYCLE_FRESH (UA_SERVERLIFECYLE_RUNNING is never used if the github search is right).

@jpfr  Whats the intention of server->state? If I get it right, it should be set to UA_SERVERLIFECYLE_RUNNING here:
https://github.com/open62541/open62541/blob/004d7d23af82f2be366a3f0a6a3fef86dc249adb/src/server/ua_server.c#L443
and reset to UA_SERVERLIFECYCLE_FRESH inside UA_Server_run_shutdown?

If that is the intention, I could do a pull request with that changes, but as I'm unsure about this I want to ask first.

## Background Information
As in the new configuration the registerServerCallback of mdns is (re)set to NULL, () here: https://github.com/open62541/open62541/blob/004d7d23af82f2be366a3f0a6a3fef86dc249adb/src/server/ua_discovery_manager.c#L90 I need a possibility call UA_Server_run_startup first, then set the callback (UA_Server_setServerOnNetworkCallback) and call UA_Server_run afterwards.
At the moment inside UA_Server_run the function UA_Server_run_startup is executed again and by set callback is reset to 0.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2019-05-07:
Why do you need to run UA_Server_run_startup before you set the callback?

Imho server->state is an artefact that we don't really need at this point.
And reducing complexity is always a good idea.

ccvca, 2019-05-07:
https://github.com/open62541/open62541/blob/004d7d23af82f2be366a3f0a6a3fef86dc249adb/examples/discovery/server_multicast.c#L270-L272

As the chain of function calls is at the moment:
`UA_Server_run_startup`->`UA_DiscoveryManager_init->dm`->`registerServerCallback = NULL;`

NoelGraf, 2021-04-18:
What is the status of the issue? Does your problem still exist?

ccvca, 2021-04-19:
After a quick check of the changes since 2019, I thick this is still the case.

teccheck, 2025-03-18:
The relevant code changed quite a bit since this issue was opened. For example, we don't have a server state anymore. Therefore, I'll close this issue

closed 2025-03-18 by teccheck