open62541/open62541 #2721: Nodeset compiler error: backend_open62541_nodes:Dimension with size 0 or value count mismatch detected, ArrayDimensions won't be copied to the Value attribute.

MartinaFelberFEM, 2019-05-10:
## Description

I am trying to setup a server with Euromap 83 and 77 types. I copied and adapted the CMakeLists.txt file for examples/nodeset from following branch: 
https://github.com/Pro/open62541/tree/test/euromap/examples/nodeset/euromap

And it worked on 2019-04-19. 

Today I pulled from master branch (latest commit: commit 4160545e0acc711e199dfd1abd097f9cff04ce15) and it doesn't compile anymore. 

I get following error(s) while generating euromap83 and euromap77 types:

backend_open62541_nodes:Dimension with size 0 or value count mismatch detected, ArrayDimensions won't be copied to the Value attribute.

I did not get further error information -> line number etc. 

I have seen that changes have been made to the nodeset compiler and wondered if this is the cause.

Thanks for your support.

## Background Information / Reproduction Steps
git clone https://github.com/open62541/open62541.git
git submodule update --init
mkdir build
cd build

cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=NO -DUA_BUILD_EXAMPLES=YES -G "Visual Studio 15 2017"

Download Euromap files:
http://www.euromap.org/en/euromap83
http://www.euromap.org/en/euromap77

Generate .bsd files with UaModeler (Unified Automation)

Change example nodeset CMakeLists.txt to generate euromap types -> see attached file.
[CMakeLists.txt](https://github.com/open62541/open62541/files/3166242/CMakeLists.txt)

Added server_euromap.c file to nodeset examples -> see attached file. (C file)
[server_euromap.txt](https://github.com/open62541/open62541/files/3166313/server_euromap.txt)

Build all with Visual Studio.

jpfr, 2019-05-10:
@pro you made some changes to the nodeset compiler lately.

It is time for unit tests now.
We should test against all known companion specs in the CI.

matkonnerth, 2019-05-10:
Hi, the error is raised in backend_open62541_nodes:208. Error is raised on variables defined in the euromap77 model:

NodeIds 6088, 6040, 6112 and 6145

All of these nodes have dataType="String" ValueRank="1" ArrayDimensions="0" and as Value a list of String with an empty string member.

ValueRank=1 ... 1-dimensional array
ArrayDimensions=0 ... array with unknown size (as described in OPC UA spec part 5)

So I think the nodeset is valid and constraint checking should handle the case when arrayDimensions=0 and a value element is there.

Pro, 2019-05-11:
Yep, there are some minor issues with specific nodesets. I will have a look at it and try to add more unit tests for well-known nodesets.

Pro, 2019-05-13:
I just created a new PR which should fix your issues.

A corresponding test branch is also available here:
https://github.com/Pro/open62541/tree/test/euromap/examples/nodeset/euromap

We can not easily integrate the euromap into unit tests because these files need some custom modifications (via UaModeler as mentioned above) to be usable.
Anyways this issue also showed up for the most recent v1.04 nodeset. Therefore it will be tested in the future.

MartinaFelberFEM, 2019-05-13:
Hello!

Thanks for your support!
I could succesfully build your branch with Visual Studio using following cmake command:
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=YES -G "Visual Studio 15 2017"

But when I execute the server_nodeset_euromap project I get a runtime exception at instantiation of "RemoveTargetVariables" method output argument. 
NodeId = 0, 15120

namespace0_generated.c line 11509
"variablenode_ns_0_i_15120_variant_DataContents0_arrayDimensions" is "nullptr".

----------------------------------------------------------------------------------------
/* OutputArguments - ns=0;i=15120 */

static UA_StatusCode function_namespace0_generated_428_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
attr.valueRank = 1;
attr.arrayDimensionsSize = 1;
UA_UInt32 arrayDimensions[1];
arrayDimensions[0] = 0;
attr.arrayDimensions = &arrayDimensions[0];
attr.dataType = UA_NODEID_NUMERIC(ns[0], 296);
UA_Argument variablenode_ns_0_i_15120_variant_DataContents[1];

UA_init(&variablenode_ns_0_i_15120_variant_DataContents[0], &UA_TYPES[UA_TYPES_ARGUMENT]);
variablenode_ns_0_i_15120_variant_DataContents[0].name = UA_STRING("RemoveResults");
variablenode_ns_0_i_15120_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[0], 19);
variablenode_ns_0_i_15120_variant_DataContents[0].valueRank = (UA_Int32) 1;
UA_STACKARRAY(UA_UInt32, variablenode_ns_0_i_15120_variant_DataContents0_arrayDimensions, 1);
UA_init(&variablenode_ns_0_i_15120_variant_DataContents0_arrayDimensions, &UA_TYPES[UA_TYPES_UINT32]);
**variablenode_ns_0_i_15120_variant_DataContents0_arrayDimensions[0] = (UA_UInt32) 0;**

----------------------------------------------------------------------------------------

I didn't change anything on the branch.
Thanks for your support

Pro, 2019-05-14:
Thanks for testing again!
I found the issue. The line `UA_init(&variablenode_ns_0_i_15120_variant_DataContents0_arrayDimensions, &UA_TYPES[UA_TYPES_UINT32]);` is not correct, i,e,, there shouldn't be any de-referencing.
The correct call is without the `&`: `UA_init(variablenode_ns_0_i_15120_variant_DataContents0_arrayDimensions, &UA_TYPES[UA_TYPES_UINT32]);`

This is now fixed. Please test again. I also updated my test/euromap branch.

commit b2c0f023 (2019-05-13): NodesetCompiler: Allow optional Description for Argument Type

Also fixes #2721
PR #2723 (2019-05-15): Update nodeset to current v1.04
Also fixes #2721
commit b1f246d4 (2019-05-15): NodesetCompiler: Allow optional Description for Argument Type

Also fixes #2721
closed 2019-05-14 by Pro