open62541/open62541 #2722: How to get a secure opcua server not accepting connection from non-secure opcua client?

bvinodkumar2008, 2019-05-10:
# Description
The [examples/client.c](https://github.com/open62541/open62541/blob/cca3ead344186f9c0aac0e177263c48c52e49c9c/examples/client.c) sample is able to connect to [examples/encryption/server_basic256sha256.c](https://github.com/open62541/open62541/blob/cca3ead344186f9c0aac0e177263c48c52e49c9c/examples/encryption/server_basic256sha256.c) sample, is there a way where the server code doesn't allow non-secure client to establish connection with a non-secure client i.e, only allow basic256sha256 security policy and reject None policy?

## Checklist

Please provide the following information:

 - [cca3ead344186f9c0aac0e177263c48c52e49c9c] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [ ubuntu 16.04] Operating system:
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

jpfr, 2019-05-12:
Yes. You can add a different set of endpoints to server config.
And you can customize the AccessControl plugin:

https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c

bvinodkumar2008, 2019-05-12:
Thanks @jpfr. Let me try this and get back to you with questions if any by EOD today

rudoq32, 2019-05-12:
I am building micro embedded profile application - non-secure. I am interested in about your (@bvinodkumar2008 ) results.

jpfr, 2019-05-13:
This should be interesting for most serious usage scenarios.
We can make a wiki page out of the learnings. And add it to the integrated documentation when ready.

bvinodkumar2008, 2019-05-14:
Hi @jpfr,
Sure, i can contribute to the documentation. I guess what we are missing here is the helper methods to be able to choose the security policy of choice both  for server and client.

After digging into the core API's like setDefaultEncryption() for secure_client.c (copy of client_encryption.c file with changes to add only basic256sha256 security policy), i was able to add only the required security policy but even after doing that i'm getting security policy rejected error from the secure_server.c (copy of server_basic256sha256 with changes to add only basic256sha256 enpoint) while the insecure client also receiving the same. 

Could you please take a look at the server and client source files at https://gist.github.com/bvinodkumar2008/bd849924ff41ca4581c81d2b7f46e0e0 and see what i could be possibly doing round? I've added the output of these there. 

Regards,
Vinod

bvinodkumar2008, 2019-05-18:
Thanks @jpfr . With the below logic In secure_server.c, i was able to achieve what i was looking for, so closing.

```
    for(int i = 0; i < config->endpointsSize; i++) {
        if(config->endpoints[i].securityMode != UA_MESSAGESECURITYMODE_SIGNANDENCRYPT) {
            config->endpoints[i].userIdentityTokens = NULL;
            config->endpoints[i].userIdentityTokensSize = 0;
        }
    }
```

closed 2019-05-18 by bvinodkumar2008