open62541/open62541 #2725: Porting open62451 to eCos 3.0 OS

Marioam, 2019-05-13:
Hello community,
our goal is to implement the open62541 library for ERTEC 200P-2 from Siemens based on eCos 3.0 OS. The cross compilation is done via Eclipse CDT with arm-none-eabi architecture under Windows 7. 

Unfortunately, the Profinet stack must be cross-compiled in the gnu89 standard. To work around this, the following two options can be made:
reprogram the source code from C99 to C89 and compile
or create a static library (*.a) from the source code and link it. 
arm-none-eabi-gcc.exe -c open62541.h open62541.c -w -flto -fno-fat-lto-objects -std=c99 -pipe -DUA_ARCHITECTURE_ECOS -DUA_ECOS_HOSTNAME=\"eCos-host\" -D__ECOS -IC:/ecos/include 
arm-none-eabi-gcc-ar.exe cr libopen62541.a open62541.o

However, both ways lead to the same linking errors (see snipped below): 
... 
c:/ertec200p/msys/1.0/local/bin/../lib/gcc/arm-none-eabi/5.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Temp\ccncnTsF.ltrans0.ltrans.o: in function `ServerNetworkLayerTCP_add': 
<artificial>:(.text+0xc0): undefined reference to `setsockopt' 
c:/ertec200p/msys/1.0/local/bin/../lib/gcc/arm-none-eabi/5.3.0/../../../../arm-none-eabi/bin/ld.exe: <artificial>:(.text+0x134): undefined reference to `getnameinfo' 
... 
c:/ertec200p/msys/1.0/local/bin/../lib/gcc/arm-none-eabi/5.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Temp\ccncnTsF.ltrans1.ltrans.o: in function `UA_DateTime_localTimeUtcOffset': 
<artificial>:(.text+0x41a0): undefined reference to `gmtime_r' 
c:/ertec200p/msys/1.0/local/bin/../lib/gcc/arm-none-eabi/5.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Temp\ccncnTsF.ltrans1.ltrans.o: in function `connection_write.lto_priv.0': 
<artificial>:(.text+0x4598): undefined reference to `send' 
... 
collect2.exe: error: ld returned 1 exit status 

Doe's somebody know what I'm missing? 

Thanks in advanced! 

 - [ ] open62541 Version (release number or git tag): current master branch
 - [ ] Other OPC UA SDKs used (client or server): none
 - [ ] Operating system: eCos 3.0
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-05-13:
@cabralfortiss do you have any suggestions?

To me this looks like there are some eCos specific system libraries missing for the linker.

cabralfortiss, 2019-05-13:
Yes I think so too. 

Look in your eCos source code for the missing functions. In eclipse, CTRL + H will help you find a specific funcition in folder and subfolders. "setsockopt" and "send" should definitely be there. Look for them, and check that they aren't behing a #ifdef somehow.

Marioam, 2019-05-22:
Thanks guys for your very quick replies. Meanwhile I still could't find a solution. 

It looks like the cause for the missing file dependencies used by OPC UA lies on the  configuretion file (ecc) of eCos, which was created by Siemens. 

I found configurable options in the ecc file to enable socket support, implement netdb and use time functions. 

Could you provide me with your ecc file for building the eCos system?

jpfr, 2021-04-20:
It seems we have no Ecos specialist in the group.
Closing until somebody steps up and/or we have a stable test environment.

closed 2021-04-20 by jpfr

Isarhamster, 2021-10-20:
It's been a while since this was asked and closed, but as I had the same issue, I would like to share some confidence that it is possible to port open62541 on ERTEC 200P-2 with moderate effort: 
I started with an eCos amagalmated open62541.c/h pair to make conversion to C89 easier.
I would recommend fixing the C89-issues directly in the Open62541 code - it is just converting all` for(size_t i=0;...)` into `size_t i; for(i=0;...)`. It would be nice if Open62541 was already C89 compatible, but no big deal.
(gcc does not support C99 properly (asm volatile is not supported but needed inside the profinet stack)).

Next, adaption to the Interniche-Stack inside the Profinet-Package needs to be done. Roughly:
```
typedef signed long int ssize_t; // Must be a signed type
struct timeval
{
    long tv_sec;     /* seconds */
    long tv_usec;    /* and microseconds */
};
typedef int     socklen_t;
#define AF_INET		2
#define AF_INET6    3
#define AF_UNSPEC AF_INET
#ifndef sockaddr_storage
#define sockaddr_storage sockaddr_in
#endif
#define ioctlsocket(a,b,c) t_setsockopt(a,SOL_SOCKET,b,c,sizeof(c))
#define ioctl ioctlsocket
#undef   UA_IPV6
#define  FD_SETSIZE  16
#define  UDP_SOCKETS    1  /* support UDP (DGRAM) under sockets API */
#define  SO_SELECT      1  /* support select() call */
#define UA_ARCHITECTURE_ECOS
#undef UA_getnameinfo
#define UA_send t_send
#define UA_recv t_recv
#define UA_sendto t_sendto
#define UA_recvfrom t_recvfrom
#define UA_htonl htonl
#define UA_ntohl ntohl
#define UA_close t_socketclose
#define UA_select(a,b,c,d,e) t_select(b,c,d, ((e)->tv_sec < 1) ? -1 : ((e)->tv_sec * 10))
#define UA_shutdown t_shutdown
#define UA_socket t_socket
#define UA_bind t_bind
#define UA_listen t_listen
#define UA_accept t_accept
#define UA_connect t_connect
#define UA_getaddrinfo getaddrinfo
#define UA_getsockopt(a,b,c,d,e) t_getsockopt(a,b,c,d,*e)
#define UA_setsockopt t_setsockopt
#define UA_freeaddrinfo freeaddrinfo
#define UA_gethostname gethostname
#define UA_getsockname t_getsockname
#define UA_inet_pton(af,src,dst) inet_pton(af, src, (char*) dst)
#if UA_IPV6
# define UA_if_nametoindex if_nametoindex
#endif

struct tm *gmtime_r(const time_t *t, struct tm *tp)
{
	*tp = *gmtime(t);
	return tp;
}

int gethostname(char* name, int len)
{
  // Implement something around PnpbNvGetpStationName();
}

// Dummy implementation for getaddrinfo - there is no multihoming on ERTEC200P, so this is good enough
int getaddrinfo(const char* hostname,
                const char* service,
                const struct addrinfo* hints,
                struct addrinfo** res)
{
	int retval = EAI_MEMORY;
	struct addrinfo* ai = 0;
		
	int len = sizeof(struct addrinfo);
	int addr_len = sizeof(struct sockaddr_in);
	ai = calloc(len);
	if (ai)
	{
		ai->ai_addr = calloc(addr_len);
		if (ai->ai_addr)
		{
			ai->ai_addrlen = addr_len;
			ai->ai_addr->sin_addr.s_addr = htonl(INADDR_ANY);
			ai->ai_addr->sin_family = AF_INET;
			ai->ai_addr->sin_port =htons(atoi(service));
			ai->ai_family   = AF_INET;
			ai->ai_flags    = 1; // AI_PASSIVE
			ai->ai_socktype = SOCK_STREAM;
			ai->ai_protocol = IPPROTO_TCP;
		}
		else
		{
			free(ai);
			ai = 0;
		}
	}
	else
	{
		retval = EAI_NONAME;
	}
	*res = ai;
	return (ai!=0) ? 0 : retval;
}

void freeaddrinfo(struct addrinfo *ai)
{
	if (ai)
	{
		free(ai->ai_addr);
		free(ai);
	}
}
```

Next issue: The inerniche stack inside the Profinet-API contains quite a few bugs - I cannot share changes to the stack as these are closed source but try for yourself and if you experience issues, come back and ask for advice. Most issues circle around the t_select and blocking.