open62541/open62541 #2726: PubSub: Parsing the DataSetMessage(s)?

theresa95, 2019-05-13:
## Description
Hey there, I have configured a OPC UA server which publishes six sensor values from the raspberry pi. Furthermore, I have altered the original tutorial_pubsub_subscribe.c a bit so that I can subscribe to the published sensor values. As per my understanding, the tutorial_pubsub_subscribe.c parses exactly ONE DataSetMessage which contains multiple DataSetFields. (As it is defined in the OPC UA Spec: )
[image]

Now to my question: From the perspective of a subscriber, I do not know how the sensor values are ordered within the DataSetMessage. I mean, well, I could find it out by running the publish and subscribe applications and look up the order of how the sensor values are printed out (because I know which values the sensors have normally).

But there has to be a better option than that.  
As seen in the image above, one DataSetMessage contains multiple DataSetFields. And as per my understanding, ONE DataSetField contains ONE DataSet. And according to the OPC UA Specification Part 14, the definition of a DataSet is as follows:

> A DataSet can be thought of as a list of name and value pairs representing an Event or A LIST OF VARIABLE VALUES. 

So in the tutorial_pubsub_subscribe.c sample, the value part of the name/value pair of each field represents a sensor value. But what does the name represent in the name/value pair? 
In the tutorial_pubsub_subscribe.c example I can access the value by: 
`UA_Double temp1 = *(UA_Double *)dsm->data.keyFrameData.dataSetFields[i].value.data;`

But if I try to access the fieldname of the dsm struct like this:
`UA_String fieldName = dsm->data.keyFrameData.fieldNames[i];`
it crashes because the fieldNames it is unable to read memory:
[image]

My question in a nutshell: I would like to know how my subscribing OPC UA Client can parse not only the sensor values, but ALSO the proper name of the sensor which the value belongs to.  

My sensor objects do also have other variables (location, type and unit) besides the sensor values. I would later on like to publish this variables as well and I would like to know which variables belong  to which sensor object. Maybe there is a way to publish the whole instantiated sensor object and not only its variable nodes ?

If you want to go through my code samples by yourself, please clone this repository:

If you need any further information on my question, please feel free to comment down below and I will go more into detail. I hope I have explained my issues understandable enough! Thank you very much in advance for any help you can possibly give me.

## Checklist

Please provide the following information:
 - [ ] Other OPC UA SDKs used (client or server):  NO
 - [x] Operating system: Windows 10 (for subscribing application) and Raspbian (OPC UA Server which publishes sensor values)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Self-contained code example attached
 - [x] Critical issue: For me it is critical as always ;)

jpfr, 2019-05-13:
> But if I try to access the fieldname of the dsm struct like this:
UA_String fieldName = dsm->data.keyFrameData.fieldNames[i];
it crashes because the fieldNames it is unable to read memory:

The fieldnames are only transmitted in the JSON encoding.
For the binary encoding, the fieldnames are not visible in the network message.
You have to read out the dataset definition initially to have the metadata and the order of fields.

theresa95, 2019-05-14:
Hello @jpfr , thank you for your fast reply! 

I would be very grateful if you could explain to me exactly how to parse this message.
Currently,  I only parse the KeyFrame-DataSetMessage which does not contain the dataset definition. But as per my understanding, the metadata and the order of the fields is defined in the DataSetMetaData. And as it is stated in the OPC UA specification Part 14: PubSub (Chapter 5.2.3 DataSetMetaData), there are multiple options for Subscriber to get the initial DataSetMetaData: 

[image]

Does that mean I will have to send a request to the OPC UA server so the server sends me the DataSetMetaData within a NetworkMessage?? So I will have to send that request to the server, wait for the response, parse the NetworkMessage for the DataSetMetaData struct, parse the fields and the order of the fields and then I can start subscribing to the published values?

How is requesting/receiving/parsing of the DataSetMetaData implemented on the OPC UA Client? Do I have to implement something on the OPC UA Server application as well or is this something the OPC UA Server can do on its own?

thank you for any help you can possibly give me!

jpfr, 2019-05-14:
You can enable in the build flag `UA_ENABLE_PUBSUB_INFORMATIONMODEL`.
Then the PubSub configuration is visible in the OPC UA information and can be accessed with a normal client.

The information model representation is standardized as of part 14 of the OPC UA spec.
But there are no fancy helpers methods at the moment.

Note that a more full-featured subscriber is implemented in an open PR.
https://github.com/open62541/open62541/pull/2631
That will land in the coming days.

theresa95, 2019-05-22:
Hello @jpfr , I have cloned the subscriber example from #2631 and I see that the subscriber implementation now uses the subscriber API calls (addPubSubConnection, addReaderGroup, addDataSetReader and addSubscribedVariables). I have looked through the code in detail but I have still problems with understanding how this subscriber implementation works. 

In the FillTestDataSetMetaData: As far as I understand, this method defines 4 fields on which to subscribe to, right? For each field, it defines a FIELD NAME as well! For example, the fourth field in this array of fields has the datatype "UA_TYPES_BOOLEAN" and the field name is "BoolToggle". Does this mean that if my server adds a DataSetField  with the fieldNameAlias = "BoolToggle", I can successfully subscribe to the published field?
Does the order of subscribed fields matter? Must the fieldNameAlias be treated as a "unique identifier" (similiar to the nodeid) in order to subscribe to the correct variable node?

> static void 
addDataSetField(UA_Server *server, int nsIndex, int numIdent, char* fieldNameAlias){
    /* Add a field to the previous created PublishedDataSet */
    UA_NodeId dataSetFieldIdent;
    UA_DataSetFieldConfig dataSetFieldConfig;
    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING(fieldNameAlias); // fieldNameAlias = "BoolToggle"
    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
    dataSetFieldConfig.field.variable.publishParameters.publishedVariable =
    UA_NODEID_NUMERIC(nsIndex, numIdent);
    dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
    UA_Server_addDataSetField(server, publishedDataSetIdent,
                              &dataSetFieldConfig, &dataSetFieldIdent);
} 

Please explain to me if I get it wrong! I would be very thankful for an explanation on how this works.

merlingeorge555, 2019-05-27:
Hello @theresa95,

Please find my answers in issue #2759 

I see the same question is been raised in that issue. It's bit difficult for me to track. If you see relevant questions can asked in multiple issues please add the reference issue number for quick tracking.

Thanks.

merlingeorge555, 2019-05-29:
Hello @theresa95,

Do you see this issue is solved? Please close this if my answer satisfy you.

Thanks.

cponcedeleon, 2019-06-05:
Hi @jpfr 

I'm trying to achieve almost the same thing as OP. You said:

> You can enable in the build flag UA_ENABLE_PUBSUB_INFORMATIONMODEL.
Then the PubSub configuration is visible in the OPC UA information and can be accessed with a normal client.

I have enabled that flag but I don't see that info anywhere. I assume it would be the DataSetMetaData node under the PDS object?

The value of that node is NULL when read by UaExpert. I tried to set it manually from the publisher like this:

`    UA_DataSetMetaDataType pMetaData = publishedDataSetConfig.config.eventTemplate.metaData;
    uint64_t fieldsSize = pMetaData.fieldsSize;
    pMetaData.fields = (UA_FieldMetaData*)UA_Array_new (pMetaData.fieldsSize + 1, &UA_TYPES[UA_TYPES_FIELDMETADATA]);
    UA_FieldMetaData_init(&pMetaData.fields[fieldsSize]);
    UA_NodeId_copy (&UA_TYPES[UA_TYPES_DATETIME].typeId, &pMetaData.fields[fieldsSize].dataType);
    pMetaData.fields[fieldsSize].builtInType = UA_NS0ID_FLOAT;
    UA_String strTmp = UA_STRING ("Datetime");
    UA_String_copy (&strTmp, &pMetaData.fields[fieldsSize].name);
    pMetaData.fields[fieldsSize].valueRank = -1;
    pMetaData.fieldsSize++;
    publishedDataSetConfig.config.eventTemplate.metaData = pMetaData;
    publishedDataSetConfig.config.itemsTemplate.metaData = pMetaData;
    const UA_PublishedDataSetConfig *newConfig = &publishedDataSetConfig;
    retVal = UA_PublishedDataSetConfig_copy(newConfig, &publishedDataSetConfig);`

I do this everytime I add a new DataSetField. `publishedDataSetConfig` is the object returned by `UA_Server_addPublishedDataSet`. However I can't seem to get it to set a value. 

Any help will be appreciated.

NoelGraf, 2021-04-18:
Does your problem still exist? Have you tried the 1.2 version in the meantime?

cpipero, 2021-04-18:
Sorry I haven’t tried it

teccheck, 2025-03-18:
This problem should be solved by now. Please reopen if necessary

closed 2025-03-18 by teccheck