open62541/open62541 #2727: Historical event reading issue

davesliu, 2019-05-13:
I saw it only implement the historial data in OPC UA server side in open62541 example code. Does it support historical event on OPC UA server side for open62541? Is there any API can be referred?

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

basyskom-meerkoetter, 2019-05-14:
No, historical events are not implemented (yet).

With the current implementation you would be able to implement a simple "Process Historian" but not an "Event Historian".

davesliu, 2019-05-15:
Thank you for your info

closed 2019-05-15 by davesliu