open62541/open62541 #2728: How initial metadata and how metadata is published if configuration version changes?

merlingeorge555, 2019-05-13:
## Description
I have four questions against the specification. If I'm not wrong the subscribers use the DataSetMetaData for decoding the values of a DataSetMessage to a
DataSet.

1. As of now ie. current implementation of OPCUA Pub/Sub how the initial metadata is published?
2. How exchange the DataSetMetaData between Publisher and Subscriber if the configuration changes?
3. Does mirror variable implementation required datasetmetadata from publisher to subscriber?
4. What is mirror variable?
## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

cpipero, 2019-05-14:
I also had some questions on this topic, because I am not sure that this configuration aspect of pub-sub is implemented yet.
I was testing with some of the examples and looked at the test fixtures and while the examples work correctly (even with my own custom nodes), the configuration info (like DataSetMetaData) is _NULL_ on the serve (publisher side).
For example, if I use the OPC Foundation .NET example I cannot configure a subscribed because it looks for that info to map the dataset fields. 
I will subscribe to this issue to see the replies in case I'm missing something that exists.

theresa95, 2019-05-14:
I have also problems understanding this concept as well. 
I am publishing the values of multiple sensor data (2 temperature, 1 pressure and so on) and I have already managed to subscribe to all of them. But I do not know neither the order of the published values, nor the fieldnames. Thus I cannot distinguish between those values. Therefore, I will have  to read out the dataset definition initially to have the metadata and the order of fields. But I do not know how to accomplish that.

I have already opened an issue (#2726 ) about that yesterday. If someone could let me know how this can be accomplished, I would be very grateful! :)

teccheck, 2025-03-18:
The specs define several ways to distribute metadata (see part 14). You can consume metadata from the publisher using a client

closed 2025-03-18 by teccheck