open62541/open62541 #2735: Alarm & Conditions

davesliu, 2019-05-15:
I saw there was a pull request to implement the Alarm & Conditions feature.
 https://github.com/open62541/open62541/pull/2083

When will it be merged to master branch?

Thanks.

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ursfassler, 2019-12-11:
Do you have an idea or plan when that the Alarm & Conditions feature (#2083) will be merged? We do like to use that feature. Is it possible to support the merging of the branch?

chruetli, 2019-12-17:
I would like to join the question. Is there a plan to do the merge or is there a blocker?

Pro, 2019-12-17:
@jpfr I will merge #2083 after a rebase. IMHO it's in a good state to be integrated into master as an experimental feature. It's been open long enough and avoiding bit rot there is quite some task.

ccvca, 2019-12-19:
@chruetli The merge has been done
@Pro Many thanks for merging! I think this can be close now.

closed 2019-12-19 by Pro