open62541/open62541 #2736: triggerEvent: can't set optional event fields

MartinaFelberFEM, 2019-05-15:
Hello!

I am currently experimenting with Pro's euromap test branch: 
https://github.com/Pro/open62541/tree/test/euromap/examples/nodeset/euromap

I can successfuly build and run the server_nodeset_euromap project on Windows. 
cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=NO -DUA_BUILD_EXAMPLES=YES -G "Visual Studio 15 2017"

Then I tried to fire some Euromap events (e.g. CycleParametersEventType). 
The event fields with mandatory modelling rule work as expected. I can see the event field values in UaExpert and WireShark. But optional fields do not work.

When I setup the event fields the function UA_Server_writeObjectProperty_scalar returns with BadNoMatch statuscode for optional event fields.

I don't think that I have accidently used a wrong qualified name for all optional fields. The event fields are all in the same type hierarchy (under CycleParametersEventType) and have the same namespace and mandatory fields work as expected.

I didn't change anything on the branch, except for following server_nodeset_euromap.c file. The code is adapted from server events tutorial. 

[server_nodeset_euromap.txt](https://github.com/open62541/open62541/files/3183057/server_nodeset_euromap.txt)

I think that the optional event fields are not generated from the nodeset compiler. But I can't find any differences in namespace_euromap83_generated.c and namespace_euromap77_generated.c between mandatory fields like "CurrentLotName" and optional fields like "AverageCycleTime" of event type CycleParametersEventType. Except for the modelling rule reference. 

Do I have to enable the generation of optional fields?
Or do I have to create a subtype of the eventtype, which overwrites all optional fields to mandatory?

Thanks for your support!

Pro, 2019-05-15:
An Event is created by instantiating a specific event type.
Only mandatory child nodes are added to the instance. This is the default behaviour for any other node, which is created from a specific type.

If you need to set optional child nodes, i.e., event fields, then currently the only way is to create a custom event subtype, which overwrites the modelling rule from optional to mandatory.

We may need to find a better way in the future to easily allow overriding a modelling rule for type instances. Not only for events, but also especially for objects.

jpfr, 2019-05-15:
Hey @Pro,
we could support this case with a helper API.
Subtyping the event type is the "nuclear option".
We could have a method to set the event fields.
- If the field variable exists --> set the value
- If the field is optional --> create the variable and set the value
- Else --> return an error code and log

MartinaFelberFEM, 2019-05-16:
Thank you very much for your answers. 
Subtyping and overriding the optional fields to mandatory works!

Pro, 2019-05-16:
Perfect!

I changed the tile of the issue and marked it as a feature request.

To summarize:

We need a method in the helper API which performs the steps as suggested by @jpfr:

   * If the field variable exists --> set the value
   * If the field is optional --> create the variable and set the value
   * Else --> return an error code and log

Starting point is here:
https://github.com/open62541/open62541/blob/84f499b9320f6993213010368480e3ccf307e81a/examples/tutorial_server_events.c#L63

E.g., we need a method `UA_Server_writeObjectProperty_scalar_optional` which does the aforementioned steps for all objects, not only events. (@jpfr do you have other name suggestions?)

jpfr, 2019-05-17:
We could just make it the default behaviour.

Muddasir-Shakil, 2022-09-20:
Hi @Pro  and @jpfr 
> * If the field is optional --> create the variable and set the value
is there any option to check if the property/variable is optional? if not then it should be checked on the ObjectType of the ObjectNode and then instantiate under ObjectNode, correct?