open62541/open62541 #2737: Receiving event notifications only for first subscription

markekk, 2019-05-16:
## Description
I have been working in project where we implement OPC UA server and client applications using open62541 library.

Server has four different event types triggered from separate nodes. Each event has their own custom fields.

Client application subscribes those four events separately and it seems that event notifications are received only for first subscription. By chancing order of subscription it is possible to confirm that transferring each event type is possible.

## Background Information / Reproduction Steps
Noticed that following log entry is written when event callback is "skipped":
```
[2019-05-16 08:56:38.350 (UTC+0300)] debug/client[0m	Could not process a notification with clienthandle 3 on subscription 3
```

I noticed that UA_Client_MonitoredItem::monitoredItemId was compared to UA_EventFieldList::clientHandle in code below. Could this be the problem?

```
static void
processEventNotification(UA_Client *client, UA_Client_Subscription *sub,
                         UA_EventNotificationList *eventNotificationList) {
    for(size_t j = 0; j < eventNotificationList->eventsSize; ++j) {
        UA_EventFieldList *eventFieldList = &eventNotificationList->events[j];

        /* Find the MonitoredItem */
        UA_Client_MonitoredItem *mon;
        LIST_FOREACH(mon, &sub->monitoredItems, listEntry) {
            **if(mon->monitoredItemId == eventFieldList->clientHandle)
                break;**
        }

        if(!mon) {
            UA_LOG_DEBUG(&client->config.logger, UA_LOGCATEGORY_CLIENT,
                         "Could not process a notification with clienthandle %u on subscription %u",
                         eventFieldList->clientHandle, sub->subscriptionId);
            continue;
        }

        if(!mon->isEventMonitoredItem) {
            UA_LOG_DEBUG(&client->config.logger, UA_LOGCATEGORY_CLIENT,
                         "MonitoredItem is configured for DataChanges. But received a "
                         "EventNotification.");
            continue;
        }

        mon->handler.eventCallback(client, sub->subscriptionId, sub->context,
                                   mon->monitoredItemId, mon->context,
                                   eventFieldList->eventFieldsSize,
                                   eventFieldList->eventFields);
    }
}
```

## Checklist
 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 16.04
 - [] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-05-16:
Thanks for the issue report.
I just created a Pull Request yesterday, which should fix the clientHandle comparison: #2734 

Also, depending on your use case you should make sure that the queueSize for the monitored item is large enough.

I.e., in the client:

```
    UA_MonitoredItemCreateRequest item;
    UA_MonitoredItemCreateRequest_init(&item);
    
    // set other parts of item
 
    // queueSize which fits your needs
    item.requestedParameters.queueSize = 20;

    UA_MonitoredItemCreateResult result =
            UA_Client_MonitoredItems_createEvent(client, subId,
                                                 UA_TIMESTAMPSTORETURN_BOTH, item,
                                                 nullptr, &eventHandlerCallback, nullptr);
```

Can you check with the code from my PR, if that fixes it.

markekk, 2019-05-16:
I built new binaries from master and now my client application receives notifications for all the subscribed events. 

Thanks for the quick response.

closed 2019-05-16 by Pro