open62541/open62541 #2738: Fire complex event with ObjectTypes

MartinaFelberFEM, 2019-05-16:
Hello!

I am currently experimenting with Pro's euromap test branch and the server events tutorial:
https://github.com/Pro/open62541/tree/test/euromap/examples/nodeset/euromap

I can successfuly build and run the server_nodeset_euromap project on Windows.
cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=NO -DUA_BUILD_EXAMPLES=YES -G "Visual Studio 15 2017"

And I could trigger some simple events. For example:
Create Euromap event "CycleParametersEventType" and set field "BoxGoodPartsCounter": 
[image]

I used the function UA_Server_writeObjectProperty_scalar as described in the events tutorial with the qualified name - UA_QUALIFIEDNAME(3, "BoxGoodPartsCounter")

But I don't know how I can trigger a complex event, which contains several ObjectTypes. The Euromap specification requests to subtype the CycleParametersEventType and add some ObjectTypes. 

[image]

How can I set the BackPressure event field?
I would have expected to find a function which takes a browsepath, but I couldn't find any. And UA_Server_writeObjectProperty_scalar only takes one qualified name. 

Many thanks in advance for your support!

Pro, 2019-05-16:
You first need to find the NodeId of the `InjectionUnitCycleParameters_1` node inside the created event instance.
You can use `UA_Server_translateBrowsePathToNodeIds` starting at the new EventNodeId, or you can also use `UA_Server_forEachChildNodeCall` or `UA_Server_browse` to get the child nodes.

As soon as you have the NodeId of the injection unit, you can again use `UA_Server_writeObjectProperty_scalar`

MartinaFelberFEM, 2019-05-16:
Thanks for your answer. 

I had to change the test eventfield because Backpressure is optional. So I am using _CushionVolume_ instead. 

Browsing for the NodeId of _InjectionUnitCycleParameters_1_ inside the new created EventNodeId worked, but I couldn't set the CushionVolume event field with function _UA_Server_writeObjectProperty_scalar_. The statuscode is BadNoMatch.

Although I could successfully browse the event field CushionVolume, get its nodeId and write the value directly with _UA_Server_writeValue_ function.

I used UA_Server_browseSimplifiedBrowsePath to browse for the nodeIds. _UA_Server_translateBrowsePathToNodeIds_ works as well and I could browse for the whole path too. So browsing seems to work.

```c
UA_NodeId Event_InjectionUnitCycleParameters_1Id;
UA_NodeId_init(&Event_InjectionUnitCycleParameters_1Id);

name = UA_QUALIFIEDNAME(5, "InjectionUnitCycleParameters_1");
bpr = UA_Server_browseSimplifiedBrowsePath(server, *outId, 1, &name);
if (bpr.statusCode != UA_STATUSCODE_GOOD || bpr.targetsSize < 1) {
   UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "UA_Server_browseSimplifiedBrowsePath - (5, InjectionUnitCycleParameters_1) did not work");
}
else {

   UA_NodeId_copy(&bpr.targets->targetId.nodeId, &Event_InjectionUnitCycleParameters_1Id);
   name = UA_QUALIFIEDNAME(4, "CushionVolume");    // attention: only use mandatory event fields
   bpr = UA_Server_browseSimplifiedBrowsePath(server, Event_InjectionUnitCycleParameters_1Id, 1, &name);

   if (bpr.statusCode == UA_STATUSCODE_GOOD && bpr.targetsSize >= 1) {
      // CushionVolume
      UA_Double CushionVolume = 4711.4711;

      // TODO: - does not work
      uStatus = UA_Server_writeObjectProperty_scalar(server, Event_InjectionUnitCycleParameters_1Id, UA_QUALIFIEDNAME(4, "CushionVolume"),
         &CushionVolume, &UA_TYPES[UA_TYPES_DOUBLE]);
      UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "field 4:CushionVolume: StatusCode %s", UA_StatusCode_name(uStatus));

      // works: why?
      UA_Variant val;
      UA_Variant_setScalar(&val, &CushionVolume, &UA_TYPES[UA_TYPES_DOUBLE]);
      uStatus = UA_Server_writeValue(server, bpr.targets->targetId.nodeId, val);
      if (uStatus != UA_STATUSCODE_GOOD) {
         UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "field 4:CushionVolume: UA_Server_writeValue StatusCode %s", UA_StatusCode_name(uStatus));
      }
```

Using _UA_Server_writeValue_ to set the event field value directly is okay for me, I just wanted to inform you that I can't get the UA_Server_writeObjectProperty_scalar function working.

Thanks for your support.

Pro, 2019-05-16:
Are you sure that `CushionVolume` is a property, i.e., `Event_InjectionUnitCycleParameters_1Id -> hasProperty -> CushionVolume`?

(You can quickly check the value of `bpr.targets->referenceTypeId` in your example)

`UA_Server_writeObjectProperty_scalar` only works on properties, not on other children.
See:
https://github.com/open62541/open62541/blob/4e38d391dde9e5d74463f0a17de245f8917675ca/src/server/ua_services_attribute.c#L1573

MartinaFelberFEM, 2019-05-17:
Hi!
Excuse me, this was my fault. `CushionVolume` has a HasComponent reference. 
Thank you very much!

PR #2083 (2019-12-17): [Review] - First Implementation of Alarms and Conditions
Facets to be implemented:
1- A&C Base Condition Server Facet (only mandatory)
2- A&C Enable Server Facet (only mandatory)
3- A&C Alarm Server Facet (only mandatory)
4- A&C Acknowledgeable Alarm Server Facet (complete)

First Commit is a test to add the ConditionId Field to Events which are Conditions when using UaExpert. The Commit is to create a PR indicating the work on the first implementation of Part 9.

#2041, fixes #1947, fixes #2738
closed 2019-05-17 by MartinaFelberFEM