open62541/open62541 #2741: how to scan connected devices without knowing ip  address

bnadem, 2019-05-16:
i want to scann devices supported opc ua withoud knowing ip address
is this possible?

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2019-05-16:
If you are in the same local network and the OPC UA Servers suport the local discory mechanisms it is possible. See https://github.com/open62541/open62541/tree/master/examples/discovery for examples.

bnadem, 2019-05-16:
funciton UA_Client_findServersOnNetwork need serverUrl as  parameter!
function UA_Client_findServers need serverUrl as parameter!

Pro, 2019-05-16:
Please read the OPC UA Specification Part 12 or a paper I wrote about this:

http://www.i6.in.tum.de/Main/Publications/Profanter2017a.pdf

You need a Local Discovery Server which allows you to achieve exactly what you are describing.

closed 2019-05-16 by Pro