open62541/open62541 #2742: how to read timestamp with value from client side "UA_Client_readValueAttribute"

bnadem, 2019-05-16:
how to  read timestamp with value from client side with old function "UA_Client_readValueAttribute" please, and not with UA_Client_Service_read

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-05-16:
Why not with UA_Client_Service_read?

bnadem, 2019-05-17:
i have done with ua_client_service_read, but i want only know ist this possibel with UA_Client_readValueAttribute?

teccheck, 2025-03-13:
I'll close this as abandoned. Please reopen if it's still relevant

closed 2025-03-13 by teccheck