open62541/open62541 #2747: Could not instantiate the SecurityPolicy for the UserToken when try connect username/password

reyanvaldes, 2019-05-20:
Hi, thank you so much for sharing this amazing work you guys are doing and supporting.
I am facing since some weeks same problem trying to connect to Kepware V6 using username & password with security police #None. I can connect if no username/password is used. I read multiple issues related, and based on recommendation I tried to include the supported security policies manually but with no success after multiple back and for. I got message 'Could not create securityContext', I am using the ca.crt and ca.key generated by the self-signed tool provided. Attached is section of code in C++ and screenshot. Please could you recommend where to looks, or the proper way to complete the token properties to allow get connection
[connect_username_password_kepware.txt](https://github.com/open62541/open62541/files/3199851/connect_username_password_kepware.txt)
Thank you so much for any info
[image]

 - [ X] open62541 Version (Master V0.3.0 Latest commit 0574b60):
 - [X ] Other OPC UA SDKs used (client or server): client
 - [ X] Operating system: Debian Jessie
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X ] Self-contained code example attached (95%)
 - [ ] Critical issue

jpfr, 2019-05-23:
"Could not create SecurityContext" is not very detailed.
It looks like a problem with the certificate.

- Do you get more information if you increase the log level?
Otherwise we will add more detailed log output to the code.
- Can you debug into the creation of the security context?

fanucwj, 2019-05-24:
the same situation when using UA_Client_connect_username() with username/password.  do you resovle this problem ?

reyanvaldes, 2019-05-24:
Sorry, Unfortunately no yet. Thanks

fanucwj, 2019-05-24:
i only  make use UA_Client_connect_username(client, "opc.tcp://190.1.1.2:412",name,secret),but can not connect to server.  use open62541.c/h  , sometimes i thought whether the library is well,and you?

fanucwj, 2019-05-24:
and i think it is very basic to use this fanuction ,but still not work. make me crayz

fanucwj, 2019-05-24:
hi.glad to see u. how long have u debug for this?------------------ 原始邮件 ------------------

 
 

 
&mdash;

reyanvaldes, 2019-05-24:
I think in some part of library very deep inside the token (user name and password) is creating but not encrypted, that is the problem, so when server reject the security token 
Thanks

reyanvaldes, 2019-05-24:
A couple of weeks, but seems it is not possible with current library version, doesn’t support it yet.

fanucwj, 2019-05-24:
how do u config the client? and i only use username/password without certify------------------ 原始邮件 ------------------

fanucwj, 2019-05-24:
do u mind to exchange wechat for convenient? where u?------------------ 原始邮件 ------------------

fanucwj, 2019-05-25:
hello 
this is my siample demo https://github.com/open62541/open62541/issues/2762
is it like this?

reyanvaldes, 2019-05-25:
Hello
I don’t mind, please what is your contact info to send you invitation. We are in USA. 
Also quick questions:
1-do you have possible solution for encrypted username/password, as individual  support service? It is very important for us. 
2- also we are trying with certificate/private key but facing other challenges. 

In general we would like to know if you offer priority support service? 

Thanks

fanucwj, 2019-05-25:
hello, i am new to study opcua and will do hard research at this project.  i would like to exchange experience on this

thanks for all

Pro, 2019-05-27:
Did you already have a look at the examples here:
https://github.com/open62541/open62541/tree/master/examples/encryption

First try to understand the example code and what is happening there. Then adapt that code to use it with a Kepware Server & Co. Especially try to use your own certificates and make sure that the example code is working for you. Then adapt it.

See also #934

heppth, 2019-06-04:
I think, this could be the same issue like #2574 and #2562. It seems like there is still a problem in the library. I have already tried to solve the problem to provide a fix, but unfortunately without success.

You can reproduce to problem with the cpp server from unified automation.

fanucwj, 2019-06-06:
i have already solved this problem successfully,referring to examples
https://github.com/open62541/open62541/tree/master/examples/encryption
thanks you guys @Pro @jpfr  contribition and help

jpfr, 2019-06-06:
You're welcome

closed 2019-06-06 by jpfr

opcua-tsn-team-kalycito, 2019-06-11:
This pull request #2804 will resolve the issue.