open62541/open62541 #2749: Unable to write different values to a node in the pubsub application

SKShah36, 2019-05-21:
## Description
I want to publish a random integer (or system time) and have the subscriber receive it. I can't write the value to the variable node more than once.
## Background Information / Reproduction Steps
I have added the variable node to the information model and I use UA_Server_writeValue function to update the integer values. 

The relevant code snippets are as follows:

*Add the variable function*
[image]

*Adding dataset field function*
[image]

*Calling the functions in run loop*
[image]
 
Attaching the full code:

[tutorial_pubsub_.zip](https://github.com/open62541/open62541/files/3203879/tutorial_pubsub_.zip)

Any help is greatly appreciated

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2019-05-22:
Have you checked the return value of UA_Server_writeValue?
Could you try to read the value after setting it? (UA_Server_readValue)

```
[...]
UA_Server_writeValue(server, timeNodeId, value);
UA_UInt64 readValue_I64 = 0; // Set to a known value
UA_Variant valueRead;
UA_Variant_init(&valueRead); // Make sure everything is initialized and does not point to random locations
UA_StatusCode  retvalRead = UA_Server_readValue(server, timeNodeId, &valueRead);
if(retvalRead == UA_STATUSCODE_GOOD && UA_Variant_hasScalarType(&valueRead, &UA_TYPES[UA_TYPES_UINT64])
{
  readValue_I64 = *(UA_UInt64*) valueRead.data;
  // Check if Values are equl
}

```

SKShah36, 2019-05-22:
I am indeed able to set the value once. The returned status is good. And I can read the value just fine. My concern is the function doesn't get called more than once. Is that the expected behaviour?

*The updated code*:
[image]

*The output:*
[image]

merlingeorge555, 2019-05-27:
Hello @SKShah36,

To publish random values continuously,

- AddRepeatedCallback function shall be used
 for Ex: UA_Server_addRepeatedCallback(<server instance>, <user defined callback function>, NULL, <Call back interval in ms>, NULL)
 - Then initialize the user defined callback function 
 - Use UA_Server_writeValue() to write values into node

**FYI,**
**UA_Server_addRepeatedCallback()** sample in https://github.com/open62541/open62541/blob/4e38d391dde9e5d74463f0a17de245f8917675ca/examples/pubsub/tutorial_pubsub_subscribe.c

**UA_Server_writeValue()** sample in https://github.com/open62541/open62541/blob/84f499b9320f6993213010368480e3ccf307e81a/examples/tutorial_server_variable.c

Answer for your question:
I am indeed able to set the value once. The returned status is good. And I can read the value just fine. My concern is the function doesn't get called more than once. Is that the expected behaviour?

>> Yes.

Hope my answers helps you.

SKShah36, 2019-05-27:
Thanks @merlingeorge555. I understand that UA_Server_writeValue() is used to set/change the value on a particular node. But in the tutorial_server_variable.c the function is only called once. My question is how or in which part of the code do I call the writing function such that it gets called again and again in order to write the values on the node. When I place it in the run, it just gets called once and hence the value gets updated only once.

The following is what I believe is the loop function but correct if I am wrong.

static int run(UA_String *transportProfile,
               UA_NetworkAddressUrlDataType *networkAddressUrl) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setDefault(config);

    UA_DataType *types = (UA_DataType*)UA_malloc(sizeof(UA_DataType));
    UA_DataTypeMember *members = (UA_DataTypeMember*)UA_malloc(sizeof(UA_DataTypeMember) * 2);
    members[0] = Point_members[0];
    members[1] = Point_members[1];

    types[0] = PointType;
    types[0].members = members;

    UA_DataTypeArray customDataTypes = {config->customDataTypes, 1, types};
    config->customDataTypes = &customDataTypes;

    /* Details about the connection configuration and handling are located in
     * the pubsub connection tutorial */
    config->pubsubTransportLayers =
        (UA_PubSubTransportLayer *) UA_calloc(2, sizeof(UA_PubSubTransportLayer));
    if(!config->pubsubTransportLayers) {
        UA_Server_delete(server);
        return EXIT_FAILURE;
    }
    config->pubsubTransportLayers[0] = UA_PubSubTransportLayerUDPMP();
    config->pubsubTransportLayersSize++;
#ifdef UA_ENABLE_PUBSUB_ETH_UADP
    config->pubsubTransportLayers[1] = UA_PubSubTransportLayerEthernet();
    config->pubsubTransportLayersSize++;
#endif

    addPubSubConnection(server, transportProfile, networkAddressUrl);
    addPublishedDataSet(server);

    addMyStructDataType(server);

    addIntegerVariable(server, 1, 52525, UA_UInt32_random());
    addTimeVariable(server, 1, 52510);
    addStringVariable(server, 0, 53535);
    addPointVariable(server, 0, 12445);

    addDataSetField(server);
    addNewDataSetField(server, 1, 52525, "Integer");
    addNewDataSetField(server, 1, 52510, "TimeInteger");
    addNewDataSetField(server, 0, 53535, "String");
    addNewDataSetField(server, 0, 12445, "Point");

    writeVariable(server, UA_NODEID_NUMERIC(1, 52510));

    addWriterGroup(server);

    addDataSetWriter(server);

    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

merlingeorge555, 2019-05-28:
Hello @SKShah36,

Question1: My question is how or in which part of the code do I call the writing function such that it gets called again and again in order to write the values on the node. When I place it in the run, it just gets called once and hence the value gets updated only once.

>> We don't see writing function methods that works repeatedly for a given interval. The solution for your requirement is using the repeated call back method which I explained above. Please see my code below that writes values to the server nodes repeatedly.

Question 2: When I place it in the run, it just gets called once and hence the value gets updated only once. The following is what I believe is the loop function but correct if I am wrong.

>> No, you are wrong. Run() called only once. Only UA_Server_run() is called for every publish interval which is present within the Run() method. But you can't write nodes with UA_Server_run() which is stack implementation. Better use the user defined callback function as explained below in code section.

`static int run(UA_String *transportProfile,
               UA_NetworkAddressUrlDataType *networkAddressUrl) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setDefault(config);

    UA_DataType *types = (UA_DataType*)UA_malloc(sizeof(UA_DataType));
    UA_DataTypeMember *members = (UA_DataTypeMember*)UA_malloc(sizeof(UA_DataTypeMember) * 2);
    members[0] = Point_members[0];
    members[1] = Point_members[1];

    types[0] = PointType;
    types[0].members = members;

    UA_DataTypeArray customDataTypes = {config->customDataTypes, 1, types};
    config->customDataTypes = &customDataTypes;

    /* Details about the connection configuration and handling are located in
     * the pubsub connection tutorial */
    config->pubsubTransportLayers =
        (UA_PubSubTransportLayer *) UA_calloc(2, sizeof(UA_PubSubTransportLayer));
    if(!config->pubsubTransportLayers) {
        UA_Server_delete(server);
        return EXIT_FAILURE;
    }
    config->pubsubTransportLayers[0] = UA_PubSubTransportLayerUDPMP();
    config->pubsubTransportLayersSize++;
#ifdef UA_ENABLE_PUBSUB_ETH_UADP
    config->pubsubTransportLayers[1] = UA_PubSubTransportLayerEthernet();
    config->pubsubTransportLayersSize++;
#endif

    addPubSubConnection(server, transportProfile, networkAddressUrl);
    addPublishedDataSet(server);

    addMyStructDataType(server);

    addIntegerVariable(server, 1, 52525, UA_UInt32_random());
    addTimeVariable(server, 1, 52510);
    addStringVariable(server, 0, 53535);
    addPointVariable(server, 0, 12445);

    addDataSetField(server);
    addNewDataSetField(server, 1, 52525, "Integer");
    addNewDataSetField(server, 1, 52510, "TimeInteger");
    addNewDataSetField(server, 0, 53535, "String");
    addNewDataSetField(server, 0, 12445, "Point");

    writeVariable(server, UA_NODEID_NUMERIC(1, 52510));

    addWriterGroup(server);

    addDataSetWriter(server);

#########Merlin code snippets ############
   /*register a user defined call back function ie. writeCallbackthat calls for every 50ms interval*/
    UA_Server_addRepeatedCallback(server, writeCallback, NULL, 50, NULL);
#####################################
    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

#########Merlin code snippets ############
/* global definition */
UA_Variant           valueVariant;
UA_Boolean           actVal;

// this function called for every 50 ms after registering
static void writeCallback(UA_Server *server, void *data) {
actVal = false;
UA_Variant_setScalar(&valueVariant, &actVal, &UA_TYPES[UA_TYPES_BOOLEAN]);
UA_NodeId node_id = UA_NODEID_NUMERIC(1, 52510);
UA_Server_writeValue(server, node_id , valueVariant);
}
########################################
`

SKShah36, 2019-05-28:
Thanks a lot, @merlingeorge555. I probably asked the wrong question which in turn confused people in giving the answer that I wanted. I have updated the question. Closing the issue. Thanks, again!

closed 2019-05-28 by SKShah36