open62541/open62541 #2752: GCC compile error on Ubuntu 19.04

mrburen121, 2019-05-22:
Hello,

I'm trying to build the simple server application from [https://open62541.org/doc/current/tutorial_server_firststeps.html](url). With cmake I created the amalgamation open62541.c/.h files.
Used CMake options:
`cmake -DBUILD_SHARED_LIBS=ON -DUA_BUILD_EXAMPLES=OFF -DUA_ENABLE_AMALGAMATION=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..`

I placed the myServer.c , open62541.c and open62541.h files in the same folder.
> /Documents/projectopcua/

Then running the command `gcc -std=c99 open62541.c myServer.c -o myServer` I get this error:

> usr/bin/ld: /tmp/ccjkhm5G.o: in function `UA_ServerConfig_setMinimal':
> myServer.c:(.text+0x130): undefined reference to `UA_ServerConfig_setMinimalCustomBuffer'
> /usr/bin/ld: /tmp/ccjkhm5G.o: in function `main':
> myServer.c:(.text+0x1ca): undefined reference to `UA_Server_getConfig'
> collect2: error: ld returned 1 exit status

Is it somethiong wrong on my side or is there a problem with the sample code from the documentation?

## Checklist

Please provide the following information:

 - [X] open62541 Version: 0574b60 (last commit)
 - [ ] Other OPC UA SDKs used (client or server):
 - [X] Operating system: Ubuntu 19.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-05-27:
This looks like you are mixing different versions of the library (i.e. `0.3` vs. `master`.

Make sure that you are compiling the `master` version (or download it here: https://open62541.org/releases/) and use the corresponding examples.

Pro, 2019-05-28:
Please re-open if this doesn't fix it.

closed 2019-05-28 by Pro

kusumy, 2019-11-18:
I have similar problem, but on the Windows 10. 
I also have used open62541.c & open62541.h from release master.