open62541/open62541 #2753: deadband in client side

tuxroger, 2019-05-22:
This is related to issue (https://github.com/open62541/open62541/issues/1838).
I am trying to configure a monitored item using a relative deadband but I cannot find anywhere (in the examples) how to configure the monitored item deadband.

Here is the snippet of the part of my code where I create a subscription:

		UA_MonitoredItemCreateRequest monitoringRequest;
		UA_MonitoredItemCreateRequest_init(&monitoringRequest);
		monitoringRequest.itemToMonitor.nodeId = UA_NODEID_STRING(item->namespace_, (char*)item->identifier_.c_str());
		monitoringRequest.itemToMonitor.attributeId = UA_ATTRIBUTEID_VALUE;
		monitoringRequest.monitoringMode = UA_MONITORINGMODE_REPORTING;
		monitoringRequest.requestedParameters.samplingInterval = item->monitorResolution_;
		monitoringRequest.requestedParameters.discardOldest = true;
		monitoringRequest.requestedParameters.queueSize = 1;

		UA_MonitoredItemCreateResult monitoringResponse = UA_Client_MonitoredItems_createDataChange(
				uaClient_,
				subscriptionId_,
				UA_TIMESTAMPSTORETURN_BOTH,
				monitoringRequest,
				NULL,
				OPCUALogger::dataChangedCallback,
				NULL);

		if(monitoringResponse.statusCode == UA_STATUSCODE_GOOD)
		{
			item->id_ = monitoringResponse.monitoredItemId;
			subscribedItems_[item->id_] = *(item);
			std::cout << Utils::now_rfc3339() << "OPCUALogger::createMonitoredItems: Success. '" << item->identifier_ << "'" << std::endl;
		}
		else
		{
			std::cout << Utils::now_rfc3339() << "OPCUALogger::createMonitoredItems: Error. '" << item->identifier_ << "'. Reason '" << UA_StatusCode_name(monitoringResponse.statusCode) << "'" << std::endl;
		}

In the issue I mentioned above, they say:

You can add the necessary filter description when the MonitoredItem is created.

https://github.com/open62541/open62541/blob/master/include/ua_client_subscriptions.h#L145

But the link is broken.

Maybe this has changed since 2018? I don't know.

Thanks in advance for your time.

Roger.

WoutvdBerg, 2019-05-22:
I guess it depends on the version of open62541 you use, my code uses version 0.3.
I fixed the link for you: [](https://github.com/open62541/open62541/blob/0.3/include/ua_client_subscriptions.h#L145)

tuxroger, 2019-05-25:
Thanks for the response. But still, I cannot see any argument to that function that lets you specify any deadband so that the server does not notify if value does not change "enough". 
Does it mean that you have to do it on your own in the client side? Meaning, receive all the notifications and discard the ones that do not meet the deadband restrictions?
Thanks in advance for your time.
Roger.

Pro, 2019-05-27:
The corresponding link for `master` is https://github.com/open62541/open62541/blob/master/include/open62541/client_subscriptions.h#L144

Please check the OPC UA specification about Monitored Items to understand how to define filters. (https://opcfoundation.org/developer-tools/specifications-unified-architecture/ free after registering).

A full example for the deadband filter in client mode can be found in our unit tests:

https://github.com/open62541/open62541/blob/573f0f085280d25250d83ce9b6ae7817a40526fc/tests/server/check_monitoreditem_filter.c#L315

Look through that file to find more examples.

closed 2019-05-27 by Pro