open62541/open62541 #2755: Structure variable value not shown correctly on client

hansimais, 2019-05-22:
Hi open62541-Team,

we are currently switching from a commercial SDK to yours, so far we are very impressed and open62541 works great.

But now there is a problem working with structures. We've created a Euromap77 structure variable on the server side.

```C
    UA_StandstillReasonType r;
    r.id = UA_STRING("Someid");
    r.lockedByMES = UA_TRUE;
    r.text = UA_LOCALIZEDTEXT("en-US", "Sometext");
    
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "EuromapVar");
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    attr.dataType = UA_TYPES[UA_TYPES_EUROMAP83_STANDSTILLREASONTYPE].typeId;  

    UA_Variant_setScalar(&attr.value, &r, &UA_TYPES[UA_TYPES_EUROMAP83_STANDSTILLREASONTYPE]);

    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "EuromapVar");
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "EuromapVar");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);

    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);

    UA_StatusCode st = UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                                                 parentReferenceNodeId, currentName,
        variableTypeNodeId, attr, NULL, &newNodeID);
    if(st != UA_STATUSCODE_GOOD) {
        UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                       "EuromapVar failed. StatusCode %s", UA_StatusCode_name(st));
    }
```

When we try to read the value from the UaExpert / Prosys, it shows an INT32 Datatype with an invalid value.

[image]

We've figured out that this might be a client's issue, because your custom_datatype example works, where the datatype is compiled into the client.

This works with the non-free SDK, without adjusting the client.

Is it a server or UaExpert issue ?
Will it be fixed in the future without adapting the client side ?
Can we fix the problem on the server side ?

We're working on the test/euromap branch in pro's repo. (4b426faf28655f548b631bb6f8d0e71889d3ce6c)

Many thanks in advance

jpfr, 2019-05-23:
The standard defines a way to represent the structure of new datatypes in the information model.
Via type dictionary representations. Just via a set of special variables.
Some clients can read out and interpret the type dictionary at runtime.

We currently do not automatically create the dictionary for generated data types.
It should however be possible to manually create them in the information model or to copy them over from another server. Also, some standard information models (for example namespace zero from the main spec) come with the type dictionaries already baked into the XML nodeset.

In principle, the UA_DataType structure contains all the information that is needed for the type dictionary.
We could build a helper method to populate the information model.

I mark this issue as "help wanted".
We can gather some information in this issue on what is required to make open62541 type handling play nice for the Euromap standard. Thereafter we need a developer to step up to the task. 😄

Pro, 2019-05-27:
This is a duplicate of #2582 and the corresponding feature request is here: #2587

Currently you need to provide a .bsd file to the nodeset compiler, and also include the content of the .bsd in the nodeset. With this UaExpert is able to 'decrypt' the Extension Object.

A corresponding example is shown in https://github.com/open62541/open62541/blob/master/examples/nodeset/testnodeset.xml

Especially the BinarySchema Node is important:
https://github.com/open62541/open62541/blob/master/examples/nodeset/testnodeset.xml#L76

[image]

The use of the BinarySchema node is deprecated. Therefore we have #2587 where we need to implement the new DataTypeDescription properties for custom structures.

@hansimais you can try to open the Euromap77 Nodeset in e.g. UaModeler and export it again. This will create a .bsd file. Then base64 encode this .bsd file and set the base64 value for the BinarySchema node. Does that fix it?

davesliu, 2019-05-28:
@jpfr 
Per your above comments, the user defined structure can be only parsed by UaExpert dynamically Via type dictionary representations.

So is there any plan that open62541 support such features?

Because for current user defined data example codes in below link, it needs both the UA server & client to know the defined data structure. So it's not convenient if we can't modify the UA client program. 

https://github.com/open62541/open62541/tree/master/examples/custom_datatype

Pro, 2019-05-28:
> @jpfr
> Per your above comments, the user defined structure can be only parsed by UaExpert dynamically Via type dictionary representations.
> 
> So is there any plan that open62541 support such features?

Yes, see #2587. It's called DataTypeDefinition

> Because for current user defined data example codes in below link, it needs both the UA server & client to know the defined data structure. So it's not convenient if we can't modify the UA client program.
> 
> [/examples/custom_datatype@`master`](https://github.com/open62541/open62541/tree/master/examples/custom_datatype) [](/open62541/open62541/tree/HEAD@{2019-05-28T14:59:44Z}/examples/custom_datatype)

No. UA Expert is able to decode custom structures using the BinarySchema node. Read my comment above. It explains that concept.

Also this section of the Specification Part 3 may be interesting for you:

[image]

davesliu, 2019-05-28:
@Pro 
Thanks for your info :-)

NoelGraf, 2021-04-18:
I would then close the issue.

closed 2021-04-18 by NoelGraf