open62541/open62541 #2757: No suitable UserTokenPolicy found for the possible endpoints occured when connect to server with username/password

fanucwj, 2019-05-24:
can not make use the fuction UA_Client_connect_username(client, "opc.tcp://190.1.1.2:412",name,secret);  always can not connected with server.
using open62541.c/h in project...  is there any bug  in the library?
anyone can help me ?please! so urge!!

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-05-27:
Hello @fanucwj ,

it is possible that the server requires the password to be encrypted.
This is independent from the SecureChannel being encrypted.

Try to enable encryption in the cmake settings.
And configure the client similar to the examples that use encryption.

fanucwj, 2019-05-27:
> 
> 
> Hello @fanucwj ,
> 
> it is possible that the server requires the password to be encrypted.
> This is independent from the SecureChannel being encrypted.
> 
> Try to enable encryption in the cmake settings.
> And configure the client similar to the examples that use encryption.
thank you very much! i think your idears were very correct.
but could you please give me more details?
expect for your answer

fanucwj, 2019-05-27:
any examples on github?

fanucwj, 2019-05-27:
> 
> 
> Hello @fanucwj ,
> 
> it is possible that the server requires the password to be encrypted.
> This is independent from the SecureChannel being encrypted.
> 
> Try to enable encryption in the cmake settings.
> And configure the client similar to the examples that use encryption.

cmake -DUA_ENABLE_ENCRYPTION=ON -DUA_BUILD_EXAMPLES=OFF -DUA_BUILD_SELFSIGNED_CERTIFICATE=ON -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
this is the cmake i used, was it correct?

fanucwj, 2019-05-27:
@jpfr [2019-05-27 00:10:29.603 (UTC-0700)] error/userland	Not connected. Retrying to connect in 1 second
[2019-05-27 00:10:30.604 (UTC-0700)] info/client	Connecting to endpoint opc.tcp://192.168.1.2:4840
[2019-05-27 00:10:30.605 (UTC-0700)] info/client	SecurityPolicy not specified -> use default #None
[2019-05-27 00:10:30.605 (UTC-0700)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2019-05-27 00:10:30.607 (UTC-0700)] info/client	TCP connection established
[2019-05-27 00:10:30.610 (UTC-0700)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-05-27 00:10:30.610 (UTC-0700)] info/client	Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-05-27 00:10:30.613 (UTC-0700)] info/client	Found 3 endpoints
[2019-05-27 00:10:30.613 (UTC-0700)] info/client	Endpoint 0 has 2 user token policies
[2019-05-27 00:10:30.613 (UTC-0700)] info/client	Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
[2019-05-27 00:10:30.613 (UTC-0700)] info/client	Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2019-05-27 00:10:30.614 (UTC-0700)] info/client	Rejecting endpoint 1: security policy not available
[2019-05-27 00:10:30.614 (UTC-0700)] info/client	Rejecting endpoint 2: security policy not available
[2019-05-27 00:10:30.614 (UTC-0700)] error/client	No suitable UserTokenPolicy found for the possible endpoints
[2019-05-27 00:10:30.615 (UTC-0700)] error/userland	Not connected. Retrying to connect in 1 second
this is my log

Pro, 2019-05-27:
Did you already check out the examples here:
https://github.com/open62541/open62541/tree/master/examples/encryption

The log already gives you the problem:
```
[2019-05-27 00:10:30.605 (UTC-0700)] info/client SecurityPolicy not specified -> use default #None
[2019-05-27 00:10:30.613 (UTC-0700)] info/client Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
```

You need to configure a security policy for your client. See
https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c#L47

Please re-open the issue if the problem still exists and share your client code

closed 2019-05-27 by Pro

fanucwj, 2019-05-27:
thank you very much
but could u give more detail?   i used the client.c for test

fanucwj, 2019-05-27:
> Did you already check out the examples here:
> https://github.com/open62541/open62541/tree/master/examples/encryption
> The log already gives you the problem:
> [2019-05-27 00:10:30.605 (UTC-0700)] [...]
sorry  i still do not understand well,i use client.c  as demo to connect server,please give me more detail?
UA_StatusCode
UA_SecurityPolicy_None(UA_SecurityPolicy *policy, UA_CertificateVerification *certificateVerification,
                       const UA_ByteString localCertificate, UA_Logger logger) {
    policy->policyContext = (void *)(uintptr_t)logger;
 //   policy->policyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#None");
    policy->policyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
modfied like this?i do not think so

jpfr, 2019-05-28:
Use this as the base: https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c

You need to generate a certificate.
Then load the certificate and private key into the client.

fanucwj, 2019-05-28:
thank you very much for your answer! much appreciate!
could you give more detail for generating certificate and usage? any example?

jpfr, 2019-05-28:
See the code in /tools/certs/create_certificate.py

fanucwj, 2019-05-28:
so i need to copy the certs on server to my computer?to my trustList?
much apprivated for you answer!