open62541/open62541 #2758: Server crashes if events are generated frequently

andranik-s, 2019-05-24:
## Description

I have an OPC UA server application that under certain circumstances generates several events in quick succession. The application crashes when a client connects to it.

The provided example reproduces this issue. It is a modification of the `tutorial_server_events.c` example where a repeated callback is used to generate events every 40ms.

The issue seems to be related to https://github.com/open62541/open62541/issues/2072.

## Background Information / Reproduction Steps

[Self-contained code example](https://github.com/open62541/open62541/files/3216409/freqevents.c.txt)

Use repeated callbacks to generate events every ~40ms then connect to the server.

Used CMake options:

```bash
cmake -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=1
```

## Logs

`UA_LOGLEVEL = 100` [logs.txt](https://github.com/open62541/open62541/files/3215964/logs.txt)

Backtrace:
==11027== Invalid read of size 8
==11027==    at 0x174348: UA_Notification_isOverflowEvent 
==11027==    by 0x174C47: UA_MonitoredItem_ensureQueueSpace 
==11027==    by 0x174632: UA_Notification_enqueue 
==11027==    by 0x178A36: UA_Event_addEventToMonitoredItem 
==11027==    by 0x178C9B: UA_Server_triggerEvent

## Checklist

Please provide the following information:
 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04.2 LTS
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-05-27:
I can not reproduce this using your code.

For me it works fine and I can see the generated events in UaExpert.

Which commit are you on? Can you reproduce the issue on current master?

jpfr, 2019-05-27:
I can confirm using the code...
Working on a solution.

andranik-s, 2019-05-27:
Just reprodused it using the current master (https://github.com/open62541/open62541/commit/573f0f085280d25250d83ce9b6ae7817a40526fc).

commit 4aa7e448 (2019-05-28): Fix #2758; Add EventQueueOverflowEventType to minimal events nodeset
closed 2019-05-28 by jpfr