open62541/open62541 #2759: Subscriber implementation with Subscriber API

theresa95, 2019-05-24:
## Description
Hey there, I have configured a OPC UA server which publishes six sensor values from the raspberry pi. Furthermore, I have altered the original tutorial_pubsub_subscribe.c a bit so that I can subscribe to the published sensor values. 

Now to my question: From the perspective of a subscriber, I do not know how the sensor values are ordered within the DataSetMessage. I mean, I could find it out by running the publish and subscribe applications and look up the order of how the sensor values are printed out (because I know which values the sensors have normally).

So as suggested by @jpfr , I have downloaded the subscriber implementation from #2631 and I see that this version of the subscriber implementation now uses the subscriber API calls (addPubSubConnection, addReaderGroup, addDataSetReader and addSubscribedVariables). I have looked through the code in detail but I have still problems with understanding how this subscriber implementation works.

In the FillTestDataSetMetaData: As far as I understand, this method defines 4 fields on which to subscribe to, right? For each field, it defines a FIELD NAME as well! For example, the fourth field in this array of fields has the datatype "UA_TYPES_BOOLEAN" and the field name is "BoolToggle". Does this mean that if my server adds a DataSetField with the fieldNameAlias = "BoolToggle", I can successfully subscribe to the published field?
Does the order of subscribed fields matter? Must the fieldNameAlias be treated as a "unique identifier" (similiar to the nodeid) in order to subscribe to the correct variable node?

    static void
    addDataSetField(UA_Server server, int nsIndex, int numIdent, char fieldNameAlias){
        /* Add a field to the previous created PublishedDataSet */
        UA_NodeId dataSetFieldIdent;
        UA_DataSetFieldConfig dataSetFieldConfig;
        memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
        dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
        dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING(fieldNameAlias); //fieldNameAlias = "BoolToggle"
        dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
        dataSetFieldConfig.field.variable.publishParameters.publishedVariable =
                                                          UA_NODEID_NUMERIC(nsIndex, numIdent);
        dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
        UA_Server_addDataSetField(server, publishedDataSetIdent,
                                                          &dataSetFieldConfig, &dataSetFieldIdent);
    }

Please explain to me if I get it wrong! I would be very thankful for an explanation on how this works.

EDIT: Furthermore, I wasn't able to build this subscriber implementation. How can I build this implementation? I tried to build it like the open62541-master:
1. git clone https://github.com/open62541/open62541.git
2. cd open62541
3. git submodule update --init --recursive 
4. mkdir build && cd build
5. <path-to>\cmake-gui.exe .. -G "Visual Studio 15 2017" // it is easier using the cmake-gui
6. Browse for the source code and the build folder: for picture lookup ma images folder and search for the image called cmakegui_open62541_building_options.PNG
7. Tick the building options UA_BUILD_EXAMPLES, UA_ENABLE_PUBSUB, UA_ENABLE_PUBSUB_INFORMATIONMODEL, UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS and choose for UA_NAMESPACE_ZERO the "FULL"-option.
8. Click "Configure"
9. By clicking "Generate", the build folder will be filled with all visual studio project files.
10. For building all examples, open the open62541.sln file in Visual Studio 2017.
11. In the _CmakePredifinedTargets folder, one can build the ALL_BUILD project. This will generate all example executables.
12. One will find the generated executables in: ..\open62541\build\bin\examples

I would really like to run this subscriber implementation and try it out by myself! 

## Checklist

Please provide the following information:

 - [x] Operating system: Windows 10 (subscriber), raspbian (server)
 - [ ] Wireshark network dump attached
 - [x] self-contained code example attached
 - [x] Critical issue: for me it is critical ;)

merlingeorge555, 2019-05-26:
Hello @theresa95,

Question 1: From the perspective of a subscriber, I do not know how the sensor values are ordered within the DataSetMessage. I mean, I could find it out by running the publish and subscribe applications and look up the order of how the sensor values are printed out (because I know which values the sensors have normally).

>> First, the subscriber implementation in the #2631 is basic hardcoded implementation. As per specification the metadata shall be published initially and subscriber should use the metadata to decode  but this commit doesn't seems to have that. The metadata is harcoded in the subscriber using FillMetadata() function. 

To make you understand the order/sequence of data published and subscribed is,
 the subscriber, subscribes the data in the reverse order in which the publisher is been sent. Say for example: If you publish:  [0] temperature_int, [1] pressure_float
then the subscriber code should be decoded in [0] float [1] int -> reverse ordered datatype should be configured in the subscriber, else you might get error in subscriber console as "Error: K value"

Question 2: As far as I understand, this method defines 4 fields on which to subscribe to, right? For each field, it defines a FIELD NAME as well! For example, the fourth field in this array of fields has the datatype "UA_TYPES_BOOLEAN" and the field name is "BoolToggle". Does this mean that if my server adds a DataSetField with the fieldNameAlias = "BoolToggle", I can successfully subscribe to the published field?

>> Yes, perfect understanding.

Question 3: Does the order of subscribed fields matter?

>> Yes it matters. The subscriber decodes using the FillTestDataSetMetaData() which should be in reversed order in which data is been published. See example in question 1 explained in detail.

Question 4: Must the fieldNameAlias be treated as a "unique identifier" (similiar to the nodeid) in order to subscribe to the correct variable node

>> Yes.

Question 5 : Build issue.

>> Change UA_NAMESPACE_ZERO the "FULL"-option. to **"REDUCED"**. This should work.

Hope this answers helps you.

theresa95, 2019-05-29:
Thank you @merlingeorge555 for your detailed answer, I really appreciate it! :) I was able to build the subscriber application by changing the UA_NAMESPACE_ZERO from "FULL" to "REDUCED". 
I have also executed the application and it seems like the subscriber application does not work as expected. As I have already described above, the subscriber application defines 4 hardcoded DataSetFields. The first field is a timestamp field. So I think this field is subscribed to the timestamp field which is published by the publish application (the publish application solely publishes timestamp field). 

But the output of the subscriber application does not show any values! It looks like this:
[image]

It seems that the message string is null. 

I have already opened a new issue on this. Please have a look at issue #2771 where I do more in detail on this topic. I would really appreciate your help on this!

merlingeorge555, 2019-05-31:
Hello @theresa95,

Can we close this issue? I have given my comments in the issue #2771 

Thanks.

theresa95, 2019-05-31:
hey merlingeorge555 ,  I have just looked over it and tried some out
things. And I would like to give a big thank you to your very detailed
answers!
They do help me a lot to understand more and more.

However, concerning your answer on issue #2771, I would like to ask you
some more questions:
Question 1: What do you mean by "use the client gui and browse the
information model when the subscriber is running"?
Which client gui do you mean? Do you mean I should run the subscriber
tutorial application and for example, UAExpert concurrently?

Question 2: I would really like to read and print the subscribed values in
the tutorial application (as I use this application as the basement for
further development).
I would not only like to print them out on the console, but store them in a
variable so I could write them in a csv file. As you have already suggested
in your answer,
I can use the UA_Server_readValue() API in a separate callback and read
repeatedly. But I have some questions in order to do so:

   - So can I define a function myFunc which calls the function
   UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant
   *outValue) and then cast the UA_Variant outValue to for example DateTime?
   Is it correct to call myFunc in ua_pubsub.c line 1098 as this function will
   get called every time a new NetworkMessage is received?
   - The second parameter of the function UA_Server_readValue is the nodeid
   of the datetime variable. But how can I get this nodeId on runtime? How can
   I retrieve the nodeId of the timestamp node which publishes the timestamp
   on the serverside?

Again, I want to thank you for your support on my questions!

Kindest regards,
Theresa

theresa95, 2019-05-31:
hey @merlingeorge555 ,  I have just looked over it and tried some out
things. And I would like to give a big thank you to your very detailed
answers!
They do help me a lot to understand more and more.

However, concerning your answer on issue #2771, I would like to ask you
some more questions:
Question 1: What do you mean by "use the client gui and browse the
information model when the subscriber is running"?
Which client gui do you mean? Do you mean I should run the subscriber
tutorial application and for example, UAExpert concurrently?

Question 2: I would really like to read and print the subscribed values in
the tutorial application (as I use this application as the basement for
further development).
I would not only like to print them out on the console, but store them in a
variable so I could write them in a csv file. As you have already suggested
in your answer,
I can use the UA_Server_readValue() API in a separate callback and read
repeatedly. But I have some questions in order to do so:

   - So can I define a function myFunc which calls the function
   UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant
   *outValue) and then cast the UA_Variant outValue to for example DateTime?
   Is it correct to call myFunc in ua_pubsub.c line 1098 as this function will
   get called every time a new NetworkMessage is received?
   - The second parameter of the function UA_Server_readValue is the nodeid
   of the datetime variable. But how can I get this nodeId on runtime? How can
   I retrieve the nodeId of the timestamp node which publishes the timestamp
   on the serverside?

Again, I want to thank you for your support on my questions!

Kindest regards,
Theresa

merlingeorge555, 2019-05-31:
Hello @theresa95,

merlingeorge555, 2019-06-01:
Hello @theresa,

**Question 1:** What do you mean by "use the client gui and browse the information model when the subscriber is running"? Which client gui do you mean? 

>> Client GUI is a open source tool that performs
>> 1. Connecting and disconnecting with publisher or subscriber
>> 2. Browsing the information model
>> 3. Showing attributes and references
>> 4. Subscribing to variable etc.., 

**Question 2:** Do you mean I should run the subscriber tutorial application and for example, UAExpert concurrently?
>> No, UAExpert is a properitary tool, but this open source tool (free OPCUA client GUI) does the same but with limited number of features. 
>> Look into this website, you will be having enough details and steps to run the tool. https://github.com/FreeOpcUa/opcua-client-gui 

**Question 3:** I would really like to read and print the subscribed values in the tutorial application (as I use this application as the basement for further development).
So can I define a function myFunc which calls the function UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant *outValue) and then cast the UA_Variant outValue to for example DateTime? Is it correct to call myFunc in ua_pubsub.c line 1098 as this function will get called every time a new NetworkMessage is received?

>> Better I would suggest, create a user defined callback in the tutorial_subscribe.c. Use the same code in  #2749 and inside that user defined callback use UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant *outValue) and then cast the UA_Variant outValue.
>> As now the node Id is required and questions seems to valid. See my answers to solve this below.

Question 4: The second parameter of the function UA_Server_readValue is the nodeid of the datetime variable. But how can I get this nodeId on runtime? How can I retrieve the nodeId of the timestamp node which publishes the timestamp on the serverside?
>> In src/pubsub/ua_pubsub.c line:1028 UA_Server_addVariableNode function the "UA_NODEID_NULL" says that the NODEID is NULL while creating target variables the NODE ID is set to NULL/with random number. So the solution is to create user define node id for each variables. So create array of user defined node string/ID on the start of the UA_Server_DataSetReader_addTargetVariables() in ua_pubsub.c and replace the "UA_NODE_NULL" as UA_NODEID_NUMERIC(1, array_of_nodeid[iteratorField]) in line:1028
>> Now the super foo loop in UA_Server_DataSetReader_addTargetVariables() in ua_pubsub.c runs till the number of target variables in tutorial_subscriber that means the nodeid for each variables are taken from hardcoded user defined array and iterate using iteratorField. So node ID for each variables as mentioned above point can be user defined.
>>[Note] the array size should be the number of target variables added in tutorial_subscribe.
>> finally use the nodeid in UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant *outValue) to read the value. Simple!!

If you need any help let me know

merlingeorge555, 2019-06-02:
Hello @theresa,

Question 1: What do you mean by "use the client gui and browse the information model when the subscriber is running"? Which client gui do you mean?

>> Client GUI is a open source tool that performs
>>   1. Connecting and disconnecting with publisher or subscriber
>>   2. Browsing the information model
>>   3. Showing attributes and references
>>   4. Subscribing to variable etc..,

Question 2: Do you mean I should run the subscriber tutorial application and for example, UAExpert concurrently?

>> No, UAExpert is a properitary tool, but this open source tool (free OPCUA client GUI) does the same but with limited number of features.
>> Look into this website, you will be having enough details and steps to run the tool. https://github.com/FreeOpcUa/opcua-client-gui

Question 3: I would really like to read and print the subscribed values in the tutorial application (as I use this application as the basement for further development).
So can I define a function myFunc which calls the function UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant *outValue) and then cast the UA_Variant outValue to for example DateTime? Is it correct to call myFunc in ua_pubsub.c line 1098 as this function will get called every time a new NetworkMessage is received?

> Better I would suggest, create a user defined callback in the tutorial_subscribe.c. Use the same code in #2749 and inside that user defined callback use UA_Server_readValue(UA_Server *server, const [...]

Question 4: The second parameter of the function UA_Server_readValue is the nodeid of the datetime variable. But how can I get this nodeId on runtime? How can I retrieve the nodeId of the timestamp node which publishes the timestamp on the serverside?

> In src/pubsub/ua_pubsub.c line:1028 UA_Server_addVariableNode function the "UA_NODEID_NULL" says that the NODEID is NULL while creating target variables the NODE ID is set to NULL/with random [...]

If you need any help let me know

NoelGraf, 2021-04-18:
I would then close the issue. Please reopen if required!

closed 2021-04-18 by NoelGraf