open62541/open62541 #2762: could not use function UA_Client_connect_username()?

fanucwj, 2019-05-25:
Dear all 
i am new to leanr open62541 and get demo test in pc with simens server,the example is
int main(int argc, char *argv[])
{
 UA_Client *client = UA_Client_new(UA_ClientConfig_default);
UA_StatusCode   status =   UA_Client_connect_username(client,"opc.tcp://151.131.1.2:4040","opcua","123");
  if(status != UA_STATUSCODE_GOOD) {
               UA_Client_delete(client);

               return status;
           }
}

the errro log :
No suitable UserTokenPolicy found for the possible endpoints
i run this program on  ubuntu  virtual machine
so i wonder if any configuration to set??
Thank you all!!!

!ATTENTION!
----------------------------------------------------------------------- ---------
-->

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

johndzilva, 2019-05-27:
Hi @fanucwj 
This is hard to debug. Could you add some information on available endpoint response from the server? (wireshark trace will do)

For open62541 server it requires UserTokenPolicy:username-policy and UserTokenType: UserName to connect

Pro, 2019-05-27:
This is a duplicate of #2757

Please do not post the same question multiple times. And next time please also fill out the issue template correspondingly. Otherwise we are not able to help you and will close the issue immediately.

closed 2019-05-27 by Pro