open62541/open62541 #2763: Can't run example: custom_datatype_server

MampfHamster, 2019-05-27:
## Description

Hello,

when i try to run the custom_dataType_server example, i get an exception throw at
```
UA_DataTypeArray customDataTypes = {config->customDataTypes, 1, types};
```
the Exception is a read access violation because config is nullptr
I used CMake options:

```bash
cmake -UA_NAMESPACE_ZERO=<FULL>
```
when i build everything wih 

```bash
cmake -UA_NAMESPACE_ZERO=<REDUCED>
```

the example runs without the exception

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 573f0f08
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Windows 7 & Visualstudio 2015
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-05-27:
Thanks for the issue report.
This probably doesn't have to do anything with custom datatypes...

If config is null, then the call to `UA_ServerConfig *config = UA_Server_getConfig(server);` failed.

Can you post the Log output? It's best that you set the loglevel to trace (`cmake -DUA_LOGLEVEL=100`)

I can not reproduce this on my Linux machine.

MampfHamster, 2019-05-28:
You were right the issue was not about the custom data types UA_Server_new() already returned a nullptr,
because UA_Server_initNS0 returned UA_STATUSCODE_BADTYPEMISSMATCH.
This all happend because i had an old Version of the submodules in my project.
After i updated it to the current version everything worked fine.

Thanks for the help

closed 2019-05-28 by MampfHamster