open62541/open62541 #2769: Build problem

AngerlFG, 2019-05-28:
## Description
I'm trying to build the library following this [guide](https://open62541.org/doc/current/building.html.). My intention is to run a server on a Raspberry Pi. I'm getting this error: 

**fatal: Not a git repository(or any of the parent directories): .git
CMAKE_BUILD_TYPE not given; setting to 'Debug' in this line**
https://i.imgur.com/DyHmlFu.png

2 times when I use the **cmake ..** command in the following process:
```
sudo apt-get install git build-essential gcc pkg-config cmake python

# enable additional features
sudo apt-get install cmake-curses-gui # for the ccmake graphical interface
sudo apt-get install libmbedtls-dev # for encryption support
sudo apt-get install check # for unit tests
sudo apt-get install python-sphinx graphviz # for documentation generation
sudo apt-get install python-sphinx-rtd-theme # documentation style

cd open62541
mkdir build
cd build
cmake ..
make

# select additional features
ccmake ..
make
```
## Background Information / Reproduction Steps
Enabled  Encryption option and Amalgation getting the error I mention

## Checklist

 - [ ] open62541 Version (release number or git tag): v0.3.0
 - [ ] Operating system: Raspbian

mlgiraud, 2019-05-29:
Is the directory a git repository, or did you download and extract the sources?
@jpfr The problem could be here (Line 43 and following in the main CMakeLists.txt):
```
if(GIT_FOUND)
    execute_process(COMMAND ${GIT_EXECUTABLE} describe --always --tags
                    RESULT_VARIABLE res_var OUTPUT_VARIABLE GIT_COM_ID WORKING_DIRECTORY ${CMAKE_SOURCE_DIR})
    if(${res_var} EQUAL 0 AND NOT OPEN62541_VER_COMMIT STREQUAL "")
        string(REPLACE "\n" "" OPEN62541_VER_COMMIT ${GIT_COM_ID} )
    endif()
endif()
```
Maybe we should first check wether or not we are in a valid git repository before executing the command?

AngerlFG, 2019-05-31:
I downloaded Single-file distribution and full source code, extracted in a folder open62541 and then all I did was in the **open62541/build** folder. 

**-Did you try calling make regardless? It should still work.**
Yes, I did and It generates .c and .h files. But when I enabled amalgation and Encryption option I get that error so I guess it doesnt enable these options . Sorry If i'm not giving enough detail, I'm totally noob "progamming".

Pro, 2019-08-30:
We recently changed how the git-based version detection works. That should fix this issue.

See #3053 and #3001

closed 2019-08-30 by Pro