open62541/open62541 #2771: Subscriber implementation not subscribing to timestamp?

theresa95, 2019-05-29:
## Description
I have downloaded the subscriber implementation from #2631 and I see that this version of the subscriber implementation now uses the subscriber API calls (addPubSubConnection, addReaderGroup, addDataSetReader and addSubscribedVariables). As suggested by @merlingeorge555 in issue #2759, I was finally able to build the subscriber implementation (by kalycity) by cmaking it with the option UA_NAMESPACE_ZERO="REDUCED" instead of "FULL".

I have run both the publish and the subscribe example. And the console outputs look like the following:
Publishing application (it is only publishing the timestamp):
[image]

Subscribing application (It is subscribing to 4 hardcoded DataSetFields. The first field is a DateTime type. So as far as I understand, it should be able to subscribe to the publishing timestamp field.) :
[image]

So as you can see in the subscriber console, the four VariableNodes are successfully added (first of which is of type DateTime). But it seems like the subscribe application is not able to parse the timestamp withing the NetworkMessage.

Also, I do not understand the console output of the publish application. It says that two nodes are not found. Why is that? Which nodes are they?

I would like to use this subscriber implementation to subscribe to my published sensorvalues. But since the application is not able to subscribe to the timestamp correctly, I will have to solve this issue first.

Any suggestions?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): I have downloaded the subscriber implementation from #2631
 - [ ] Other OPC UA SDKs used (client or server): No
 - [x] Operating system: Windows 10
 - [x] Critical issue

merlingeorge555, 2019-05-29:
Hello @theresa95,

Sounds good. 

Question 1:
Subscribing application (It is subscribing to 4 hardcoded DataSetFields. The rst field is a DateTime type. So as far as I understand, it should be able to subscribe to the publishing timestamp field.) 

>> Yes, the implementation seems to have a publisher that publishes the dateandtime for every 100ms interval. And also the subscriber should subscribe the same. Your understanding seems to be perfect.

Question 2: So as you can see in the subscriber console, the four VariableNodes are successfully added (first of which is of type DateTime). But it seems like the subscribe application is not able to parse the timestamp withing the NetworkMessage.

>> I see you misunderstood the console print in the subscriber, because if I'm not wrong you expected a print near the "Message Receive:". And it is empty because it is just a print statement in **"UA_ReaderGroup_subscribeCallback"** in /src/pubsub/ua_pubsub.c in line:2392

>> if you want to see the value of the "dateandtime" from the subscriber information model, use the client gui and browse the information model when the subscriber is running. Then you can see the data change. The subscriber implementation doesn't seems to have a UA_server_readValue() API to read the variable value from the information model and display the value in the console print. If you want to read and print in the tutorial application use the UA_server_readValue() API in a separate call back and read repeatedly 

Question 3: Also, I do not understand the console output of the publish application. It says that two nodes are not found. Why is that? Which nodes are they?

>> In the **publisher** if you browse the information model using the client tool you can find the objects without the parent node. And also it's not mandatory that objects should have a parent node name. Its a optional in modelling rule. That's the reason for this "Node id not found" console print. But to my understanding you don't need to worry about this, it means that it won't affect the publisher and subscriber working.

Question 4: I would like to use this subscriber implementation to subscribe to my published sensorvalues. But since the application is not able to subscribe to the timestamp correctly, I will have to solve this issue first.

>> You don't need to solve anything. It seems to be working fine.

Hope my answer helps.

Thanks,
Merlin

johndzilva, 2019-05-30:
@theresa95 If you really need your published sensor values on console, add a UA_LOG_INFO() function in the below mentioned line based on the log level with which you wanna operate.

https://github.com/opcua-tsn-team-kalycito/open62541/blob/23aec0cc21edc973bc6ba1e85eeeffbcb6170cd0/src/pubsub/ua_pubsub.c#L1097

theresa95, 2019-05-31:
hey @merlingeorge555 and @johndzilva ,
I would like to give a big thank you to your answers! 

As I using the subscribe application as a basement for further extended the application, I would not only like to print the timestamp value to the console, but also store them in a variable to write them in a csv file.

As I understand correctly, one way to do this is to define a function myFunc which calls UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant *outValue). myFunc then casts the outValue to DateTime and writes it into the csv file.  myFunc is called in ua_pubsub.c line 1098 where it is executed everytime a new NetworkMessage is received.

BUT: as you can see in the interface of the UA_Server_readValue(UA_Server *server, const UA_NodeId nodeId, UA_Variant *outValue), the second parameter is the nodeId of the serverside datetime node.
From the perspective of the subscriber, I do not know this nodeId. So how can I retrieve this nodeId on runtime (not hardcoded)?
As the subscribe application defined unique fieldNames for 4 fields (in the fillTestDataSetMetaData() mehthod), is it possible to get the nodeId by fieldname?

Thank you both for your help !

johndzilva, 2019-06-03:
@theresa95 Use the TranslateBrowsePathsToNodeIds Service to find the NodeId

theresa95, 2019-06-06:
hey @merlingeorge555,

 I would like to ask you how the client (which is subscribed to the
published sensorvalues) can WRITE a value on the server over the network.
Like: I want the Client to set the temperature value to 0.
How can I accomplish this?

I was thinking about something like this:

static void updateCurrentTime(UA_Server *server) {
    UA_DateTime now = UA_DateTime_now();
    UA_Variant value;
    UA_Variant_setScalar(&value, &now, &UA_TYPES[UA_TYPES_DATETIME]);
    UA_NodeId currentNodeId = UA_NODEID_STRING(1,
"current-time-value-callback");
    UA_Server_writeValue(server, currentNodeId, value);

}

But I cannot use the UA_Server_writeValue method because this is a
serverside method and I want to set the value from the client side.
Any suggestions?

Thanks a lot in advance!
Best regards

NoelGraf, 2021-04-18:
Does your problem still exist? Have you tried the 1.2 version in the meantime?

jpfr, 2021-09-23:
@theresa95 use the Client api to write.

closing due to inactivity.

closed 2021-09-23 by jpfr