open62541/open62541 #2773: Trying to read variable value periodically and asynchronously

tuxroger, 2019-05-29:
Hi, I am currently subscribing to a STATUSVALUE change of a certain value but it produces a lot of noise.
After applying deadband filter (thanks for the quick support you gave me), I am receiving very little data back.
As far as I remember, there is a way to specify a deadband (for the value) but you can tell the server to notify the client periodically so that even if the value does not change at all, you receive the notification anyway.
For instance, when you start data logging of a machine and one subsystem is working. Using status value data change I understand that the client will not receive any answer from the server until the subsystem stops. Anyway, I would like to receive the isRunning flag every 5 minutes, just in case.
Does it have anything to do with the  UA_DATACHANGETRIGGER_STATUSVALUETIMESTAMP flag?
In that case, how do you work with that flag?
I have not seen any test/example that is using this enum value in the sources...
Thanks in advance for your time!
Roger

RavilN, 2019-11-13:
Hi,
In general, in OPC UA, if your data is not changing for long time (can be just data is not changing, or deadband settings filter them out), and you want anyway periodically receive latest values, then client can call method Objects/Server/ResendData.
It takes as argument subscription ID, and after calling it publish response(s) will be sent to the client with up to date values.
I tried to run this with the server_ctt application in open62541, but method call returns BadInternalError, probably it is not implemented yet.
Unified Automation demo server handles it properly (verified by Wireshark).

NoelGraf, 2021-04-18:
What is the status of the issue? Can the issue be closed?

teccheck, 2025-03-13:
As there has been no further activity it seems this is resolved

closed 2025-03-13 by teccheck