open62541/open62541 #2774: LDS-ME does not work on Windows devices

royalheadwear, 2019-05-29:
## Description

Hello.

I'm currently trying to run the example LDS-Server on multiple device-combinations with different operating systems. I am using the following device-combinations:

| # | first device | second device |
|----------------------------|------------|---------------|
| 1 | Raspberry Pi with raspian stretch | Raspberry Pi with raspian stretch |
| 2 | Raspberry Pi with raspian stretch | Desktop PC with Windows 10 |
| 3 | Desktop PC with Windows 10 | Raspberry Pi with raspian stretch |

All devices are connected via Ethernet.

I'm just using the following three applications from the examples:

 - ``discovery_server_lds``
 - ``discovery_server_multicast``
 - ``discovery_client_find_servers``

I always start an instance of the ``discovery_server_lds`` on each device. Additionally I run an instance of the ``discovery_server_multicast`` on the first device. Then I run ``discovery_client_find_servers`` on the second device to show if I can find the running server instances.

With device-combination 1 everything works fine. But if I replace one of the devices with a Windows 10 machine, the servers on the other device can not be found. As far as I understood the LDS-ME-feature can be used to find servers and endpoints in the local network no matter on which device the server is running - as long as there is a LDS-ME on that device, too.

## Background Information / Reproduction Steps

I used the following CMake options to generate the library files and compile the examples:

``` BUILD_SHARED_LIBS                OFF
 CLANG_FORMAT_EXE                 CLANG_FORMAT_EXE-NOTFOUND
 CMAKE_BUILD_TYPE                 Debug
 CMAKE_INSTALL_PREFIX             /usr/local
 LIB_INSTALL_DIR                  lib
 MDNSD_LOGLEVEL                   300
 UA_ARCHITECTURE                  posix
 UA_BUILD_EXAMPLES                ON
 UA_BUILD_TOOLS                   OFF
 UA_BUILD_UNIT_TESTS              OFF
 UA_ENABLE_AMALGAMATION           OFF
 UA_ENABLE_COVERAGE               OFF
 UA_ENABLE_DISCOVERY              ON
 UA_ENABLE_DISCOVERY_MULTICAST    ON
 UA_ENABLE_ENCRYPTION             OFF
 UA_ENABLE_EXPERIMENTAL_HISTORI   OFF
 UA_ENABLE_HISTORIZING            OFF
 UA_ENABLE_METHODCALLS            ON
 UA_ENABLE_NODEMANAGEMENT         ON
 UA_ENABLE_QUERY                  OFF
 UA_ENABLE_STATIC_ANALYZER        OFF
 UA_ENABLE_SUBSCRIPTIONS          ON
 UA_ENABLE_SUBSCRIPTIONS_EVENTS   OFF
 UA_LOGLEVEL                      100
 UA_NAMESPACE_ZERO                REDUCED
```

As far as I could see with device-combination 1 the LDS on device 1 reacted on the startup of the LDS on device 2. This unfortunately was not the case in device-combination 2 and 3.

### Logs
#### Device combination 1
##### LDS on device1 (called NASpberrypi) with Raspian stretch:
```
pi@NASpberrypi:~/git/open62541/build $ bin/examples/discovery_server_lds
[2019-05-29 16:31:13.505 (UTC+0200)] info/network       TCP network layer listening on opc.tcp://NASpberrypi:4840/
[2019-05-29 16:31:13.506 (UTC+0200)] info/server        Multicast DNS: add record for domain: LDS-NASpberrypi._opcua-tcp._tcp.local.
[2019-05-29 16:31:13.608 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:31:13.709 (UTC+0200)] info/server        Multicast DNS: found server: opc.tcp://NASpberrypi:4840
[2019-05-29 16:31:13.709 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:31:14.413 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:31:16.419 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:31:20.278 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[... 57 lines cut ...]
[2019-05-29 16:51:14.104 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Der Vorgang wurde erfolgreich beendet.
```
##### LDS on device2 (called NASpberrypi) with Raspian stretch:
```
pi@NASpberrypi:~/git/open62541/build $ bin/examples/discovery_server_lds
[2019-05-29 16:51:16.010 (UTC+0200)] info/network       TCP network layer listening on opc.tcp://NASpberrypi:4840/
[2019-05-29 16:51:16.011 (UTC+0200)] info/server        Multicast DNS: add record for domain: LDS-NASpberrypi._opcua-tcp._tcp.local.
[2019-05-29 16:51:16.112 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:51:16.213 (UTC+0200)] info/server        Multicast DNS: found server: opc.tcp://NASpberrypi:4840
[2019-05-29 16:51:16.213 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:51:16.916 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
[2019-05-29 16:51:18.921 (UTC+0200)] debug/server       Multicast error: Can not write to socket. Resource temporarily unavailable
```

Watching these logs I recognized the tremendous number of *"Multicast error: Can not write to socket."*-messages. Because they appeared in all three device-combinations, I estimated that these messages did not cause the issues. 

## Checklist

 - [x] open62541 Version (release number or git tag): aed211f46e93e17462124d7d5a9342e216e0a415
 - [x] Other OPC UA SDKs used (client or server): None
 - [x] Operating system: Windows 10 & Raspbian stretch
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: If you need more logs then let me know
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached: Just examples were used
 - [ ] Critical issue

royalheadwear, 2019-05-31:
Today I tried to figure out where the actual issue of the communication was. therefore I installed Wireshark on the Windows 10 and the Raspian device. First there just was traffic from the the Raspian device to the Multicast group (224.0.0.251) shown on the Raspbian device. The traffic from the Windows Desktop device was not shown. 
On the Windows device there was no traffic with 224.0.0.251 as destination over the ethernet adapter. When I randomly switched on my wifi adapter additionally suddenly the instances of LDS found each other. After turning off the wifi adapter again, the LDS on the pi could find the LDS on the Windows desktop but not the other way arround.
Because of this behavior I assume:
1. the windows device is not able to receive multicast messages via wifi
1. the windows device is not able to send multicast messages via ethernet

The first point maybe occures because of some issues with UDP, multicast and wifi.

Is there any reason why the second issue occures? Do I need additional configuration to send multicast messages via the ethernet device?

jpfr, 2019-06-03:
This has happened before.
Windows does not know which IP range the device has for multicast.
So it tries the first network adapter. If that is not the correct one, no messages flow.

I have not found a good way to display the issue to users.
If there is a way, we should make debugging this a little easier.

Stasik0, 2019-06-17:
Just my 2 cent:

could/should we add an explicit multicast interface selection, by copying this line:
https://github.com/troglobit/mdnsd/blob/d4fc074d8a6ad72477bfbd86dacef96df1d09d49/src/mdnsd.c#L140

into here:
https://github.com/open62541/open62541/blob/6fe8453249dcfacc0983789fc9ea65a4ce2aa237/src/server/ua_discovery_manager.c#L22

at least for sending?

jpfr, 2019-06-25:
Thanks @Stasik0 !

@Pro, this sounds like simple fix.
You're the master of mdns. Green light?

Pro, 2019-06-25:
Yes, this should work. @jpfr feel free to create a PR which adds this line. If the unit tests run through it should at least not break existing code.

Stasik0, 2019-06-25:
I could try, bit will need to see/refresh the config stricture. Should the interface be passed through nl config structure?

Pro, 2019-06-25:
I guess this mDNS socket should also be refactored inside the Network Refactoring PR (#2408, @mlgiraud ?)

So probably it's enough to do a quick-fix here and then do a nicer implementation in #2408 

In #2694 I added a specific struct to the Server config for the discovery stuff. Maybe you can extend that one, i.e. here:
https://github.com/open62541/open62541/pull/2694/files#diff-b7dcf9e9bfaffd31b52d7fcb397b6d28R77

and add an optional IP address field for interface selection.
It's best if we can just use the first interface (the current code) if that field is not set. Otherwise try to use the given interface.

Does that make sense?

Stasik0, 2019-06-25:
it does, i guess the code i have seen lets you pick a "sting" interface and also an ip, which one would you preffer? Or should we take both?
Is it worth fixing in the code or should we wait until 2408?

Stasik0, 2019-06-30:
@royalheadwear:  could you test my pr #2856 please?

https://github.com/open62541/open62541/blob/9baf744268cd8b076838be7371c31924603c34dd/examples/discovery/server_multicast.c#L255

Pro, 2019-07-02:
@royalheadwear the fix from @Stasik0 is now in the master branch. That should fix it for you.
If not please reopen.

closed 2019-07-02 by Pro

royalheadwear, 2019-07-03:
I've just found time to check the changes. I've added a the Line from @Stasik0 's comment including the device's IP-addresses of the network interfaces the device should use. 

With this the Raspberry Pi's LDS could find the Windows LDS. Unfortuantly the Windows LDS does not show any messages beginning with "Multicast DNS: found server [...]". The Windows LDS even seems not to be found by itself.

I've recorded a network dump with wireshark on the windows device and added it as a screenshot (I added a filter to just show mdns-messages). The record was done while startup of the both LDSs. The packages seem to arrive at the Windows device, but the LDS doesn't use them.

[image]

Has anybody the same issue?

commit 1de8b24a (2019-07-02): First attempt of mDNS interface selection (#2856)

fixes #2774