open62541/open62541 #2776: How to compile the very most basic example with the Nodeset compiler?

lintamacar, 2019-05-29:
## Description

## Background Information / Reproduction Steps

I'm following the [documentation](https://open62541.org/doc/0.3/nodeset_compiler.html) for the nodeset compiler and before I try to do anything custom, I would like to compile the sample xml file exactly as it is shown in the tutorial. 

I'm pretty sure I have version 0.3. I grabbed that [tarball](https://api.github.com/repos/open62541/open62541/tarball/v0.3.0) on the [homepage](https://open62541.org/). I'm on a Compute Module right now (a sort of Raspberry Pi) running Debian—CLI only. I don't think my OS matters here because I get almost the same issue on my Windows 10 laptop. 

I summarized the issue I'm having in a [Stack Overflow post](https://stackoverflow.com/questions/56352062/how-can-i-compile-the-example-xml-file-from-the-open62541-tutorial). Basically, I just cd'd to *~/code/open62541-open62541-6249bb2/tools/nodeset_compiler/* and tried to run the command **python ./nodeset_compiler.py --types-array=UA_TYPES --existing ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml myNS.xml myNS**. Of course, I copied the sample xml code and saved it in that directory as *myNS.xml*, and I also found a [source](https://github.com/OPCFoundation/UA-Nodeset/blob/master/Schema/Opc.Ua.NodeSet2.xml) online to get the *Opc.Ua.NodeSet2.xml* file and dropped it in that *deps/ua-nodeset/Schema/* directory. 

This is the error I'm currently having: 
```
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing myNS.xml
Traceback (most recent call last):
  File "./nodeset_compiler.py", line 178, in <module>
    ns.allocateVariables()
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/nodeset.py", line 322, in allocateVariables
    n.allocateValue(self)
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/nodes.py", line 291, in allocateValue
    self.value.parseXMLEncoding(self.xmlValueDef, dataTypeNode, self)
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/datatypes.py", line 161, in parseXMLEncoding
    val = self.__parseXMLSingleValue(el, parentDataTypeNode, parent)
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/datatypes.py", line 281, in __parseXMLSingleValue
    extobj.value.append(extobj.__parseXMLSingleValue(ebodypart, parentDataTypeNode, parent, alias=None, encodingPart=e))
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/datatypes.py", line 223, in __parseXMLSingleValue
    alias=alias, encodingPart=enc[1], valueRank=enc[2] if len(enc)>2 else None)
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/datatypes.py", line 198, in __parseXMLSingleValue
    t.parseXML(xmlvalue)
  File "/root/code/open62541-open62541-6249bb2/tools/nodeset_compiler/datatypes.py", line 330, in parseXML
    self.value = int(unicode(xmlvalue.firstChild.data))
ValueError: invalid literal for int() with base 10: ''
```

I really don't understand what I'm doing differently from the tutorial here, or what I could be doing wrong. Does this work for anyone else? 

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

NoelGraf, 2021-04-18:
Does your problem still exist? Have you tried the 1.2 version in the meantime?

teccheck, 2025-03-13:
There hasn't been any response and this issue should be resolved

closed 2025-03-13 by teccheck

lintamacar, 2025-03-27:
I guess better late than never, but I don't even work at the company where
I had this question anymore lol