open62541/open62541 #2777: How to notify the client when variable node value was changed?

asor, 2019-05-30:
## Description
I would like to use events to notify the client when variable node value was changed. But I have problem how the client will get the data because events could not transfer data.
Is it possible to create property for BaseEventType for data transfer or client should get the data from originId in UA_Server_triggerEvent method?
Or maybe I am totally wrong and events are not the correct way for this purpose?
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_ENABLE_METHODCALLS='ON' -DCMAKE_BUILD_TYPE='Debug' -DUA_NAMESPACE_ZERO='FULL' -DUA_ENABLE_SUBSCRIPTIONS='ON' -DUA_ENABLE_SUBSCRIPTIONS_EVENTS='ON' -DUA_ENABLE_AMALGAMATION='ON'
```

## Checklist

Please provide the following information:

 - [x] open62541 Version : 0.4-mainline
 - [x] Other OPC UA SDKs used (client or server):  Server
 - [x] Operating system: Windows/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-05-30:
You can create your own subtype of the BaseEventType and then create an instance of that event.
Your subtype could then e.g. have an additional field for your custom data

jpfr, 2019-05-31:
This sounds like a case for datachange monitoreditems.
Events are overkill to monitor for data changes.

asor, 2019-06-03:
Thank you! Works very good.
I will put example code here if someone else wonder how to do it :)

 ```
/* Add event new field */
  UA_VariableAttributes vtAttr = UA_VariableAttributes_default;
  UA_StatusCode ret = UA_STATUSCODE_GOOD;
  UA_Byte initValue = 0;
  UA_NodeId outNewNodeId;

  vtAttr.dataType = UA_TYPES[UA_TYPES_BYTE].typeId;
  vtAttr.valueRank = UA_VALUERANK_SCALAR_OR_ONE_DIMENSION;
  vtAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Data");

  UA_Variant_setScalar(&vtAttr.value, &initValue, &UA_TYPES[UA_TYPES_BYTE]);

  ret = UA_Server_addVariableNode(server, UA_NODEID_NULL,
      UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE),
      UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
      UA_QUALIFIEDNAME(0, "Data"),
      UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
      vtAttr, NULL, &outNewNodeId);

  if (ret != UA_STATUSCODE_GOOD)
  {
    printf("%s(%d) Error %s \n ", __FUNCTION__, __LINE__, UA_StatusCode_name(ret));
    return OPCUA_STATUS_BAD;
  }

  ret = UA_Server_addReference(server, outNewNodeId, UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
      UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY), true);

  if (ret != UA_STATUSCODE_GOOD)
  {
    printf("%s(%d) Error %s \n ", __FUNCTION__, __LINE__, UA_StatusCode_name(ret));
    return OPCUA_STATUS_BAD;
  }
```
Write value to event field 
```
  UA_Variant Val;
  UA_StatusCode ret = UA_STATUSCODE_GOOD;

  ret = UA_Server_writeObjectProperty(hServer,
      EventNodeId, UA_QUALIFIEDNAME(0, "Data"), Val);
```

closed 2019-06-03 by jpfr

rudoq32, 2019-06-04:
Hi all. I see this issue is closed. I have one or two additional questions. In OP UA Part 4 [p. 60] 5.12.1.2 Sampling interval they say that _The Server may support data that is collected based on a sampling model or generated based on an **exception-based model**. The fastest supported sampling interval may be equal to 0, which indicates that the data item is exception-based rather than being sampled at some period. An exception-based model means that the underlying system does not require sampling and reports data changes._

That requires something like this (bellow), right?
`serverConfig->publishingIntervalLimits.min = 50.0;`
`serverConfig->publishingIntervalLimits.max = 3600.0 * 1000.0;`

`serverConfig->samplingIntervalLimits.min = 0.0;	// zero means event/exception based based`
`serverConfig->samplingIntervalLimits.max = 24.0 * 3600.0 * 1000.0;`

But once I tried to create monitored item with 
`UA_MonitoringMode monitoringMode = UA_MONITORINGMODE_REPORTING;`
`UA_TimestampsToReturn timestampsToReturn = UA_TIMESTAMPSTORETURN_BOTH;`
`UA_Double samplingInterval = 0.0;`
`UA_Int32 queueSize = 0;`
`// Configuration for subscription`
`UA_Double requestedPublishingInterval = 50.0;`
I am getting _BadInternalError_. I used UA_ENABLE_SUBSCRIPTIONS_EVENTS, UA_ENABLE_AMALGAMATION, UA_ARCHITECTURE_FREERTOSLWIP (among others).

Does it mean that my monitored item has to be Event based type?  (@Pro )

Thanks Rudo