open62541/open62541 #2778: OPEN62541 compilation

JustinKieu, 2019-05-30:
Hello,
I am just trying to build a simple client/server, but am running into the issue when compiling on Visual Studio. When using the latest master file from github, I run into the issue with ALL_BUILD being denied. After looking around, I found out that I need to set the starting point of the project somewhere else as that is related to the CMAKE and it is necessary. When using the 0.3 version from the website, I used the zip file and attempted to compile on visual studio. I got an error that says
[image]

```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag): 0.4
 - [ x] Other OPC UA SDKs used (client or server): 
 - [ x

] Operating system:Windows

traversaro, 2019-05-30:
When you get the error related to cmd.exe, typically checking out the output tab (near Error List) for more verbose errors can be useful.

JustinKieu, 2019-05-30:
Ok so i have this weird error where it says ua_namespace0.h  cannot be found, but I ran into this error for both the single source and the master files for the 
0.3 version. This is my other one
[image]

traversaro, 2019-05-30:
The actual error is `ModuleNotFoundError: No module named 'six'`, that refers to the fact that in the python installation used by open62541 to run its generators, the python six module is missing. 
This is a requirement described in the docs for 0.3 version, see https://open62541.org/doc/0.3/building.html#building-with-cmake-on-windows for more info (the mention is missing in the latest docs because the requirement has been removed from the latest version of the library, see https://github.com/open62541/open62541/pull/2493 .

Related issues:
* https://github.com/open62541/open62541/issues/1250 
* https://github.com/open62541/open62541/issues/1338

JustinKieu, 2019-05-30:
Ok thank you, I have to talk to my IT about getting a certificate for downloading six as it is blocked. Would this fix this error too?
[image]

traversaro, 2019-05-30:
I guess it will, that file is generated by a python script that uses `six`, so it is understandable why the build system does not  find it.

closed 2019-05-30 by JustinKieu

JustinKieu, 2019-05-30:
Is there a reason for many of the files not being able to be found in the master file? The myServer.c file has trouble looking for all the files that are required and I'm not sure if I should go and put the location of the file for every single one. My fellow coworker ran into an issue of many redefinition errors that we weren't sure if it was from us having to manually find each file and re-edit the include.