open62541/open62541 #2781: Historical data question

luibass92, 2019-06-04:
Hello everyone,

I am doing my first steps into the history plugin.
I am wondering how the history "actors" (Database, Backend and Gathering) should be used.

I have understand that I could have only one Database and only one Gathering in my project  while I may have different settings for each Node to historize and, consequently, different Backend.
I could also use always the same backend but, in general, I guess it should be different for each Node.
Then, through the Gathering's  registerNodeId(...) function I can register each node with its settings that include its backend implementation.

Is this vision correct or am I missing something?

I will tag @peter-rustler-basyskom and @basyskom-meerkoetter since I have already found your posts on this topic. 

Thanks,
Luigi

juangburgos, 2019-08-08:
https://blog.basyskom.com/author/prus/

teccheck, 2025-03-13:
There is an example for this now. Please take a look at that if it's still relevant, five years later

closed 2025-03-13 by teccheck