open62541/open62541 #2782: Unable to make pubsub files

mrburen121, 2019-06-04:
## Description
I'm unable to generate the pubsub_ethernet.h and pubsub_udp.h files. These are necessary to run the example found here [](https://open62541.org/doc/current/tutorial_pubsub_publish.html). I'm running on the newest version of the master branch, last commit #e84eaf3.
```
#include <open62541/plugin/pubsub_ethernet.h>
#include <open62541/plugin/pubsub_udp.h>
```

## Background Information / Reproduction Steps
```
cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -UA_ENABLE_PUBSUB=ON -DCMAKE_BUILD_TYPE=Debug -DUA_BUILD_EXAMPLES=ON ..
```
[image]

```
make
sudo make install
```
Below the end of the installed files.
[image]

Is the cmake command that I'm using wrong?

Kind regards,

Chris

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): #e84eaf3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu 19.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-05:
That is odd.
All the files listed in `default_plugin_headers` are normally installed.
And we add the pubsub files here.

```
if(UA_ENABLE_PUBSUB)
    list(APPEND default_plugin_headers ${PROJECT_SOURCE_DIR}/plugins/include/open62541/plugin/pubsub_udp.h)
    list(APPEND default_plugin_sources ${PROJECT_SOURCE_DIR}/plugins/ua_pubsub_udp.c)
    if(UA_ENABLE_PUBSUB_ETH_UADP)
        list(APPEND default_plugin_headers ${PROJECT_SOURCE_DIR}/plugins/include/open62541/plugin/pubsub_ethernet.h)
        list(APPEND default_plugin_sources ${PROJECT_SOURCE_DIR}/plugins/ua_pubsub_ethernet.c)
    endif()
endif()
```

Note that you need to activate UA_ENABLE_PUBSUB_ETH_UADP as well to get PubSub for raw Ethernet (and the header for it).

mrburen121, 2019-06-06:
Hello,

I created a new VM vith the same Ubuntu version. Now with the cmake commando:
`cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -UA_ENABLE_PUBSUB=ON -UA_ENABLE_PUBSUB_ETH_UADP=ON -DCMAKE_BUILD_TYPE=Debug -DUA_BUILD_EXAMPLES=ON ..`

The pubsub_ethernet.h and pubsub_udp.h are now created. I have no idea what was wrong with the other Ubuntu VM. But now the make commando crashes with the following error.
[image]
The pubsub tutorials are not able to be created and give some errors. It looks like some references can't be found. Am I missing some PUBSUB CMAKE commands?

jpfr, 2019-06-06:
Disabble -DBUILD_SHARED_LIBS=ON.
The subscribe example depends on internal API that is not exposed in the shared lib.

The PR with the full subscriber will land in the next days.

mrburen121, 2019-06-07:
Thank you, I'm able to create the samples.
I  will look into the problem with my old VM to find out why it wasn't creating the correct files.

closed 2019-06-07 by mrburen121