open62541/open62541 #2783: Updating nodes (deleting and adding nodes) during runtime from dedicated thread

Nik654, 2019-06-04:
## Description
Hello, 

I'm using server from open62541 SDK version 0.3, following the [documentation ](https://open62541.org/doc/0.3/server.html)and here it says "When creating dynamic node instances at runtime", means that nodes can be added, removed during runtime without restart.
What is not clear is behavior in following use-case.
Is it possible to update nodes (delete nodes and add new ones) during runtime from different thread, than one in which server is running.
## Background Information / Reproduction Steps
What kind of behavior is to be expected if following steps are done:
1. Load sever configuration
2. Create server instance
3. Add nodes (add few object instances and in each instance few variables)
4. Run the server in one thread (UAExpert can connect and browse nodes)
5. From second thread i am updating nodes (deleting nodes and adding new nodes), it can happen that this is done while in UAExpert nodes are browsed.

Can this approach be used with open62541 SDK, in other words is open62541 thread-safe.

If approach that we have taken is not recommended for open62541 SDK, what is your suggestion, how to resolve this possible issue?

Used CMake options:

```bash
cmake -DUA_ENABLE_FULL_NS0=ON -DUA_ENABLE_ENCRYPTION=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3 release
 - [x] Other OPC UA SDKs used (client or server): UAExpert used as OPC client
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2019-06-04:
Open62541 thread-safe in total, there are some minor things already implemented, but most of the library is not thread-safe.

To overcome this issue, I use the following code for the main loop (C++):
```
    std::unique_lock<std::remove_reference<decltype(OPCUA_Mutex)>::type> l_opcUA(OPCUA_Mutex, std::defer_lock);
    while (*run) {
      l_opcUA.lock();
      UA_Client_run_iterate(pClient.get(), 10);
      l_opcUA.unlock();
      std::this_thread::yield(); // < This line ensures that a waiting thread is executed
      std::this_thread::sleep_for(std::chrono::microseconds(10));
    }
```
OPCUA_Mutex is a std::mutex which is shared with the pointer to the OPC UA Client.

jpfr, 2019-06-05:
Hey @ccvca,

thanks for providing the answer.
Can we preserve this for the future?

Feel free to add you answer to a "Frequently Asked Questions" page in the wiki.

Nik654, 2019-06-10:
Just to check if i got you right @ccvca

What you are suggesting is to create a mutex in my code which will make access to open62541 SDK serialized (thread-safe)?

I should do something like this:

    std::mutex open62541Mutex;

THREAD 0: Thread where server loop is running
===============================================
    UA_StatusCode retval = UA_Server_run_startup(server);
    if(retval != UA_STATUSCODE_GOOD)
        return retval;
    while(*running) {
        open62541Mutex.lock();
        UA_Server_run_iterate(server, true);
        open62541Mutex.unlock();
    }
    return UA_Server_run_shutdown(server);
	
THREAD 1: Thread where delete and add of nodes is being called
===============================================
    open62541Mutex.lock();
    UA_Server_deleteNode(server, requestedNewNodeId, deleteReferences);
    UA_Server_addObjectNode(server, requestedNewNodeId, parentNodeId, referenceTypeId, browseName, typeDefinition, attr, nodeContext, outNewNodeId);
    UA_Server_addVariableNode(server, requestedNewNodeId, parentNodeId, referenceTypeId, browseName, typeDefinition, attr, nodeContext, outNewNodeId);
    UA_Server_setVariableNode_dataSource(server, requestedNewNodeId, dataSource);
    open62541Mutex.unlock();

ccvca, 2019-06-11:
Yes that is the way I'm using this. Instead of locking a mutex, I'm using unique_lock (https://en.cppreference.com/w/cpp/thread/unique_lock) and call the lock/unlock in the loop on this.
Also suggest to add these lines to you thread 0 loop, otherwise my other thread was very unlucky of locking the mutex (it has take up to 5s):
```
std::this_thread::yield(); // < This line ensures that a waiting thread is executed
std::this_thread::sleep_for(std::chrono::microseconds(10));
```

@jpfr I've no problem preserving this in an FAQ, but as this is a C library a solution in with pthread would be more useful.

rsnitsch, 2023-06-27:
The methods `UA_Server_addVariableNode` and others are marked as `UA_THREADSAFE` so I think this issue can be resolved.

I have yet to find a method that is *not* threadsafe (take that with a grain of salt because I'm new to open62541).

teccheck, 2025-03-13:
As mentioned by @rsnitsch these methods are thread safe now

closed 2025-03-13 by teccheck