open62541/open62541 #2784: Extending an existing standard model

miehe-dup, 2019-06-04:
## Description

I need an information model based on the standard EUROMAP 82.2 for open62541. The standard can be accessed [here](http://www.euromap.org/en/euromap82-2/).

## Background Information / Reproduction Steps

EUROMAP 82.2 depends on the EUROMAP 83.1 standard, which is included as ZIP distribution. Both depend on the device integration model OPC UA DI.

I can easily find the model files needed to integrate Opc.Ua.Di into open62541.  
However, using the datatypes generator and XML nodeset compiler, it seems impossible to compile the EUROMAP standards, as they only provide NodeSet2.xml, csv and xsd formats but no bsd files.

How can we proceed without rewriting our whole application on top of a different OPC UA stack?

Also, there's the question on how to create a Model.xml file (ref. https://opcua.rocks/custom-information-models/) based on the EUROMAP standard. But I guess, that's for another day.

As he's done some really sophisticated posts on this topic, I'll tag @Pro in this question.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0.3.0
 - [X] Other OPC UA SDKs used (client or server): none
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-07:
Are the bsd definitions contained in the nodeset-xml in a base64 blob?

Pro, 2019-06-07:
Thanks for your question.

There are a few solutions you could do:

1. If the your XML nodeset (in this case 82.2) does not define any custom datatypes, you do not need to care about the .bsd file and just call the nodeset generator without it. An example is given here:
https://github.com/open62541/open62541/blob/fcbbf7db7a579aa3a245582ec5ee2a5622d1fa65/examples/nodeset/CMakeLists.txt#L93

2. If your model has custom types but no separate .bsd file, you have two ways to get it:
2.1  Open the .xml file with UaModeler (from Unified Automation) and re-export it as .xml. This will also create the needed .bsd file. UaModeler is free if you want to export less than 100 nodes.
2.2. As @jpfr already mentioned, the .xml should normally include the content of the .bsd inside its XML definition.  There is a node with the Browse Name `Default Binary`, and then a child node which contains some base64 encoded string. Copy this string, use some online base64 converter and then save the converted string as .bsd file

3. Wait until #2587 is fixed (hopefully in the next two weeks) and then you do not need the deprecated .bsd anymore.

Regarding your additional question on how to create a custom information model based on my tutorial you linked:
I will extend the tutorial with such an example. It may be difficult if the Model.xml file is not given. If that's the case, then you probably need to write your NodeSet2.xml by hand or use some GUI tool. I will go in more detail into this in the tutorial. Let's not mix two issues here.

miehe-dup, 2019-06-07:
> your XML nodeset (in this case 82.2) does not define any custom datatypes [...] just call the nodeset generator without it
Well, I'm completely oblivious when it comes to CMake and wrote my own Makefile to include the single-file open62541 distribution.

What I'm more curious about: How did you determine that 82.2 does not define custom datatypes?  
What about BaseDataType.Enumeration.ControllerTypeEnumeration?

> UaModeler is free if you want to export less than 100 nodes.
Actually, the free version does only allow to export 10 nodes (basically useless here)

> There is a node with the Browse Name `Default Binary`, and then a child node which contains some base64 encoded string.
I found a node called `TypeDictionary` (`TypeDictionary_BinarySchema`) which seems to work, and it also has content for EUROMAP 82.2. Thanks!

miehe-dup, 2019-06-07:
Am I doing this right? ("$" = command prompt)

```
$ open62541/generate_datatypes.py --namespace=1 --type-csv=../model/Opc.Ua.Di.NodeSet2.csv --type-bsd=../model/Opc.Ua.Di.Types.bsd --no-builtin open62541/ua_types_di
$ open62541/generate_datatypes.py --namespace=2 --type-csv=../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.csv --type-bsd=../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.Types.bsd --no-builtin open62541/ua_types_eu83
$ open62541/generate_datatypes.py --namespace=3 --type-csv=../model/Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.csv --type-bsd=../model/Opc_Ua.EUROMAP82_2.RC1_00_1.Types.bsd --no-builtin open62541/ua_types_eu82

$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --existing ../model/Opc.Ua.NodeSet2.xml --xml ../model/Opc.Ua.Di.NodeSet2.xml open62541/ua_namespace_di
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Generating Code
INFO:__main__:NodeSet generation code successfully printed

$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EU83 --existing ../model/Opc.Ua.NodeSet2.xml --existing ../model/Opc.Ua.Di.NodeSet2.xml --xml ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml open62541/ua_namespace_eu83
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml
Traceback (most recent call last):
  File "open62541/nodeset_compiler/nodeset_compiler.py", line 178, in <module>
    ns.allocateVariables()
  File "open62541/nodeset_compiler/nodeset.py", line 322, in allocateVariables
    n.allocateValue(self)
  File "open62541/nodeset_compiler/nodes.py", line 291, in allocateValue
    self.value.parseXMLEncoding(self.xmlValueDef, dataTypeNode, self)
  File "open62541/nodeset_compiler/datatypes.py", line 167, in parseXMLEncoding
    self.value = [self.__parseXMLSingleValue(xmlvalue, parentDataTypeNode, parent)]
  File "open62541/nodeset_compiler/datatypes.py", line 206, in __parseXMLSingleValue
    t.parseXML(xmlvalue)
  File "open62541/nodeset_compiler/datatypes.py", line 572, in parseXML
    if not self.stringRepresentation == xmlvalue.localName:
AttributeError: 'NodeId' object has no attribute 'stringRepresentation'
```

closed 2019-08-19 by miehe-dup