open62541/open62541 #2785: Client certificates handling on open62541 server

Nik654, 2019-06-04:
## Description
Hello,

We are using server from open62541 SDK version 0.3, following the [documentation](https://open62541.org/doc/0.3/server.html), but regarding client certificates handling on server it seems there is not a lot of information.
We have a few questions regarding the usage of client certificates in the open62541 based server application.
## Background Information / Reproduction Steps
We are trying to set up and use trusted/revocation lists, and we are loading those lists from the filesystem.
UA_ENABLE_ENCRYPTION is enabled, and a server certificate is loaded, and passed through to the configuration method (UA_ServerConfig_new_basic256sha256).
If we for example assume, that on the first time startup of the server, both of those lists (trusted/revocation) are empty, what exactly happens when a OPC client connects to our server and sends its certificate.
Does it automatically go to "revocation" list, and if it does, is there a way to access that certificate, so we can inspect it, and potentially "trust" it by moving it to the trusted list (from our application which is using open62541 SDK with server public API).
If client certificate doesn't automatically go to "revocation" list, is there any way to manage trusted, and revoked certificates using the open62541 SDK server public API (without changing SDK or SDK plugins code)?

Used CMake options:

```bash
cmake -DUA_ENABLE_FULL_NS0=ON -DUA_ENABLE_ENCRYPTION=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3 release
 - [x] Other OPC UA SDKs used (client or server): UAExpert as OPC client
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

MarkusWeinand, 2019-06-06:
Hello,
if the TrustList is empty the Server will accept every Client Certificate.
If you want to do your own check of the certificate you have to provide your own Method for verifyCertificate.
You can set this Method in the Server-Config `config->certificateVerification.verifyCertificate` as a function Pointer.
The standard Method is here [ua_pki_certificate.c](https://github.com/open62541/open62541/blob/0.3/plugins/ua_pki_certificate.c) in Line 50.

Nik654, 2019-06-07:
Thank you for your response, i will close the issue.

closed 2019-06-07 by Nik654