open62541/open62541 #2787: Variables

JustinKieu, 2019-06-05:
Hello,
I was wondering if there was a way to read a variable from the client onto the server. I was able to set up a variable on the client, but can't read it on the server. I am simply just increment the variable on the client by one and am writing it onto the variable using UA_Client_writeValueAttribute which I suspect where I may be going wrong.
Thanks!

## Checklist

Please provide the following information:

 + [x ] open62541 Version (release number or git tag): 0.4(master) 
 - [ x] Operating system: Windows

jpfr, 2019-06-05:
Yes, you can read variables.
Use the Read service for it.

You can take one of the client code examples as the starting point.

JustinKieu, 2019-06-05:
I tried that, but I think my server is not reading the variable.
```
static void
    addVariable(UA_Client *client, UA_Int32 myInteger) {
    /* Define the attribute of the myInteger variable node */
    UA_VariableAttributes attr = UA_VariableAttributes_default;

    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    attr.description = UA_LOCALIZEDTEXT("en-US", "the answer");
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "the answer");
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Add the variable node to the information model */
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Client_addVariableNode(
        client, myIntegerNodeId, parentNodeId, parentReferenceNodeId, myIntegerName,
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
}

static void
writeVariable(UA_Client *client, UA_Int32 myInteger) {
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the answer");

    /* Write a different integer value */

    UA_Variant myVar;
    UA_Variant_init(&myVar);
    UA_Variant_setScalar(&myVar, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    UA_Client_writeValueAttribute(client, myIntegerNodeId, &myVar);
    printf("%i", myInteger);
}

UA_Boolean running = true;
static void
stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Received Ctrl-C");
    running = 0;
}

int
main(void) {
    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));

    UA_Boolean waitInternal = false;
    UA_Int32 value = 1;

    addVariable(client, value);

    while(running) {

        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                         "Not connected. Retrying to connect in 1 second");
            /* The connect may timeout after 1 second (see above) or it may fail
             * immediately on network errors */
            /* E.g. name resolution errors or unreachable network. Thus there should be a
             * small sleep here */
            UA_sleep_ms(1000);
            continue;
        }

        UA_Client_run_iterate(client, 1000);

        /* update value and write it to the node */
        value++;
        writeVariable(client, value);
        UA_UInt16 timeout = UA_Client_run_iterate(client, waitInternal);

        /* Now we can use the max timeout to do something else. In this case, we
           just sleep. (select is used as a platform-independent sleep
           function.) */

        UA_sleep_ms(2000);
    }
```
Thats my client where I'm just having it constantly increase the value by 1 every 2 seconds.
This is my server where I am trying to have it read the node the.answer and print out the value as the client increments. It skips over the if statement when I attempt to debug and I am not sure why.
```
 while(running) {

        UA_Server_run_iterate(server, 1000);

        /* Read attribute */
        UA_Int32 value = 0;
        printf("\nReading the value of node (1, \"the.answer\"):\n");
        UA_Variant *val = UA_Variant_new();
        retval = UA_Server_readValue(server, UA_NODEID_STRING(1, "the.answer"), val);

        if(retval == UA_STATUSCODE_GOOD && UA_Variant_isScalar(val) &&
           val->type == &UA_TYPES[UA_TYPES_INT32]) {
            value = *(UA_Int32 *)val->data;
            printf("The value is %i\n", value);
        }
        UA_sleep_ms(1000);
    }

```

JustinKieu, 2019-06-05:
I figured out the issue. Thanks!

closed 2019-06-05 by JustinKieu