open62541/open62541 #2788: Support for handling subscriptions & monitored items externally

lkaino, 2019-06-05:
## Description

Feature request:
Possibility to sample monitored items externally, not using the timer callbacks from the SDK.

I would implement it so that I get callbacks from the SDK whenever a monitored item is registered or removed. The callback would receive a context pointer to the monitored item which can later be passed to the SDK when providing a new sample for the item. It should also be possible for the callback to update the revised sampling interval in case of registering a new item.

I can currently implement this by setting the monitoredItemRegisterCallback in the server config. The problem is that the callback only receives the node ID for the item. I would then have to use UA_Server_write() to update new samples to the server. This is inefficient when using other node ID types than numeric as a hash for the node ID needs to be calculated on every sample.

This functionality is currently possible in the Unified Automation High Performance SDK.

## Background Information / Reproduction Steps

My use case is following:
I have several threads running on different periods (10ms, 100ms, 1000ms) providing the values that are then monitored through the OPC UA server. The OPC UA server is running on a separate thread with lower RT priority. When the monitored items are sampled from the server thread context, I cannot guarantee that each sample is recorded (assuming that the sampling interval is equal to the period on which the value is updated). By having this feature it would be possible to sample the values to the server from the threads that are producing the values.

jpfr, 2019-06-05:
>This is inefficient when using other node ID types than numeric as a hash for the node ID needs to be calculated on every sample.

Hmm okay. Do you have some numbers on the increase in the CPU load?
That would be important to know to make a tradeoff decision.

> When the monitored items are sampled from the server thread context, I cannot guarantee that each sample is recorded (assuming that the sampling interval is equal to the period on which the value is updated).

In general, to make sure you get all the updates, make the sampling interval just a bit faster than the update rate. Note that many clients can monitor a value. All with different intervals and time offsets from the last sample.

What you propose would introduce a tight coupling between data generation and he clients. I don't think it is a good idea.

Also, the overhead can be quite large. If a process generates values at many kHz and a client monitors at 1/s, we'd have to check all the time when to actually take the sample. Or overwrite intermediate samples.

lkaino, 2019-06-14:
I don't have measurements on the CPU overhead, but one can assume that calculating hash for long strings is considerably slower compared to passing a monitoredItem pointer.

In our system all values are produced at fixed intervals: 10 ms, 100 ms and 1000 ms. I know the producing interval for each value, so if a client requests the item at a faster interval than it is being produced, I can revise the interval.

If a value is produced at 10 ms rate, but a client requests it at 1000 ms rate, I will sample it at 1000 ms. According to Nyquist theorem, the client would have to sample the value at half the interval the value is produced, which is wasting CPU cycles on the server.

Another example:
We have a value produced at 10 ms rate. Two clients connect to the server, one monitors the value at 10 ms and the other at 100 ms. Every 100 ms I would need to read the value from my data store twice, to serve both clients. If it would be possible for me to manage the monitored items myself, I would always sample the value once, but every tenth time I would provide the sample also for the slower client.

These optimizations might sound small, but in our system we have multiple connected clients with each one reading over 2000 nodes per second. When scaling up the CPU overhead starts to add up.

jpfr, 2019-06-15:
Variables have a `MinimumSamplingInterval` attribute.
So for the values that are created at 1000ms, you can limit to exactly that.

See also the questions raised in #2814 on *Exception-based DataChange MonitoredItems*.

Feel free to make a design proposal for an additional API.
We need a proposal to see how much effort it is.

jpfr, 2021-09-23:
Now we can have a sampling interval of 0.0.
Then, every write triggers a check of the MonitoredItem.

closed 2021-09-23 by jpfr