open62541/open62541 #2789: What do you need to start the "example/server_encryption" ?

SebsSa, 2019-06-05:
## Description

I'm not sure what exactly I need to do to get the "example/server_encryption.c" server up and running.

## Background Information / Reproduction Steps

Hello guys, with your great tutorial I was able to create a simple server and also test it successfully. But I wanted to make my server more secure.

So I started with a simple login query. (User/Password)
It works but UAExpert gives me the "BadSecurityModeInsufficient -Error".

> BadSecurityModelInsufficient probably means that i try to use a "none" encrypted 
> communication.
 
After some research I came across the topic server-encrypting.
I was able to create the 
- server_cert.der
- server_key.der

`gcc -std=c99 open62541.c server_encryption.c -o mySecuredServer`

it throws :

```
server_encryption.c: In function ‘main’:
server_encryption.c:53:9: warning: implicit declaration of function ‘UA_ServerConfig_setDefaultWithSecurityPolicies’ [-Wimplicit-function-declaration]
         UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/cch5J7Ld.o: In function `UA_ByteString_clear':
server_encryption.c:(.text+0x1c7): undefined reference to `UA_clear'
/tmp/cch5J7Ld.o: In function `main':
server_encryption.c:(.text+0x4bb): undefined reference to `UA_Server_getConfig'
server_encryption.c:(.text+0x4f4): undefined reference to `UA_ServerConfig_setDefaultWithSecurityPolicies'
collect2: error: ld returned 1 exit status

```
Anyone have any idea what happened here? 

## Checklist

Please provide the following information:

 - [] open62541 Version (release number or git tag): 
 - [x] Other OPC UA SDKs used (client or server): UAExpert (Client)
 - [x] Operating system: Debian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-05:
Did you enable encryption in the cmake settings?

SebsSa, 2019-06-05:
after checking `cmake -L` in the build folder ->its OFF 

But now I'm a little confused. I did it this way: : 
`cmake UA_ENABLE_ENCRYPTION=ON ..`
`make`
What's wrong with that?

Tomorrow I'll reset everything and try again.

jpfr, 2019-06-05:
you can use ccmake or cmake-gui to see and adjust the cmake settings with more comfort.

SebsSa, 2019-06-06:
Okay big thanks, that seems to have fixed one of the problems, but I still get this :
```
/tmp/ccJjTAZx.o: In function `UA_ByteString_clear':
server_encryption.c:(.text+0x1c7): undefined reference to `UA_clear'
/tmp/ccJjTAZx.o: In function `main':
server_encryption.c:(.text+0x4bb): undefined reference to `UA_Server_getConfig'
server_encryption.c:(.text+0x4ef): undefined reference to `UA_ServerConfig_setDefaultWithSecurityPolicies'
collect2: error: ld returned 1 exit status
```
Do I need to add the linker optio when invoking gcc ?

jpfr, 2019-06-06:
You need to link against open62541.
Try to add `-lopen62541` and `-L<path_to_open62541>`.

With security you need to add the mbedtls link targets as well.

SebsSa, 2019-06-06:
I'm not sure if I made it better or worse.

I enabled "UA_ENABLE_AMALGAMATION" and got the new .h and .c files. 
I replaced the old ones and tried again with : 

`sudo gcc -std=c99 server_encryption.c open62541.c -o myServer `

Now I get this message :

```
/tmp/ccmGnYXg.o: In function `mbedtls_pk_get_len':
open62541.c:(.text+0x23f8): undefined reference to `mbedtls_pk_get_bitlen'
/tmp/ccmGnYXg.o: In function `certificateVerification_verify':
open62541.c:(.text+0x58ee1): undefined reference to `mbedtls_x509_crt_init'
open62541.c:(.text+0x58f08): undefined reference to `mbedtls_x509_crt_parse'
open62541.c:(.text+0x58f8c): undefined reference to `mbedtls_x509_crt_verify_with_profile'
open62541.c:(.text+0x59032): undefined reference to `mbedtls_x509_crt_free'
/tmp/ccmGnYXg.o: In function `certificateVerification_verifyApplicationURI':
```
Now I'm at the point of thinking about this, 

@jpfr : 
> With security you need to add the mbedtls link targets as well.

right?

jpfr, 2019-06-06:
Yes. It depends on your Linux distribution which libraries you need to add.
-lmbetls -lmbedx509 -lmbedcrypto, and so on,

SebsSa, 2019-06-06:
Thank you very much, sir,

I'll summarize it quickly for successors:

Use the latest version.
Better use ccmake or cmake-gui (clearer)
Enable : UA_ENABLE_ENCRYPTION and if necessary UA_ENABLE_AMALGAMATION (for new .h and c. open62541 files)

closed 2019-06-06 by SebsSa

anvitu, 2020-01-09:
Hello,
I'm trying to add a server certificate to a Simple Server. The certificate is loaded in the configuration server but my client Unified Automotion does not find the certificate.
I compile with this and have no errors:
`gcc -std=c99 open62541.c test.c -o test`

I think that is related to the commented above, it needs to compile with the flag UA_ENABLE_ENCRYPTION, but if I compile with
` gcc -std=c99 -DUA_ENABLE_ENCRYPTION -lopen62541 test.c -o test -lrt`

it will throw:
`/usr/local/lib/libopen62541.so: undefined reference to `__fdelt_chk@GLIBC_2.15'
/usr/local/lib/libopen62541.so: undefined reference to `clock_gettime@GLIBC_2.17'
collect2: ld devolvió el estado de salida 1
`

Here is the piece of code that add the certificate.
```
	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	config->serverCertificate = loadCertificate();
        if(config->serverCertificate.length > 0)
              UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Certificate loaded");

	UA_ServerConfig_setDefault(config);
```
Any help would be appreciate.

mojaxx, 2020-02-02:
hi 
i make and select the options in ccmake but i still got this error why ??

server.c:17:20: fatal error: common.h: No such file or directory
 #include "common.h"
                    ^
compilation terminated.

Regards