open62541/open62541 #2790: BadSessionIdInvalid from Client

MarkusWeinand, 2019-06-05:
I have setup a OPC-UA Server on a Siemens S7-1500 and try to wirte Data to it from a Client.
My client code looks like this:
`
 int main(int argc, char *argv[]) {
    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
    UA_Boolean value = true;
    while (true) {
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://192.168.25.131:4840");
        if (retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Not connected. Retrying to connect in 1 second");
            UA_sleep_ms(1000);
            continue;
        }
        value = !value;
        UA_Variant myVariant;
        UA_Variant_init(&myVariant);

        if (UA_Client_getState(client) == UA_CLIENTSTATE_SESSION) {
            UA_Variant_setScalarCopy(&myVariant, &value, &UA_TYPES[UA_TYPES_BOOLEAN]);

            UA_StatusCode retval = UA_Client_writeValueAttribute(client, UA_NODEID_STRING(4, "Start_Zaehler"), &myVariant);
            if (retval != UA_STATUSCODE_GOOD) {
                UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Something went wrong while trying to write to the Server");
            }

            UA_Variant_clear(&myVariant);
            UA_Client_run_iterate(client, 0);
        }
        
        retval = UA_Client_run_iterate(client, 30000);
        if (retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Something went wrong");
        }
    };

    /* Async disconnect kills unprocessed requests */
    // UA_Client_disconnect_async (client, &reqId); //can only be used when connected = true
    // UA_Client_run_iterate (client, &timedOut);
    UA_Client_disconnect(client);
    UA_Client_delete(client);

    return EXIT_SUCCESS;
}
`

The Client toggles the Value of the Node between treu and false every 30 Seconds.
But after a some time I get this Message "Received a ServiceFault response with StatusCode BadSessionIdInvalid". It is no Error log it just shows up under info. The client is running on a Raspberry Pi. It seems like the Client doesnt renew the Session by itself. I have tried to set the requestedSessionTimeout and secureChannelLifeTime to a very high number, but it did not change anything.
Has anybody had the same Problem before?
Does anybody know a fix or workaround for it?

Also i apologize because I have no clue how this Description thing below works.

## Description

## Background Information / Reproduction Steps

Used CMake options:
cmake .. -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=300 -DCMAKE_BUILD_TYPE=Debug -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON

```cmake .. -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=300 -DCMAKE_BUILD_TYPE=Debug -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): Current Master Branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Raspberry Pi
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x ] Self-contained code example attached
 - [ ] Critical issue

MarkusWeinand, 2019-06-11:
I added a wiresharkdump. It looks like the server is not responding to anything. The first response from the server comes after my writerequest and he immediately sends ServiceFault.
But I can connect with UA_Expert just fine.

jpfr, 2019-06-11:
Thanks for providing the dump.

So I see two open connections.
Did you manually remove some of the response messages?
I only partially see the SecureChannel and session handshake.

Only the second WriteRequest returns a failure in the provided dump:

> ServiceResult: 0x80270000 [BadSessionNotActivated]

But even before that the second open session is returning this on the publish requests:

> ServiceResult: 0x800a0000 [BadTimeout]

So it seems whatever causes the problem, it translates from one open session to the other.
Have you tried to connect our client without the UAExpert at the same time?

Sometimes the Siemens OPC UA servers get problems if many connections are open at the same time.

We can continue debugging when we see all the response messages.
In total, there seems to be a lot going on in the network. Is this wifi?

MarkusWeinand, 2019-06-12:
Thanks for your response,
I dont know why my tcpdump didnt capture all the packages. Here is an new tcpdump with hopefully all package.
[tcpdump.zip](https://github.com/open62541/open62541/files/3279959/tcpdump.zip)
The communication is not via wifi, but I know that there is a lot of traffic in our network.
But the Client has 192.168.25.31 and the Siemens S7 192.168.25.131, this should help filtering the tcpdump.

I didn't even know I had the UA Client running, this time I only had the open62541 Client running, but still got the same result.

But you can see that in the CreateSessionResponse SessionTimout is 30 sec. 
Now I just read the current server time to keep the session alive, like so.
```
UA_Variant value;
UA_Client_readValueAttribute(client, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME), &value);
```
I dont know if this is a good way of keeping the session alive or if there is a better solution?

JeremyMahieu, 2020-12-08:
I've got the same from Siemens S7-1500 (et200sp cpu) and Siemens SIMATIC RF680R RFID reader. I use the UPC UA Refference client's keepalive mechanism. First it respons with a `BadSessionNotActivated` then the next keepalive message is repsonden with `BadSessionIdInvalid` after which the clients know something is wrong and reconnects. 

Seems related: https://github.com/FreeOpcUa/python-opcua/issues/800
My issue: https://github.com/OPCFoundation/UA-.NETStandard/issues/1207

bitflower, 2020-12-16:
Hi, I'm using `node-opcua` and have a similiar behaviour @JeremyMahieu: I receive 10.000s of `ServiceFault` with `BadSessionNotActivated` followed by 5-6 `BadSessionIdInvalid` after which my whole process seems to restart:

[image]

bitflower, 2020-12-17:
In my case I had the `requestedSessionTimeout` set too low, around 5000. It needed 60000 though.

tobydox, 2021-05-05:
> 
> ```
> UA_Variant value;
> UA_Client_readValueAttribute(client, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME), &value);
> ```
> 
> I dont know if this is a good way of keeping the session alive or if there is a better solution?

Indeed this was a good hint. Adding a (publish) subscription of `UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME` (`i=2258, ns=http://opcfoundation.org/UA/`) makes the session errors for Siemens PLCs gone here. However we're using QtOPCUA with open62541 backend so your pure open62541-based example implementation likely would require additional code involving `UA_Client_Subscriptions_create()` and `UA_Client_MonitoredItems_createDataChange()`.

closed 2025-03-13 by teccheck